sebanc/brunch #130: Problems installing Chrome OS from live system to HDD

Ubsefor, 2020-05-12:
So, basically, after executing `chromeos-install -dst /dev/sda` from root user in live system and waiting for quite a while for partitions to flash, the only thing I get after trying to boot from HDD is  «brunch: chromeOS not found» and dropout to shell.
All partitions seem to flash successfully, at least they are labeled properly.

Is there any way to copy USB stick directly to HDD from ChromeOS live system? Or should I try installing it from linux live system (f.e Ubuntu) ?

sebanc, 2020-05-12:
Weird, this issue is a new one.

What you can do if your usb flashdrive is smaller in size than your hdd is run "dd if=<usb_flashdrive> of=<hdd> bs=1M" in crosh shell.

When you first boot from hdd, go to tty2 (CTRL+ALT+F2) and run "sudo resize-data" then reboot and hopefully all will be fine.

Ubsefor, 2020-05-12:
> What you can do if your usb flashdrive is smaller in size than your hdd is run "dd if=<usb_flashdrive> of= bs=1M" in crosh shell.

Am doing this from Ubuntu Live System, hopefully this will help. Actually, came to this solution not too long before you answered

> When you first boot from hdd, go to tty2 (CTRL+ALT+F2) and run "sudo resize-data" then reboot and hopefully all will be fine.

I’ll do that and report if there are any other issues

Ubsefor, 2020-05-12:
Well, it boots up and works without USB Drive Now, but resize-data did nothing

sebanc, 2020-05-12:
is the size of your hdd correctly recognized ?

Ubsefor, 2020-05-12:
> is the size of your hdd correctly recognized ?

Yes, it is correct: 465,8 Gb just as it is in linux distros

Ubsefor, 2020-05-12:
Maybe there is an error in GPT table? I will try to fix it with gparted from Ubuntu Live System

Ubsefor, 2020-05-12:
Ubuntu recognized the whole disk as a zfs filesystem and rpool volume! 
I recreated the GPT and now dd’ing Chrome OS again, will keep updated if anything else comes

Actually this may be the reason why Chrome OS didn’t install correctly in the first place

Ubsefor, 2020-05-12:
Well, those zfs leftovers were indeed the root of the evil, now as I checked – chrome OS installs just like in installation instructions, and dd method also worked!

So a bit of advise to all, facing unknown errors or having something not working out: check your partitions and partition table

closed 2020-05-12 by Ubsefor