sebanc/brunch #131: Cannot Access new Chrome OS installation

akashdur, 2020-05-13:
I followed the guide and installed Chrome OS on one of my partitions.Now clicking chrome os in the grub menu does nothing,just shows the background image(which grub uses).
Note: Installation was using done using already installed Debian system.
Tried both NTFS and ext4 file systems,didn't work

sebanc, 2020-05-14:
Hi,

How did you create the grub config ?

akashdur, 2020-05-16:
Hello,
I used grub-customizer to create new menu entry using the config appeared at the end of installation.
Thanks

sebanc, 2020-05-16:
Some distro appears to have a grub efi which is not compatible with brunch. In that case, the first thing to try is to replace your distro's grubx64.efi (usually located at /boot/efi/EFI/\<distro name>/) by the brunch grubx64.efi located in /efi/boot on partition 12.

akashdur, 2020-05-27:
that worked.
Thanks

closed 2020-05-27 by akashdur