sebanc/brunch #132: Cant change channel from stable

NO-ob, 2020-05-13:
The option to change the channel is greyed out for me and my account is owner, it says it's owner on the lockscreen and I have no other accounts on the machine. I've already tried reinstalling and it made no difference. I have also tried adding enable_updates to the args in the grub editor but that didn't let me enable it either

I used samus as the recovery image.

sebanc, 2020-05-14:
Hi,

"samus" recovery image has 2 known issues:
- flags can only be changed in guest mode
- the update channel has to be changed using command line (sudo update_engine_client --channel=dev-channel -update)

Regarding updates, do you use the dual boot procedure or is it single boot ? Could you run sudo edit-grub-config and take a picture of the screen ?

NO-ob, 2020-05-14:
Okay I set the update channel to dev channel via the method you just posted it gives a timeout error though so i'm not sure if it can find any updates, it also cant find any in the settings after changing to the dev channel.

Current version is  80.0.3987.162

grub.cfg is 

set timeout=2
menuentry "ChromeOS" {
regexxp --set disk "(^.+)(,gpt)" root
linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug enable_updates
initrd ($disk,7)/initramfs.img
}

Is it worth trying out rammus? I went with samus and my cpu is a second gen intel though

Also my system is single boot, I do have macOS on another ssd but grub doesn't see it

sebanc, 2020-05-14:
You nearly had it right, just replace "enable_updates" by "options=enable_updates"

NO-ob, 2020-05-14:
That worked thankyou for the help

closed 2020-05-18 by sebanc