sebanc/brunch #133: Suspend crashes, Matebook X Pro Intel i5-8250U, uhd620.

steinano, 2020-05-14:
Evertyhing on this laptop works, but after suspend the screen is still black.
I've tried the boot argument suspend_s3 it didnt make any difference.
Also, there have been quite a few other crashes.

I tried first the Brunch r81 k4.19 stable 20200510 and then tried the older Brunch r80 k4.19 stable 20200407 but no difference.
Rammus 80 image.

sebanc, 2020-05-14:
Hi,

Do you have a dual gpu ? If so try to add "module_blacklist=nouveau" to the kernel command line.

Else, could you please attach a dmesg (run "dmesg > ~/Downloads/dmesg.txt in crosh shell and attach the dmesg.txt file from your downloads folder).

steinano, 2020-05-14:
Mine doesn't have dual GPU.
Attached dmesg
[dmesg.txt](https://github.com/sebanc/brunch/files/4627892/dmesg.txt)

sebanc, 2020-05-14:
Could you update brunch to the latest version and confirm if you still have the suspend issue ?

steinano, 2020-05-14:
Yes I did start with the newest version , tried the older one just to see if it worked better. 
Do you need a dmesg with the latest brunch?

sebanc, 2020-05-14:
Thanks for the confirmation, the dmesg you have already provided should be enough for now, I will have a closer look at it.

sebanc, 2020-05-19:
Hi,

Just to say that I observed some weird behavior with the current chromiumos kernel source regarding networking and suspend for which I have not been able to clearly identify the origin of the issue.
I expect that this is caused by a bad commit which will be corrected in the next kernel release. I will update this issue once a release with a newer kernel is available.

steinano, 2020-05-20:
Thanks for the update! Now it has started crashing soon after boot for me, as soon as I start doing something in the browser it lags before the screen turns black (mouse pointer still visible). Not sure if this is connected with the crash on suspend, however the whole system is very unstable. Running off an ISO on ext4 SSD.

sebanc, 2020-05-20:
It would be helpful to have a dmesg after the computer starts crashing if you can get one.

steinano, 2020-05-20:
Here is var/log/messages, does that help.
When it crashed just after boot was just before all the ��������������������.
There was also other crashes maybe in the same log.
[messages.txt](https://github.com/sebanc/brunch/files/4655094/messages.txt)

sebanc, 2020-05-21:
Could you try the release here:
https://github.com/sebanc/brunch-testing/releases

steinano, 2020-05-22:
I tried it, it still didn't survive a suspend. The screen turned back on after about 1 minute, it seemed to work, but it didn't accept any input from keyboard or mouse (except could move mouse pointer). So I had to force shut off. Here is /var/log/messages the latest crash is towards the end
[messages.txt](https://github.com/sebanc/brunch/files/4665944/messages.txt)

sebanc, 2020-06-02:
Hi, I had good feedbacks about suspend on the latest "brunch-testing" repository release, could you try it ?

closed 2020-06-26 by sebanc