sebanc/brunch #135: MacBookAir5,2 no wifi

lindraden, 2020-05-14:
There doesn't seem to be any wifi at the setup/start. I've tried sammus and ramus firmwares just in case it was something to do with that. I've looked at some closed issues about the wifi but they are for lenovo laptops. Ctrl+Alt+T or Alt+F2 also doesn't work to bring up a terminal.

sebanc, 2020-05-14:
Hi, 

could you press "e" quickly when you see the grub screen and add "options=broadcom_wl" to the kernel command line ?

Unfortunately to have this setup automatically, you will need to use the terminal.

CTRL+ALT+T only works after signin in chromeos with your user
CTRL+ALT+F2 should normally bring up tty2 as soon as the boot screen appears

lindraden, 2020-05-14:
Yep I've tried that. I saw the broadcom.sh script get executed successfully

sebanc, 2020-05-14:
Have you been able to access the crosh shell (CTRL+ALT+T) or tty2 (CTRL+ALT+F2) ? Without that it will be difficult to try to debug it

lindraden, 2020-05-14:
I have tethered my phone via USB so I got past the setup. Still can't see any wifi connections though

Yes I have access to the crosh shell now

sebanc, 2020-05-14:
What is the output of "sudo rfkill list all" in crosh shell ?

lindraden, 2020-05-14:
hci0: Bluetooth
Soft and Hard Block: No
phy0: Wireless Lan
Soft and Hard Block: No
brcmwl-0: Wireless Lan
Soft and Hard Block: No

sebanc, 2020-05-14:
What's the output of "sudo rmmod brcmfmac" ? Do you see wifi networks after ?

lindraden, 2020-05-14:
sudo rmmod brcmfmac - ERROR brcmfmac is not currently loaded
sudo rmmod wl - exit 0, but took out my wifi completely even if I try to turn it on. I assume thats what that command does?

sebanc, 2020-05-14:
could you add "module_blacklist=wl" to the kernel command line ? (before or after "options=broadcom_wl")

lindraden, 2020-05-14:
Done. Restarted. Wifi doesn't turn on anymore. After running `sudo rmmod wl`

sebanc, 2020-05-14:
Could you remove boot without any additional option and attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach the dmesg.txt from your downloads folder).

lindraden, 2020-05-14:
[dmesg.txt](https://github.com/sebanc/brunch/files/4628123/dmesg.txt)
[lspci.txt](https://github.com/sebanc/brunch/files/4628124/lspci.txt)

Attached both files

sebanc, 2020-05-14:
Thanks ! You definitely need "options=broadcom_wl" but I don't really get why it's not working.

Could you boot again with "options=broadcom_wl" and provide a dmesg and the output of "lsmod" command ?

lindraden, 2020-05-14:
[dmesg_broadcom_wl.txt](https://github.com/sebanc/brunch/files/4628391/dmesg_broadcom_wl.txt)
[lsmod.txt](https://github.com/sebanc/brunch/files/4628392/lsmod.txt)

sebanc, 2020-05-14:
Sorry, i need a last dmesg with "options=broadcom_wl" and without module_blacklist=wl

lindraden, 2020-05-14:
[dmesg_broadcom_wl(1).txt](https://github.com/sebanc/brunch/files/4628477/dmesg_broadcom_wl.1.txt)

sebanc, 2020-05-14:
Could you post the outpit of "sudo lspci -n" ?

lindraden, 2020-05-15:
[lspci_new.txt](https://github.com/sebanc/brunch/files/4631608/lspci_new.txt)

thank you for your help!

sebanc, 2020-05-15:
Could you boot with "options=broadcom_wl", run "sudo modprobe lib80211 lib80211_crypt_tkip" and check if networks appears ?

lindraden, 2020-05-15:
No networks :(

sebanc, 2020-05-15:
I don't know macbooks very well, is the efi secureboot available in macbooks ? If so, is it enabled on your device ?

lindraden, 2020-05-15:
Hmm. I dont think so.. =\

sebanc, 2020-05-15:
The module is correctly loaded and should normally work, i don't why it is not working.

Could you verify that you wifi is not blocked by specific keys on the keyboard like airplane mode or something like that ?

lindraden, 2020-05-15:
Wifi is on. It's just when it scans it can't find any wifi access points anywhere. Normally when the module isn't working even if I turn it on it says its off. But it does say its on.
Even phone hotspot doesn't show up. It's very odd.

DJ8ones, 2020-07-08:
Was this one ever solved? I don't have a Macbook, but I do have a similar wifi card: 
Broadcom BCM4352 802.11ac Wireless Network Adapter (rev 03). 
[14e4:43b1]

I've gone through all the steps above and get the same results -- wifi appears to be on, but can't find any networks. I've attached relevant outputs. Let me know if I should try something else. Thanks!

[lspcin.txt](https://github.com/sebanc/brunch/files/4891255/lspcin.txt)
[lspcik.txt](https://github.com/sebanc/brunch/files/4891256/lspcik.txt)
[lsmod.txt](https://github.com/sebanc/brunch/files/4891257/lsmod.txt)
[lspci.txt](https://github.com/sebanc/brunch/files/4891258/lspci.txt)
[dmesg.txt](https://github.com/sebanc/brunch/files/4891259/dmesg.txt)

DJ8ones, 2020-07-08:
I realized my bluetooth can't connect either. Is it possible that the issue is airplane mode button for my model? This link describes exactly what I'm seeing. The only difference is that trying to press the airplane mode buttons does nothing for me.

[https://neverware.zendesk.com/hc/en-us/community/posts/360042264393-Wifi-not-Working-properly-](url)

lindraden, 2020-07-08:
No it was never solved. Kinda just moved on. I did try to install other OS's on my macbook air and it also had no wifi until I installed some additional drivers. This was PopOS!.

sebanc, 2020-07-09:
@lindraden could you post an lsmod and dmesg from PopOS ?

sebanc, 2020-07-11:
@DJ8ones I tried to add the ubuntu "broadcom_sta" driver instead of the one I am currently using in the latest testing release, could you give it a try ? (validate the github invite first)
https://github.com/sebanc/brunch-testing/releases

If it does not work it most likely means that your wifi adapter is not supported by the broadcom_wl driver and that linux is using the NdisWrapper package/driver which would be difficult to add to ChromeOS.

DJ8ones, 2020-07-11:
I will give it a shot, although I don’t have access to the machine for another week. Sorry!

liqili, 2022-07-25:
cool, I can confirm append options=broadcom_wl works for my macbook air 2014.