sebanc/brunch #136: A little help with broadcom wifi drivers

clandestinz, 2020-05-14:
As the title says. I couldn't get the broadcom wifi drivers working. I don't see make installed. No dkms. Ideas?

sebanc, 2020-05-14:
Hi, the broadcom_wl dkms driver is already included but you have to add "options=broadcom_wl" to the kernel command line to activate it. (run "sudo edit-grub-config" and add "options=broadcom_wl" after cros_debug)

clandestinz, 2020-05-14:
im dumb i forgot about the options=

RuhanN7, 2020-05-15:
added the code and rebooted.. still wifi not working.

sebanc, 2020-05-15:
@RuhanN7 what's your laptop model ?

RuhanN7, 2020-05-15:
hp 15q-ds0004tu. There is no Linux Beta support too.

sebanc, 2020-05-15:
For wifi, could you attach a dmesg to this issue (run "dmesg > ~/Downloads\dmesg.txt" and attach the dmesg.txt file from your downloads folder). Also, could you post the output of "sudo lspci" ?

For linux beta, verify that you are using rammus recovery image and that virtualisation or vt-x is enabled in the bios.

RuhanN7, 2020-05-15:
Where to run this (dmesg > ~/Downloads\dmesg.txt) command? in Chrome browser terminal, it's not working.

Output of "sudo lspci"
chronos@localhost / $ sudo lspci
00:00.0 Host bridge: Intel Corporation Device 31f0 (rev 03)
00:00.1 Signal processing controller: Intel Corporation Device 318c (rev 03)
00:00.3 System peripheral: Intel Corporation Device 3190 (rev 03)
00:02.0 VGA compatible controller: Intel Corporation Device 3184 (rev 03)
00:0e.0 Audio device: Intel Corporation Device 3198 (rev 03)
00:0f.0 Communication controller: Intel Corporation Device 319a (rev 03)
00:12.0 SATA controller: Intel Corporation Device 31e3 (rev 03)
00:14.0 PCI bridge: Intel Corporation Device 31d6 (rev f3)
00:14.1 PCI bridge: Intel Corporation Device 31d7 (rev f3)
00:15.0 USB controller: Intel Corporation Device 31a8 (rev 03)
00:1f.0 ISA bridge: Intel Corporation Device 31e8 (rev 03)
00:1f.1 SMBus: Intel Corporation Device 31d4 (rev 03)
01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
02:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device d723

RuhanN7, 2020-05-15:
I used rammus recovery in installation but Virtulisation was not enabled. I just enabled it and got Linux Beta option.. Thanks for that. Now Wifi and Bluetooth is the Problem.

sebanc, 2020-05-15:
Sorry the correct command is  "dmesg > ~/Downloads/dmesg.txt"

RuhanN7, 2020-05-15:
This the the dmesg code ouput
[dmesg.txt](https://github.com/sebanc/brunch/files/4634823/dmesg.txt)

sebanc, 2020-05-15:
@RuhanN7 your wifi module seems to be "rtl8188eus" which has no driver in the mainline linux kernel, I will try to include it and will update this issue when it is.

RuhanN7, 2020-05-15:
I will be greatful to you. If you solve the issue. I don't know the wifi module number. Probably you found it. 
Thank you in advance.

RuhanN7, 2020-05-15:
@sebanc can the "rtl8188eus" driver be installed through Terminal for now?

If possible, I got the Driver file.

Can you guide me by giving commands?

sebanc, 2020-05-16:
@RuhanN7 

Could you try the release in the "brunch-testing" repo (not this one) and boot with "options=rtl8188eu" added to the kernel command line ?

RuhanN7, 2020-05-16:
Didn't understand the brunch testing thing. But I can add the Line in Kernel Command and reboot. Can you elaborate?

RuhanN7, 2020-05-16:
@sebanc do I have install anything or just add the rtl command to kernel line and reboot?

sebanc, 2020-05-16:
Go back to my github account, instead of clicking on "brunch", look for "brunch-testing", download the release that is there and install it.
Before you boot the new release, add "options=rtl8188eu"

RuhanN7, 2020-05-16:
Did what you said. Wifi and Bluetooth still not working. Lost my Ethernet also.

sebanc, 2020-05-16:
Could you post a dmesg ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads folder to this issue)

sebanc, 2020-05-16:
Actually I forgot to include the firmware, could you download the below file:
https://github.com/lwfinger/rtl8188eu/raw/master/rtl8188eufw.bin

Then run "sudo cp ~/Downloads/rtl8188eufw.bin /lib/firmware/" and reboot.

If it still does not work, please attach the dmesg

sebanc, 2020-05-16:
Also, could you try booting with "options=rtl8188eu module_blacklist=r8188eu" added to the kernel command line ?

clandestinz, 2020-05-16:
love your support. keep it up

RuhanN7, 2020-05-16:
@sebanc should I reboot after installing firmware or should I enter the command after installing the firmware then reboot?

sebanc, 2020-05-16:
Enter the command before rebooting

RuhanN7, 2020-05-16:
@sebanc "sudo cp ~/Downloads/rtl8188eufw.bin /lib/firmware/rtlwifi/" this command didn't have any return. Is it meant to be that way?

It was like this..
chronos@localhost / $ sudo cp ~/Downloads/rtl8188eufw.bin /lib/firmware/rtlwifi/

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

chronos@localhost / $ sudo cp ~/Downloads/rtl8188eufw.bin /lib/firmware/rtlwifi/
chronos@localhost / $

RuhanN7, 2020-05-16:
@sebanc If you say it's ok. Then I will continue to add the command and reboot.

sebanc, 2020-05-16:
yes it is ok, you can reboot now

RuhanN7, 2020-05-16:
[dmesg.txt](https://github.com/sebanc/brunch/files/4638785/dmesg.txt)

Not working.. Ethernet too..

sebanc, 2020-05-17:
Sorry, just re-reading it and I think I mixed 2 different issues at some point, your wifi driver 
seems to be rtl8723de which I have also added at the same time, could you try to boot with "options=rtl8723de" ?

RuhanN7, 2020-05-17:
Should I delete last 2 commands?
Option enable and Blacklist?

RuhanN7, 2020-05-17:
> Also, could you try booting with "options=rtl8188eu module_blacklist=r8188eu" added to the kernel command line ?

Should I delete this two then reboot amd after that enter the new commands?

sebanc, 2020-05-17:
Yes please delete these 2

RuhanN7, 2020-05-17:
It worked. Wifi is online. Should I bother you with other issues? like Bluetooth?

RuhanN7, 2020-05-17:
Thanks a lot for the Wifi fix.

sebanc, 2020-05-17:
Could you send a new dmesg for bluetooth and explain the bootscreen and file transfer speed points ?

RuhanN7, 2020-05-17:
Earlier dmesg won't work?
Bluetooth not working at all.
Boot Screen shows enormous coding everytime I turn on the pc.
File transfer speed is too slow.

sebanc, 2020-05-17:
1) I need a new dmesg because the wifi driver was not loaded on previous one and wifi/bt are linked.
2) add "console=" to the kernel command line
3) File transfer speed only depends on the device you are using
4) add "options=enable_updates" to the kernel command line

RuhanN7, 2020-05-17:
added this "options=enable_updates" command after the rtl command to the kernel command line, it turned off wifi like before. Then I added "options=enable_updates" first then added "options=rtl8723de" . The wifi works now but not the update.

Wouldn't "console=" this command create the same problem that I just mentioned above?

RuhanN7, 2020-05-17:
[dmesg.txt](https://github.com/sebanc/brunch/files/4639834/dmesg.txt)
Here is the dmesg.

sebanc, 2020-05-17:
options have to be grouped together sperated by commas, in your case "options=rtl8723de,enable_updates". then you can a space and "console="

sebanc, 2020-05-17:
Sorry but I would need another dmesg taken after a fresh reboot

RuhanN7, 2020-05-17:
I will send you one. when it is done.

RuhanN7, 2020-05-17:
Updates and Boot Screen fixed. Thanks agiain.
[dmesg.txt](https://github.com/sebanc/brunch/files/4640093/dmesg.txt)
Dmesg for bluetooth.

RuhanN7, 2020-05-18:
This is the last major issue. Check if you can attend it please.

sebanc, 2020-05-20:
Could you try the release I have just uploaded in the "brunch-testing" repo (not this one) and attach another dmesg if bluetooth still does not work ?

RuhanN7, 2020-05-20:
Brunch r81 k4.19 testing 20200520
 This one?

sebanc, 2020-05-20:
Yes exactly

RuhanN7, 2020-05-20:
Internet is too slow because of Cyclone in my area. 
Will send the report to you , when it is done.

RuhanN7, 2020-05-20:
[dmesg.txt](https://github.com/sebanc/brunch/files/4658319/dmesg.txt)
Didn´t  Work.

sebanc, 2020-05-20:
Could you try to replace the content of /etc/init/rtl8723de.conf with:
```
start on stopped udev-trigger

script
	rmmod btrtl
	insmod /lib/modules/rtl8723de.ko
	modprobe btrtl
end script
```

And then reboot

RuhanN7, 2020-05-20:
Where to find the content and replace?

sebanc, 2020-05-20:
copy the text above
run "sudo nano /etc/init/rtl8723de.conf" in crosh shell
Delete the text in the file
Press CTRL+SHIFT+V to paste
then CTRL+X, "y" to save and enter

RuhanN7, 2020-05-21:
[dmesg.txt](https://github.com/sebanc/brunch/files/4660531/dmesg.txt)
Bluetooth still not working and The wifi stopped working after the command.

RuhanN7, 2020-05-22:
Can you give me the earlier script?

sebanc, 2020-05-22:
Here it is:
```
start on stopped udev-trigger

script
	insmod /lib/modules/rtl8723de.ko
end script
```

sebanc, 2020-05-22:
@RuhanN7 Could you try the new build there ?
https://github.com/sebanc/brunch-testing/releases

RuhanN7, 2020-05-22:
After installing the new build, Do I have edit any command?

sebanc, 2020-05-22:
No, no need to modify the script

RuhanN7, 2020-05-22:
[dmesg.txt](https://github.com/sebanc/brunch/files/4667751/dmesg.txt)
Didn´t work. Lost my Wifi again.

RuhanN7, 2020-05-22:
How to fix Wifi in this build?

[21 more replies]

closed 2020-06-02 by sebanc