sebanc/brunch #137: Chrome OS shows blank screen after rebooting

Aduser345, 2020-05-14:
I followed all the steps to create and install Chrome OS in HDD and even have dual boot setup for linux mint, windows and Chrome OS.
It booted up fine and I setup everything (Google account, etc) and then rebooted to disable developer mode and tried to boot it.
It boots but there is nothing except the mouse like if windows is running but explorer.exe isn't there
Please help to fix this issue as I d ont have much knowledge using shell

Aduser345, 2020-05-14:
Also why does it keep rebuilding rootfs

clandestinz, 2020-05-14:
I don't think it's a really good idea to dual boot chromeos. Try installing it on a different ssd or just go with one os

Aduser345, 2020-05-14:
Nevermind I reinstalled it and now it works okay

Aduser345, 2020-05-14:
Just that rebuilding rootfs step is painful

sebanc, 2020-05-15:
Hi,

Rebuilding rootfs happens when you do something which requires different patches to be applied on the main rootfs, notably disabling dev mode also sets selinux as enforcing in the rootfs and therefore requires the rootfs to be rebuilt.

More generally, rebuilding rootfs occurs:
- On first boot or after a brunch/chromeos update
- If you disable / enable dev_mode
- If you change an option
- if you modify partition 7

After you have fully setup your system, rebuilding rootfs should only occur after an update.

Unless, there are still opened points in this issue, I will close it.

Aduser345, 2020-05-15:
So it showing rebuilding rootfs while normal boot is completely normal?
Or it shouldn't show that?

sebanc, 2020-05-18:
Rebuilding rootfs is only normal in those 4 situations:
- On first boot or after a brunch/chromeos update
- If you disable / enable dev_mode
- If you change an option
- if you modify partition 7

Aduser345, 2020-05-18:
Well then it isn't normal cause it does it every boot
Nevertheless I lost my Chrome partition grub config 
Nice guide tho keep it up!!👍🏻

closed 2020-05-18 by Aduser345