sebanc/brunch #139: Lenovo L390 Yoga Stylus?

ldti, 2020-05-14:
Anyone managed to get the stylus menu working? The stylus itself seems to work fine...

sebanc, 2020-05-15:
Hi, which recovery image are you using ?

ldti, 2020-05-15:
I am using rammus.

sebanc, 2020-05-15:
Hi,

Indeed, I notice that the stylus menu is not present. I think it is a regression that i did not notice when I switched from octopus to rammus as base recovery image.

Could you try octopus recovery image and see if the stylus menu is present there ?

ldti, 2020-05-15:
I will.

ldti, 2020-05-15:
ok, that's odd. the recovery download page doesn't have any links, all of a sudden!

sebanc, 2020-05-15:
Indeed it seems not to be working for now

ldti, 2020-05-15:
What's your opinion on using Nami instead? Seems to be closer the the Lenovo yoga chrome book, hardware wise..

sebanc, 2020-05-15:
Nami should work too I think, it's just that i never tried with it but with octopus, also I have not yet implemented updates on nami.

ldti, 2020-05-15:
OK. I'll go with octopus then.
As soon as the site is back up..

ldti, 2020-05-15:
Well, tried it. Same result as previously. Stylus works, but no stylus menu.

sebanc, 2020-05-15:
Thanks for trying. That's weird, it worked at some point, I will try to see what's wrong with it.

ldti, 2020-05-15:
LMK if i can help in some way.

ldti, 2020-05-15:
I am getting the stylus message, but not the button. See picture. 
[image]

This was taken on Nami

sebanc, 2020-05-16:
Could you add the line "--has-internal-stylus" to /etc/chrome_dev.conf ?

ldti, 2020-05-16:
Addded. Didn't help :(

sebanc, 2020-05-17:
Hi, just added "--has-internal-stylus" to /etc/chrome_dev.conf on rammus image and it appeared:

[image]

ldti, 2020-05-17:
welp, unless i am doing something very wrong...
[image]

sebanc, 2020-05-17:
Is the stylus toolbox enabled in settings ?

ldti, 2020-05-17:
I don't have anything in settings mentioning stylus...
[image]

sebanc, 2020-05-17:
I have a stylus menu in devices but i am still on r80 ? Which version are you on ?

ldti, 2020-05-17:
Updated to r81...

sebanc, 2020-05-17:
Updated to r81, I still have the stylus menu in settings:
[image]

Another weird thing happens, despite the "--has-internal-stylus" being from removed from /etc/chrome_dev.conf, I still have the stylus toolbox so it seems like the stylus setting is synced somewhere.

ldti, 2020-05-17:
Is it possible thst it's dependent on the hardware?

Also, I could try to delete backups, if I can find them..

[Screenshot 2020-05-17 at 11 42 47 AM]<https://user-images.githubusercontent.com/22224731/82141062-a0fe5d00-9833-11ea-9321-27142464726a.png>

sebanc, 2020-05-17:
Maybe you can add "--has-internal-stylus" to /etc/chrome_dev.conf, delete google sync and reboot:
https://support.google.com/chromebook/answer/2914794?hl=en

ldti, 2020-05-17:
I just deployed a clean rammus image on a usb drive. added has internal stylus and rebooted. still the same effect exactly.

What hardware are you testing on?

sebanc, 2020-05-17:
On a Surface Go. I will keep thinking about this issue but I am not sure what it could be related to

ldti, 2020-05-17:
Different stylus hardware?

sebanc, 2020-05-17:
Different touchscreen hardware maybe, with "--has-internal-stylus" I don't even have to use a stylus, the icon just appeared.

I did not try bringing a stylus as mine has no more battery, maybe the icon would have appeared directly.

ldti, 2020-05-17:
Could be, indeed. The stylus does work here, but no icon...

ldti, 2020-05-17:
Here's something interesting:
chrome://version does not show the --has-internal-stylus in the command line!
Can you check on your side?

ldti, 2020-05-17:
It's even worse! the chrome_dev.conf seems to be overwritten after a restart!

ldti, 2020-05-17:
ok, so after keeping it in debug mode, the setting is there. still no stylus menu...

sebanc, 2020-05-17:
What's mostly weird is that you don't even have the stylus menu in settings, i think it must be a pre-requisite for the stylus toolbox but i have currently no idea why it is missing.

ldti, 2020-05-17:
very odd...

sebanc, 2020-05-17:
Apparently there was a flag "force-enable-stylus-tools" at some point, not sure if it is still there (i cannot check i am under linux right now):
https://www.reddit.com/r/chromeos/comments/602y2j/stylus_menu/

ldti, 2020-05-17:
it's nt , unfortunately.

https://www.reddit.com/r/chromeos/comments/602y2j/stylus_menu/

ldti, 2020-05-17:
What is the touch hardware in the surface go, in DMESG?

sebanc, 2020-05-17:
```
[    4.051301] input: ELAN9038:00 04F3:261A as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-1/i2c-ELAN9038:00/0018:04F3:261A.0005/input/input8
[    4.051463] input: ELAN9038:00 04F3:261A Pen as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-1/i2c-ELAN9038:00/0018:04F3:261A.0005/input/input11
```

Indeed my touchscreen expose a "Pen" device, does yours ?

ldti, 2020-05-17:
```
[    1.590866] usb 1-6: Product: Pen and multitouch sensor
[    1.595247] input: Wacom Co.,Ltd. Pen and multitouch sensor Touchscreen as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:5159.0001/input/input6
[    1.596097] input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:5159.0001/input/input7
[    1.596905] hid-generic 0003:056A:5159.0001: input,hiddev0,hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-6/input0
[    1.599548] input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:5159.0002/input/input8
[    1.600366] input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:5159.0002/input/input9
[    1.601149] input: Wacom Co.,Ltd. Pen and multitouch sensor Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:5159.0002/input/input10
[    1.601885] hid-generic 0003:056A:5159.0002: input,hiddev1,hidraw1: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-6/input1
[   18.943528] input: Wacom Pen and multitouch sensor Finger as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:5159.0001/input/input14
[   18.943682] wacom 0003:056A:5159.0001: hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-6/input0
[   19.005475] input: Wacom Pen and multitouch sensor Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:5159.0002/input/input18
[   19.007547] wacom 0003:056A:5159.0002: hidraw1: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-6/input1

sebanc, 2020-05-17:
That's a lot of input devices but input18 is correctly recognized as "Pen".

ldti, 2020-05-17:
yep...

sebanc, 2020-05-17:
Can you post the output of "udevadm info /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:5159.0002/input/input18" ?

ldti, 2020-05-17:
```
P: /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:5159.0002/input/input18
E: ABS=1000d000003
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:5159.0002/input/input18
E: EV=1b
E: ID_BUS=usb
E: ID_INPUT=1
E: ID_INPUT_TABLET=1
E: ID_MODEL=Pen_and_multitouch_sensor
E: ID_MODEL_ENC=Pen\x20and\x20multitouch\x20sensor
E: ID_MODEL_ID=5159
E: ID_PATH=pci-0000:00:14.0-usb-0:6:1.1
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_6_1_1
E: ID_REVISION=0002
E: ID_SERIAL=Wacom_Co._Ltd._Pen_and_multitouch_sensor
E: ID_TYPE=hid
E: ID_USB_DRIVER=usbhid
E: ID_USB_INTERFACES=:030000:
E: ID_USB_INTERFACE_NUM=01
E: ID_VENDOR=Wacom_Co._Ltd.
E: ID_VENDOR_ENC=Wacom\x20Co.\x2cLtd.
E: ID_VENDOR_ID=056a
E: KEY=1e03 0 0 0 0 0
E: MODALIAS=input:b0003v056Ap5159e0111-e0,1,3,4,k140,141,149,14A,14B,14C,ra0,1,18,1A,1B,28,m0,lsfw
E: MSC=1
E: NAME="Wacom Pen and multitouch sensor Pen"
E: PHYS="usb-0000:00:14.0-6/input1"
E: POWERD_ROLE=external_input
E: POWERD_TAGS=usable_when_docked usable_when_laptop usable_when_tablet usable_when_display_off wakeup wakeup_only_when_usable inhibit role_external_input
E: POWERD_TAGS_INHIBIT=inhibit
E: POWERD_TAGS_USABLE=usable_when_docked usable_when_laptop usable_when_tablet usable_when_display_off
E: POWERD_TAGS_WAKEUP=wakeup wakeup_only_when_usable
E: PRODUCT=3/56a/5159/111
E: PROP=2
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: UDISKS_PRESENTATION_HIDE=0
E: UNIQ=""
E: USEC_INITIALIZED=19009848

sebanc, 2020-05-17:
```
P: /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-ELAN9038:00/0018:04F3:261A.0005/input/input19
E: ABS=1000d000003
E: DEVPATH=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-ELAN9038:00/0018:04F3:261A.0005/input/input19
E: EV=1b
E: ID_INPUT=1
E: ID_INPUT_TABLET=1
E: ID_PATH=pci-0000:00:15.1-platform-i2c_designware.1
E: ID_PATH_TAG=pci-0000_00_15_1-platform-i2c_designware_1
E: ID_SERIAL=noserial
E: KEY=c03 0 0 0 0 0
E: MODALIAS=input:b0018v04F3p261Ae0100-e0,1,3,4,k140,141,14A,14B,ra0,1,18,1A,1B,28,m4,lsfw
E: MSC=10
E: NAME="ELAN9038:00 04F3:261A Pen"
E: PHYS="i2c-ELAN9038:00"
E: POWERD_ROLE=internal_stylus
E: POWERD_TAGS=usable_when_laptop usable_when_tablet wakeup wakeup_only_when_usable inhibit role_internal_stylus
E: POWERD_TAGS_INHIBIT=inhibit
E: POWERD_TAGS_USABLE=usable_when_laptop usable_when_tablet
E: POWERD_TAGS_WAKEUP=wakeup wakeup_only_when_usable
E: PRODUCT=18/4f3/261a/100
E: PROP=2
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: UNIQ=""
E: USEC_INITIALIZED=88842671
```

ldti, 2020-05-17:
``` E: POWERD_ROLE=external_input```
``` E: POWERD_ROLE=internal_stylus ```

hmmm...

I wonder if something can be done about that?

sebanc, 2020-05-17:
You can try: (and verify if it worked after a reboot, i am not sure it will)
`echo 'ENV{ID_INPUT_TABLET}=="1", ENV{POWERD_ROLE}="internal_stylus"' | sudo tee /lib/udev/rules.d/99-test.rules`

ldti, 2020-05-17:
dmesg | grep Pen:
```[    1.599773] usb 1-6: Product: Pen and multitouch sensor
[    1.603905] input: Wacom Co.,Ltd. Pen and multitouch sensor Touchscreen as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:5159.0001/input/input6
[    1.604647] input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:5159.0001/input/input7
[    1.605426] hid-generic 0003:056A:5159.0001: input,hiddev0,hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-6/input0
[    1.607846] input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:5159.0002/input/input8
[    1.608622] input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:5159.0002/input/input9
[    1.609396] input: Wacom Co.,Ltd. Pen and multitouch sensor Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:5159.0002/input/input10
[    1.610048] hid-generic 0003:056A:5159.0002: input,hiddev1,hidraw1: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-6/input1
[   18.950784] input: Wacom Pen and multitouch sensor Finger as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:5159.0001/input/input13
[   18.951071] wacom 0003:056A:5159.0001: hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-6/input0
[   19.012610] input: Wacom Pen and multitouch sensor Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:5159.0002/input/input16
[   19.013915] wacom 0003:056A:5159.0002: hidraw1: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-6/input1
```

udevadm info /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:5159.0001/input/input13 :

``` P: /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:5159.0001/input/input13
E: ABS=260800000000003
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:5159.0001/input/input13
E: EV=b
E: ID_BUS=usb
E: ID_INPUT=1
E: ID_INPUT_TOUCHSCREEN=1
E: ID_MODEL=Pen_and_multitouch_sensor
E: ID_MODEL_ENC=Pen\x20and\x20multitouch\x20sensor
E: ID_MODEL_ID=5159
E: ID_PATH=pci-0000:00:14.0-usb-0:6:1.0
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_6_1_0
E: ID_REVISION=0002
E: ID_SERIAL=Wacom_Co._Ltd._Pen_and_multitouch_sensor
E: ID_TYPE=hid
E: ID_USB_DRIVER=usbhid
E: ID_USB_INTERFACES=:030000:
E: ID_USB_INTERFACE_NUM=00
E: ID_VENDOR=Wacom_Co._Ltd.
E: ID_VENDOR_ENC=Wacom\x20Co.\x2cLtd.
E: ID_VENDOR_ID=056a
E: KEY=400 0 0 0 0 0
E: MODALIAS=input:b0003v056Ap5159e0111-e0,1,3,k14A,ra0,1,2F,35,36,39,mlsfw
E: NAME="Wacom Pen and multitouch sensor Finger"
E: PHYS="usb-0000:00:14.0-6/input0"
E: POWERD_ROLE=external_input
E: POWERD_TAGS=usable_when_docked usable_when_laptop usable_when_tablet usable_when_display_off wakeup wakeup_only_when_usable inhibit role_external_input
E: POWERD_TAGS_INHIBIT=inhibit
E: POWERD_TAGS_USABLE=usable_when_docked usable_when_laptop usable_when_tablet usable_when_display_off
E: POWERD_TAGS_WAKEUP=wakeup wakeup_only_when_usable
E: PRODUCT=3/56a/5159/111
E: PROP=2
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: UDISKS_PRESENTATION_HIDE=0
E: UNIQ=""
E: USEC_INITIALIZED=19014922
```

udevadm info /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:5159.0002/input/input16:
```P: /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:5159.0002/input/input16
E: ABS=1000d000003
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:056A:5159.0002/input/input16
E: EV=1b
E: ID_BUS=usb
E: ID_INPUT=1
E: ID_INPUT_TABLET=1
E: ID_MODEL=Pen_and_multitouch_sensor
E: ID_MODEL_ENC=Pen\x20and\x20multitouch\x20sensor
E: ID_MODEL_ID=5159
E: ID_PATH=pci-0000:00:14.0-usb-0:6:1.1
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_6_1_1
E: ID_REVISION=0002
E: ID_SERIAL=Wacom_Co._Ltd._Pen_and_multitouch_sensor
E: ID_TYPE=hid
E: ID_USB_DRIVER=usbhid
E: ID_USB_INTERFACES=:030000:
E: ID_USB_INTERFACE_NUM=01
E: ID_VENDOR=Wacom_Co._Ltd.
E: ID_VENDOR_ENC=Wacom\x20Co.\x2cLtd.
E: ID_VENDOR_ID=056a
E: KEY=1e03 0 0 0 0 0
E: MODALIAS=input:b0003v056Ap5159e0111-e0,1,3,4,k140,141,149,14A,14B,14C,ra0,1,18,1A,1B,28,m0,lsfw
E: MSC=1
E: NAME="Wacom Pen and multitouch sensor Pen"
E: PHYS="usb-0000:00:14.0-6/input1"
E: POWERD_ROLE=internal_stylus
E: POWERD_TAGS=usable_when_docked usable_when_laptop usable_when_tablet usable_when_display_off wakeup wakeup_only_when_usable inhibit role_external_input
E: POWERD_TAGS_INHIBIT=inhibit
E: POWERD_TAGS_USABLE=usable_when_docked usable_when_laptop usable_when_tablet usable_when_display_off
E: POWERD_TAGS_WAKEUP=wakeup wakeup_only_when_usable
E: PRODUCT=3/56a/5159/111
E: PROP=2
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: UDISKS_PRESENTATION_HIDE=0
E: UNIQ=""
E: USEC_INITIALIZED=19017707
```

So we need to change the other one too, somehow...

sebanc, 2020-05-17:
I would say it should be enough, there is still no stylus in settings menu ?

ldti, 2020-05-17:
Unfortunately not...

sebanc, 2020-05-17:
I would say that's probably not related then... I will try to think of something else

ldti, 2020-05-17:
got it..

sebanc, 2020-05-20:
Well, I have no other idea for now, so let's try your idea, could you run the below commands, reboot and see if something changed:
```
echo 'ENV{ID_INPUT_TABLET}=="1", ENV{POWERD_ROLE}="internal_stylus"' | sudo tee /lib/udev/rules.d/99-test.rules
echo 'ENV{ID_INPUT_TOUCHSCREEN}=="1", ENV{POWERD_ROLE}="internal_touchscreen"' | sudo tee -a /lib/udev/rules.d/99-test.rules
```

ldti, 2020-05-20:
Unfortunately, no change. though the devices in dmesg do now identify as internal stylus and touchscreen.