sebanc/brunch #143: EXT4-fs not mounting with journaling

Inotamira, 2020-05-15:
I don't think this is related to my Coreboot Grub payload at all, as the issue I'd mentioned before I understood the problem, was me not setting the correct root partition. After doing so I can use the grub.cfg on partition 12 straight away and the system boots no problem. I bring this up for clarification for sebanc should he come to assist me, to simplify, Coreboot is not the problem I am 100% certain of it at this point.

So, my system is a Lenovo Thinkpad T400, 6GB of ram, an Intel Core 2 Duo P8400, with the Intel Corporation Mobile 4 Series Chipset Integrated Graphics Controller. The system is installed using the Samus as I'm well older than 3rd generation through Linux Mint as most guides recommend. Everything goes smoothly until the system starts trying to set itself up.

Once it boots, it starts expanding the FS, and then everything starts to go wrong, it tries mkdir a good several dozen times, and fails repeatedly, eventually says that it can't mount the EXT4-FS on sda1 with journaling, and failed into self-repair mode forever. I don't think the hardware is directly a stopping point at the moment, as it seems to be an issue with it managing the file systems at the moment for whatever reason.

sebanc, 2020-05-15:
The issue with filesystems not correctly mounted/recognized can be caused by many underlying issues. Could you attach as zip in this issue a a short video of the boot process ?

Inotamira, 2020-05-15:
> The issue with filesystems not correctly mounted/recognized can be caused by many underlying issues. Could you attach as zip in this issue a a short video of the boot process ?

You want that in slowmotion so you can catch everything :P ? Gimme like 5 minutes.

Inotamira, 2020-05-15:
https://drive.google.com/open?id=10xG2zd9ggPe8hxNcRQ3wIQoc_lpTvS-9
Hopefully a drive link is fine, videos too big for github.

sebanc, 2020-05-15:
Thanks for the video, the messages that are displayed means that the os has been build with cpu instructions which are not supported by your cpu.

Unfortunately, all the devices that were compatible with your cpu are now AUE.

I was not aware that certain devices were not compatible with samus, I will add this information on the main page.

Inotamira, 2020-05-15:
> AUE

? I cannot for the life of me figure out what that's supposed to unabreviate to, but that's genuinely disappointing. Thanks.

sebanc, 2020-05-15:
sorry it is a chromeos term to say that the device is considered end of life and not updated anymore by google.

closed 2020-05-18 by sebanc