sebanc/brunch #144: Trackpad movement not working on asus E200HA laptop

swarshall, 2020-05-15:
Hi guys, brunch works great on my laptop, except for the trackpad.

- Left click works on the two buttons (left and right) and on the surface.

- Right click walks on the surface with two fingers

- But no pointer movement possible

Thanks  for your time and sorry for my English.

[dmesg.txt](https://github.com/sebanc/brunch/files/4634703/dmesg.txt)

sebanc, 2020-05-15:
Could you post the output of "udevadm info /devices/pci0000:00/808622C1:02/i2c-2/i2c-FTE1000:00/0018:0B05:0101.0003/input/input12" ?

swarshall, 2020-05-15:
chronos@localhost / $ udevadm info /dev/input/mouse1
P: /devices/pci0000:00/808622C1:02/i2c-2/i2c-FTE1000:00/0018:0B05:0101.0003/input/input12/mouse1
N: input/mouse1
S: input/by-path/platform-808622C1:02-mouse
E: DEVLINKS=/dev/input/by-path/platform-808622C1:02-mouse
E: DEVNAME=/dev/input/mouse1
E: DEVPATH=/devices/pci0000:00/808622C1:02/i2c-2/i2c-FTE1000:00/0018:0B05:0101.0003/input/input12/mouse1
E: ID_INPUT=1
E: ID_INPUT_TOUCHPAD=1
E: ID_PATH=platform-808622C1:02
E: ID_PATH_TAG=platform-808622C1_02
E: ID_SERIAL=noserial
E: MAJOR=13
E: MINOR=33
E: POWERD_ROLE=internal_touchpad
E: POWERD_TAGS=usable_when_laptop usable_when_display_off role_internal_touchpad
E: POWERD_TAGS_USABLE=usable_when_laptop usable_when_display_off
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: USEC_INITIALIZED=39701393

sebanc, 2020-05-15:
Could you try pressig FN+F7 ?
source: https://askubuntu.com/questions/1045102/touchpad-recognized-but-cursor-not-moving

sebanc, 2020-05-15:
Sorry it is not at all the same model but could you try to see if there is a key combination that could block the touchpad.

sebanc, 2020-05-15:
Also, could you try booting with "module_blacklist=asus_wmi" added to the kernel command line (after cros_debug)

swarshall, 2020-05-15:
I check but it's not working (module_blacklist and key combination : FN+F7 and FN+F9)

sebanc, 2020-05-15:
Could you replace the first block in /etc/gesture/40-touchpad-cmt.conf with:
```
Section "InputClass"
    Identifier      "touchpad"
    MatchIsTouchpad "on"
    MatchDevicePath "/dev/input/event*"
    Driver          "cmt"
    Option          "Tap Minimum Pressure" "0"
    Option          "AccelerationProfile" "-1"
    Option          "Scroll Buttons" "0"
    Option          "Scroll Axes" "1"
    Option          "Tap Minimum Pressure" "10.0"

    # CMT devices potentially process keyboard events
    Option          "XkbModel" "pc"
    Option          "XkbLayout" "us"
EndSection
```

swarshall, 2020-05-15:
No, it does not work, my trackpad is surely an elantech, there may be something to do with these models

sebanc, 2020-05-15:
Could you try this one also when you have the time ?

Section "InputClass"
    Identifier      "touchpad"
    MatchIsTouchpad "on"
    MatchDevicePath "/dev/input/event*"
    Driver          "cmt"
    Option          "Tap Minimum Pressure" "0"
    Option          "AccelerationProfile" "-1"
    Option          "Scroll Buttons" "0"
    Option          "Scroll Axes" "1"
    Option          "SendCoreEvents" "On"

    # CMT devices potentially process keyboard events
    Option          "XkbModel" "pc"
    Option          "XkbLayout" "us"
EndSection

sebanc, 2020-05-15:
Could you also try adding "psmouse.elantech_smbus=0" to the kernel command line ?

swarshall, 2020-05-16:
Really weird this problem nothing makes the trackpad work. Do you have an idea ?

sebanc, 2020-05-17:
Someone ported the cmt driver to linux and added its config files:
https://github.com/hugegreenbug/xf86-input-cmt/tree/master/xorg-conf

Could you try some of the configs that are there, for example for 50-touchpad-cmt-elan.conf:
```
# Configure touchpads to use Chromium Multitouch (cmt) X input driver
Section "InputClass"
    Identifier      "touchpad elan"
    MatchIsTouchpad "on"
    MatchDevicePath "/dev/input/event*"
    Option          "Touchpad Stack Version" "1"
    Option          "Horizontal Resolution" "33"
    Option          "Vertical Resolution" "33"
    Option          "Integrated Touchpad" "1"
    Option          "Input Queue Delay" "0.017"
    Option          "Two Finger Vertical Close Distance Thresh" "35.0"
    Option          "Fling Buffer Suppress Zero Length Scrolls" "0"
EndSection
```
Take the "Option" lines:
```
    Option          "Touchpad Stack Version" "1"
    Option          "Horizontal Resolution" "33"
    Option          "Vertical Resolution" "33"
    Option          "Integrated Touchpad" "1"
    Option          "Input Queue Delay" "0.017"
    Option          "Two Finger Vertical Close Distance Thresh" "35.0"
    Option          "Fling Buffer Suppress Zero Length Scrolls" "0"
And add them to the main config in the first section of /etc/gesture/40-touchpad-cmt.conf.
```
Important: do not replace the full section, just add the options you want to try as the config misses some chromeos features

sebanc, 2020-05-21:
Forgot to ask, which recovery image did you use ?

swarshall, 2020-05-24:
Hi I hope you are well. I finally had time to try and I must tell you that it works very well!
Small problem all the same the physical right button of the trackpad does not work. Any idea for this?

Otherwise thank you for your help and for brunch :) Can we make a donation on github?

sebanc, 2020-05-25:
Hi,

Thanks for the donation offer, I really appreciate the gesture :) I am not rich but i would prefer that my work would serve charity/hospitals, if you can make a small donation to one that would be perfect.

Regarding touchpad physical right click, it is disabled when gestures are enabled and replaced by two fingers tap so I would say it's normal.

Which config did you use to make it work ? Was it exactly adding the below lines ?

```
    Option          "Touchpad Stack Version" "1"
    Option          "Horizontal Resolution" "33"
    Option          "Vertical Resolution" "33"
    Option          "Integrated Touchpad" "1"
    Option          "Input Queue Delay" "0.017"
    Option          "Two Finger Vertical Close Distance Thresh" "35.0"
    Option          "Fling Buffer Suppress Zero Length Scrolls" "0"
```

closed 2020-06-02 by sebanc

rockmanvnx6, 2020-11-10:
Hi, @sebanc, sorry for digging this up. I have the exact problem on Brunch 80 with the same model, `alt_touchpad_config1` and `alt_touchpad_config2 `does not work either. 

I just wonder when you were referring to the 
```
    Option          "Touchpad Stack Version" "1"
    Option          "Horizontal Resolution" "33"
    Option          "Vertical Resolution" "33"
    Option          "Integrated Touchpad" "1"
    Option          "Input Queue Delay" "0.017"
    Option          "Two Finger Vertical Close Distance Thresh" "35.0"
    Option          "Fling Buffer Suppress Zero Length Scrolls" "0"
```

Do I boot into chromeos and modify the `/etc/gesture/40-touchpad-cmt.conf`?