sebanc/brunch #145: HP Envy x360 AMD Status (Image: grunt r80)

PranavSakthivel, 2020-05-15:
Hi, I understand that AMD hardware has much lower chances of working, but I wonder if it is possible to troubleshoot. The laptop is a HP 15m-cp0011dx:
- AMD Ryzen 2500U CPU
- AMD Vega 8 iGPU
- 16GB DDR4

Installation process seemed to go fine (16GB USB flash drive), and it rebuilt the rootfs with no issues. When it tries to boot Chromeos after that step, the system black screens and reboots immediately to GRUB for ChromeOS, and this repeats indefinitely. Was wondering if there are any boot options I can try, or a different image. I had no luck with nomodeset.

Thanks.

sebanc, 2020-05-15:
Hi,

Sure we can try to debug it. First, could you take a short video of the boot process (from grub to crash without rootfs rebuilding) ?

PranavSakthivel, 2020-05-15:
Hi, here is the video of the full boot. Sorry about the glare. Thanks.

https://youtu.be/K8I0owIo2fE

sebanc, 2020-05-15:
Hi, thanks for the video it's perfect, unfortunately we can see "trap: invalid opcode" error messages which is characteristic of cpu instructions included in the build which are not compatible with your cpu. I don't really know which one as I did not look through it for AMD but there is no chance your device will work with this image.

PranavSakthivel, 2020-05-15:
Hi, that is unfortunate, thank you for the response.

sebanc, 2020-05-16:
The rumor has it that AMD Ryzen chromebooks are on their way:
https://www.reddit.com/r/chromeos/comments/ghypby/just_found_this_on_amds_websiteryzen_powered/

The related recovery image when it will be out might give better results.

sebanc, 2020-05-17:
Hi,

I attempted something related to cpu instructions, could you try the release I uploaded on the "brunch-testing" repository (not this one) ?

PranavSakthivel, 2020-05-17:
Hi, thank you very much for the build. I gave it a try by making the chromeos img from WSL. Rootfs seemed to build fine, but I believe the same issue happened again. I used the same Grunt 80 image. Here is the video of the boot: 

https://www.youtube.com/watch?v=7lS0qJUuI5A

Thanks.

sebanc, 2020-05-17:
Thanks for trying, indeed it is the same issue... Could you just confirm the name of the archive you downloaded "brunch_r81_k4.19_testing_20200517.tar.gz" or "brunch_r81_k4.19_testing_20200517_v2.tar.gz" ? (I updated the release at about the same time you tested it)

PranavSakthivel, 2020-05-17:
The name of the file I tried is **brunch_r81_k4.19_testing_20200517.tar.gz**. I'll go ahead and give the v2 a try then, thanks.

PranavSakthivel, 2020-05-17:
Just tested the V2, seems to have the same behavior unfortunately.

sebanc, 2020-05-17:
Indeed, globally both attempts have not managed to fix the cpu instructions issue. Hope the rizen device comes out soon with a more compatible AMD recovery image.

PranavSakthivel, 2020-05-17:
Seems like it is a typo or rumor so far, but fingers crossed. Thanks for all the help.

closed 2020-06-02 by sebanc

xeu100, 2020-12-19:
#722