sebanc/brunch #146: Fingerprint sensor support ?

skandavc18, 2020-05-16:
Everything works fine in my Lenovo Yoga 730. 
Wanted to know if support for fingerprint sensor can be added. There is support for fingerprint sensor in Ubuntu.

Would be helpful if you could tell me the steps to add those Ubuntu drivers to Chrome OS.

ldti, 2020-05-17:
ChromeOS doesn't support it yet, AFAIK..

sebanc, 2020-05-17:
Hi,

it's unlikely that it will happen, it would require a motivated dev with the same hardware to make the linux fingerprint sensor driver output data the same way as the chromeos one (which is rather complicated).

skandavc18, 2020-05-17:
> ChromeOS doesn't support it yet, AFAIK..

It does. There are many Chromebooks which support fingerprint sensor.
example : https://www.androidpolice.com/2020/05/02/asus-chromebook-flip-c436-review-premium-hardware-poor-value/

skandavc18, 2020-05-17:
> Hi,
> 
> it's unlikely that it will happen, it would require a motivated dev with the same hardware to make the linux fingerprint sensor driver output data the same way as the ChromeOS one (which is rather complicated).

Ok. Anyways thanks for this wonderful project.

closed 2020-05-20 by skandavc18