sebanc/brunch #149: System failed to detect ChromeOS install on USB and launches an init shell instead

khanhtranngoccva, 2020-05-16:


sebanc, 2020-05-16:
Hi,

Could you post the text that appears when you run "chromeos-install.sh" to create the usb ?

Also, what size is your usb flashdrive ?

khanhtranngoccva, 2020-05-16:
[image]

khanhtranngoccva, 2020-05-16:
There is one device /dev/sdb as well.

sebanc, 2020-05-16:
Ok got it, brunch comes with a timeout of 10 seconds to make sure all devices are loaded (usually all devices are loaded in 1/2 seconds). Your usb device took 25 seconds to be loaded which is not normal and is the origin of this issue.

Could you try to plug it on another usb port or looking for suspicious settings in your bios ?

You can also try another usb flashdrive.

khanhtranngoccva, 2020-05-16:
How do I set its timeout?

sebanc, 2020-05-16:
The timeout is in the code you cannot change it

khanhtranngoccva, 2020-05-16:
Where is the code of the timeout please, and is it readable?

sebanc, 2020-05-16:
It's in the initramfs, if you want to modify it, you have to unpack it and repack it, i will not provide support for that.

khanhtranngoccva, 2020-05-16:
OK. I have got another issue too, would you mind answering?

sebanc, 2020-05-16:
No what is it ?

khanhtranngoccva, 2020-05-16:
https://github.com/sebanc/brunch/issues/150

closed 2020-05-16 by sebanc

khanhtranngoccva, 2020-05-16:
How to unpack/repack initramfs without kernel panic, please?