sebanc/brunch #150: ChromeOS 78+ failed to run ARC across reboots.

khanhtranngoccva, 2020-05-16:
This error is caused by signal 15, signal 9 or EOF bugs. I do not know how to fix it.

khanhtranngoccva, 2020-05-16:
[image]

sebanc, 2020-05-16:
If android is able to setup, it means that all is working correctly on kernel side. If arc refuses to launch in subsequent reboots, it means that there is something in your google profile / apps you have downloaded... which breaks the chromeos android container.

My only suggestions are:

- Deleting google sync:
https://support.google.com/chromebook/answer/2914794?hl=en

- Clearing cache & data from Google Play Services:
https://support.google.com/googleplay/answer/7512202?hl=en

closed 2020-05-16 by sebanc