sebanc/brunch #151: Installing rammus/brunch on different notebooks & same google account means everywhere the same app?

jordanvanbergen, 2020-05-16:
I have installed rammus / brunch / chrome OS on 3 different notebooks and trieds the MSI i9 Prestige P75 Creator 9SF as well and it is all working except for the webcam, see: https://imgur.com/G89IjbM 

One question. Due to booting rammus / brunch with same google account am I correct that Chrome OS sees all 3 different notebook as 1 pieace of hardware? 

If I install an app from the play store and reboot into another machine it will be installed automatically so I can't have different apps on different hardware. 

Is this standard / normal ?

khanhtranngoccva, 2020-05-16:
Uninstalled apps will not be synced.

jordanvanbergen, 2020-05-16:
> Uninstalled apps will not be synced.

I know but the question is: are all installations seen as 1 hardware device instead of 3 different ones?

jordanvanbergen, 2020-05-16:
I was able to get the webcam running by the way

[    2.116079] uvcvideo: Found UVC 1.00 device HD Webcam (5986:211c)
[    2.138367] input: HD Webcam: HD Webcam as /devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0/input/input10

It's working with the camera app. So even a live USB Chrome OS bootable device is nice this way on this bad ass machine.

sebanc, 2020-05-16:
@jordanvanbergen Indeed historically I made the choice to have only one device name because it was easier and never looked back at it.

I can probably fix this but it's quite a core part of brunch, so I will take some time to think about it and make sure that i am not breaking things.

sebanc, 2020-05-16:
Also, how did you fix the camera ?

jordanvanbergen, 2020-05-16:
> @jordanvanbergen Indeed historically I made the choice to have only one device name because it was easier and never looked back at it.
> 
> I can probably fix this but it's quite a core part of brunch, so I will take some time to think about it and make sure that i am not breaking things.

I think you could leave it like it is. But if you have (like me) 100 apps installed. A new device will first take 1+ hour to be up2date as it first will download/install all android applications. 

So I think just answer that it's like what I thought and that you keep it like this. If you want it different on different machines you just use another build other than rammus.

jordanvanbergen, 2020-05-16:
> Also, how did you fix the camera ?

options=updates_enabled
installed the latest chrome os release 81.0.4044.141 (May 8, 2020)
After reboot the webcam works.

hussainalhaddad, 2020-05-17:
> > Also, how did you fix the camera ?
> 
> options=updates_enabled
> installed the latest chrome os release 81.0.4044.141 (May 8, 2020)
> After reboot the webcam works.

options=enable_updates

jordanvanbergen, 2020-05-17:
> > > Also, how did you fix the camera ?
> > 
> > 
> > options=updates_enabled
> > installed the latest chrome os release 81.0.4044.141 (May 8, 2020)
> > After reboot the webcam works.
> 
> options=enable_updates

Yeah... that's what I meant ;-)

sebanc, 2020-06-25:
Apparently even if I would keep original devices names in brunch, the apps sync behavior would be the same as chromeos only seems to take into account the login to sync apps: https://www.reddit.com/r/chromeos/comments/hfsb5e/issues_with_multiple_chromebooks/

closed 2020-06-25 by sebanc