sebanc/brunch #152: ChromeOS UEFI menu entry turns into ?? icon after clicking shutdown button

khanhtranngoccva, 2020-05-16:
There is no error when using sudo poweroff or sudo reboot. Other ChromiumOSes doesn't suffer this, but Project Croissant OSes do.

khanhtranngoccva, 2020-05-16:
2 menu entries with disk model and name was replaced by 1 entry with ??.
[image]
[image]

sebanc, 2020-05-16:
I don't understand what you mean, 2nd printscreen is before chromeos install and first printscreen is after chromeos install ?

khanhtranngoccva, 2020-05-16:
No, the first screen happened when rebooting after clicking the shutdown button.

sebanc, 2020-05-16:
The usb disk is not changed by the reboot therefore it is not an issue with chromeos but with your bios. Maybe for some obscure reason it does not happen with chromiumos distro but anyway it's not chromeos related. All I can say is try to update to the latest bios.

closed 2020-05-16 by sebanc