sebanc/brunch #154: force_tablet_mode disables PS2, USB Type-C over TB3 controller, I2C cameras

midi1996, 2020-05-16:
Hello, thank you for this amazing project, really fixed a lot of issues that I had with chromeOS especially tablet mode, however I'm still having some issues here and there and I hope I can find solutions for them. Here is my tablet setup:

- Model: HP elite X2 G2 1012
- CPU: i5-7300U
- GPU: Intel HD 630
- Storage: WD Blue SN500 (500GB, nvme)
- Screen: 2736*1824, with wacom touch + pen (WCOM483E, over I2C)
- Wireless: DW1560 (works with broadcom_wl and bluetooth has its firmware uploaded)
- Detachable keyboard and mouse: Kyb is PS2 connected to a controller inside the laptop, touchpad is USB connected (supports up to 5 touch points)
- USB Type-C: is not connected to the PCH's XHCI controller but through the Thunderbolt3 controller, works on all linux distros from 2016 and later.

So here are my main issues:

- When using `force_tablet_mode` the detachable keyboard stops working, the FN keys (that are controlled by EC events) work just fine, no idea about the rest why it doesn't work. Disabling this option (and doing a powerwash) would bring back the keybaord.
- I have a Type-C dongle with a USB hub and some display outputs and Ethernet, the display output (DP-alt) works fine, however the USB portion of it doesnt work. Since it's not connected through the laptop's chipset XHCI, it uses the thunderbolt 3 XHCI controller (NOT Thunderbolt 3 controller itself, but just the XHCI controller). The only way of getting working is by plugging it before starting chromeOS and then use it without unplugging it or going to sleep.
- I also have a windows hello camera, sadly it's connected through the iGPU (as what I can see from windows device manager), it has 3 sensors that are defined in the ACPI as LNK0,1,2. They only work on windows, I havent seen anyone getting it working on Linux, but idk if that would be the case with chromeOS.
- booting chromeOS through grub gives me `error: no suitable video mode found. Booting in blind mode`, it boots, I just cant see the logs like on the other devices I tried this on.

Some logs and files:
[dmesg_chromeos.txt](https://github.com/sebanc/brunch/files/4639114/dmesg_chromeos.txt)
[logs_chromeOS.txt](https://github.com/sebanc/brunch/files/4639115/logs_chromeOS.txt)

sebanc, 2020-05-20:
Hi,

1) What's your use case for "force_tablet_mode", this option has been created mainly pure tablets that do not come with a keyboard and for which tablet mode cannot be detected otherwise.
If you have a detachable keyboard, we might be able to setup udev rules to trigger tablet mode, could you run "sudo udevadm monitor", detach and re-attach the keyboard and post the output here ?

2) This might be a kernel issue, I am working on it.

3) Ipu3 cameras are not properly supported in the linux kernel and will not work with Chrome OS.

midi1996, 2020-05-20:
Thanks for your answer.

1. Yes, I usually force it since by default it's not detected if it's in tablet mode or not, it just show a regular laptop/desktop mode. Forcing it would make the UI work in tablet mode then when I plug my keyboard deck it would switch to desktop mode (since it detects a USB mouse/trackpad plugged in). Here is the log, I started seeing some stuff and I might get an idea on how to make it out, but still the keyboard part still doesnt work (it's PS2, not USB, the trackpad is USB). Thanks a lot.
[udevman_log.txt](https://github.com/sebanc/brunch/files/4656568/udevman_log.txt)
2. Thank you.
3. Yeah I lost all hopes of getting it working on anything other than windows, seems like it's not properly supported and it's more effort than it's worth (I dont really use it, it's just that with these online meetings nowadays, I can use my other laptop for that just fine) Thanks!

I might also look into fixing the USB trackpad to work with cmt drivers but not really a priority of mine (I usually work in tablet mode most of the time with chromeOS).

midi1996, 2020-05-20:
1. So I made a udev rule for dock and undock as follows ```ACTION=="add" , ATTRS{idVendor}=="044e" , RUN+="/bin/echo '0' /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode"``` and ```ACTION=="remove" , ATTRS{idVendor}=="044e" , RUN+="/bin/echo '1' /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode"``` (using sysfs_tablet_mode) and it works fine to switch between the mods, however the rotation sensor has been disabled (???), here is lsmod with `force_tablet_mode` and `sysfs_tablet_mode`, they both look the same. 
[lsmod_sysfs.txt](https://github.com/sebanc/brunch/files/4656940/lsmod_sysfs.txt)
[lsmod_force.txt](https://github.com/sebanc/brunch/files/4656942/lsmod_force.txt)
Here are dmesg outputs for each setup 
[dmesg_force.txt](https://github.com/sebanc/brunch/files/4657035/dmesg_force.txt)
[dmesg_sysfs.txt](https://github.com/sebanc/brunch/files/4657037/dmesg_sysfs.txt)

sebanc, 2020-05-20:
I think there is 1 normal behavior and 1 issue:
- Normal behavior: when tablet mode is triggered, chromeos disables the touchpad and other input devices it considers internal, I have no control over that as far as I can see (no config file, lags...)
- Issue: the sensors are not starting properly which is not related to the keyboard, tablet mode has to be enabled though.

Could you try the release I have just uploaded in the "brunch-testing" repo (not this one), It contains an attempt to fix both of your issues ?

Also, tablet mode should work out of the box if you boot with "options=sysfs_tablet_mode" on the kernel command line.

midi1996, 2020-05-20:
Sure in a moment, preparing a USB.

midi1996, 2020-05-20:
ok so I tried it and here are my observations:
- the keyboard "works" with rotation but with these issues:
  - https://streamable.com/92were (it's a video showing the twitching)
  - udevadm monitor output [udevmon.txt](https://github.com/sebanc/brunch/files/4658275/udevmon.txt)
  - This happens only when I dock the keyboard
  - The trackpad works but it's wonky (because of those add/removes) and the keyboard too but it works (so that's good so far)

- The type C now properly works! with everything (display out and USB)

here is the dmesg output 
[dmesg-new.txt](https://github.com/sebanc/brunch/files/4658304/dmesg-new.txt)

sebanc, 2020-05-20:
Could you try to replace the content of /lib/udev/rules.d/99-tablet-mode.rules with the following one ?
```
ACTION=="add", SUBSYSTEMS=="usb", ATTR{idVendor}=="0416", ATTR{idProduct}=="c141", RUN+="/bin/bash -c 'echo 0 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode'"
ACTION=="remove", SUBSYSTEMS=="usb", ATTR{idVendor}=="0416", ATTR{idProduct}=="c141", RUN+="/bin/bash -c 'echo 1 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode'"
ACTION=="add", SUBSYSTEMS=="usb", ATTR{idVendor}=="044e", ATTR{idProduct}=="1216", RUN+="/bin/bash -c 'echo 0 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode'"
ACTION=="remove", SUBSYSTEMS=="usb", ATTR{idVendor}=="044e", ATTR{idProduct}=="1216", RUN+="/bin/bash -c 'echo 1 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode'"
```

midi1996, 2020-05-20:
Sure, I just did will report soon. Also about the type C, maybe  I spoke too soon but maybe it wont work after a sleep/wake cycle, I need to test it more later on. Is there a way to force chrome OS to enter S3 mode without the timeout?

midi1996, 2020-05-20:
Ok, I dont know what happened but I lost rotation option again after a simple reboot.

midi1996, 2020-05-20:
ok with those new entries I lost rotation but it's not twitching like before.

sebanc, 2020-05-20:
Does it still go to tablet mode ? If not, that's why rotation is broken

midi1996, 2020-05-20:
> Does it still go to tablet mode ? If not, that's why rotation is broken

It does go to tablet mode, the dock becomes a simple home bar and chrome switches to tablet mode (big icons and buttons).

midi1996, 2020-05-21:
Ok so I think I understand what's going on a bit. On boot, when the keyboard is unplugged, the tablet mode is set to `0` no matter what. What I think that should be done is a check whether the trackpad exists or not on boot, and set the tablet mode accordingly or force `1` at boot then wait for udev to kick in?

midi1996, 2020-05-21:
OK, I think I fixed it, in a horrible way.
I made a simple script and put it in `/usr/bin` and named `tmod` and it goes like this
```bash 
#!/bin/bash

if [[ "$(lsusb | awk '{print $6}' | grep 044e | cut -c-4)" ]]; then
        echo 0 | tee /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode
#       cat /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode
else
        echo 1 | tee /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode
#       cat /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode
fi
```

Then changed the rule to this:

```
ACTION=="add" , ATTRS{idVendor}=="044e" , RUN+="/bin/bash /usr/bin/tmod"
ACTION=="remove" , ATTRS{idVendor}=="044e" , RUN+="/bin/bash /usr/bin/tmod"
```

I know it's quite eh, but it works and I don't know why is that instead of running the command directly from udev rule. Do you have any cleaner solution?

sebanc, 2020-05-22:
The udev rule seems too wide so I changed it a bit and also tried to simplify the rest.

/usr/bin/tmod:
`echo $1 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode`

/lib/udev/rules.d/99-tablet-mode.rules:
```
ACTION=="add", SUBSYSTEMS=="hid", ATTRS{idVendor}=="044e" , RUN+="/bin/bash /usr/bin/tmod 0"
ACTION=="remove", SUBSYSTEMS=="hid", ATTRS{idVendor}=="044e" , RUN+="/bin/bash /usr/bin/tmod 1"
```

 Does it still work with this ?

midi1996, 2020-05-22:
> The udev rule seems too wide so I changed it a bit and also tried to simplify the rest.
>
> /usr/bin/tmod:
> `echo $1 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode` [...]

ok, so you're checking for a subsystem `hid` with idVend `044e` which doesn't exist. I changed that to `KERNEL=="*:044E:1216.*"` and now it works BETTER. I noticed that the the reason why it borks is because the USB briefly reconnects which make it run in a loop (no idea why tho) so here are my rules:
```
ACTION=="add", SUBSYSTEMS=="hid", KERNEL=="*:044E:1216.*" , RUN+="/bin/bash /usr/bin/tmod 0"
ACTION=="remove", SUBSYSTEMS=="hid", KERNEL=="*:044E:1216.*" , RUN+="/bin/bash /usr/bin/tmod 1"
```

However I'm looking into making this more generic and not device specific by feeding the device/vendor ids through the boot arg, I'm trying to do that by modifying 99-tablet_mode.sh in patches folder to read `sysfs_tablet_mode.vvvv.pppp` where `vvvv` is the vendor ID and `pppp` is the product ID, then generate the needed files if needed. (First time sh scripting and since it's not bash, it's a quite challenging for me.)

sebanc, 2020-05-22:
As I used "SUBSYSTEMS" with an "S", I expected it would work by looking at underlying devices but I am not an expert in udev rules.

Anyway, I really like that udev rule you came up with and I would like for it to become the new standard in brunch, however I would prefer to avoid using the "tmod" script. Do tablet mode / sensors still work if you directly pass the full commands in the RUN+= argument ?
```
RUN+="/bin/bash 'echo 0 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode'"
RUN+="/bin/bash 'echo 1 > /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode'"
```

Also, I think that using those udev rules if included with force_tablet_mode option (not the case currently) should solve the issue of incorrect tablet mode on boot:
- Tablet mode is enabled by default
- udev launches, if the keyboard is present, tablet mode is set to 0, otherwise it remains 1.

midi1996, 2020-05-22:
> As I used "SUBSYSTEMS" with an "S", I expected it would work by looking at underlying devices but I am not an expert in udev rules.

Yeah I saw that and when I was running the tests (`udevadm test`) the rule wasn't executed (but it's properly read).

> Anyway, I really like that udev rule you came up with and I would like for it to become the new standard in brunch, however I would prefer to avoid using the "tmod" script. Do tablet mode / sensors [...]

Yes it works! (you forgot `-c`, I added it in my lines)

> Also, I think that using those udev rules if included with force_tablet_mode option (not the case currently) should solve the issue of incorrect tablet mode on boot:
> 
> * Tablet mode is enabled by default
> * udev launches, if the keyboard is present, tablet mode is set to 0, otherwise it remains 1.

Yes I added it to patched/99-tablet_mode.sh and it work too, actually that's the behaviour that I was looking for! Thank you!

sebanc, 2020-05-22:
Nice ! If you don't mind, could you attach your patches/99-tablet_mode.sh in this issue as .zip or as .txt, I will use it to be sure I don't mess things up when modifying my patch.

midi1996, 2020-05-22:
[99-tablet_mode_u.txt](https://github.com/sebanc/brunch/files/4668846/99-tablet_mode_u.txt)

Here it is, it's literally the regular one with the options added. Also I thrown the same rules for sysfs section (just in case, since it was all testing). I'm trying to figure out how dash parses this and maybe work out a way to read the dev/prod IDs of a certain device (USB in this case) for the switching.

sebanc, 2020-05-22:
Thanks ! Unless you personally need it, don't bother too much with the command line parsing, I think that even if it would be implemented, it would be too complicated for most users.

midi1996, 2020-05-22:
I think I figured it out, and I will post the resulting script soon once I'm done with testing. 👌

midi1996, 2020-05-22:
Here is the tablet_mode script I'm using now:
[99-tablet_mode.zip](https://github.com/sebanc/brunch/files/4669294/99-tablet_mode.zip)

Usage: add `options=tablet_mode.AB12.CD34` where `AB12` is the **Vendor ID** and `CD34` is the **Product ID**. I thought of switching between `hid` and `usb` SUBSYSTEMS, but I dont have any devices to test that (properly) on.

sebanc, 2020-05-23:
Wow, that's great work ! Thanks :)

I have to think about the best way of doing this but I will add this approach as a complementary one to the current approach as the current one allows it to work out of the box for inexperienced users.

sebanc, 2020-05-23:
I am thinking of adding this new feature as an option named "hid_tablet_mode". I just made small modifications to the code, you will find the updated version below:
[99-hid_tablet_mode.sh.zip](https://github.com/sebanc/brunch/files/4671952/99-hid_tablet_mode.sh.zip)

Thanks again for providing this new method :)

midi1996, 2020-05-23:
No problem man! Thank you for bringing this project to life! Now I just have to figure out how to make my USB trackpad work with cmt drivers (if it's even possible).

closed 2020-06-02 by sebanc

Insiro, 2022-11-29:
how can i use this scripts??
i want to set on my laptop