sebanc/brunch #155: Issues getting linux beta to work.

B3ATDROP3R, 2020-05-16:
I have a dell studio 1569. Android apps work fine but the linux beta fails to start.

sebanc, 2020-05-17:
Hi,

I think gpu acceleration is included in the recovery image but will not work with your gpu. Could you try to turn it off in chrome://flags and then try crostini again ?

Note: with samus recovery image, you can only access chrome://flags through guest mode.

B3ATDROP3R, 2020-05-17:
I have an Intel b450m with integrated graphics. I set the flag when I was in guest mode and Linux beta still doesn't work. I am using sammus

sebanc, 2020-05-17:
Are you booting from usb or have you installed brunch to hdd ?

B3ATDROP3R, 2020-05-17:
I am booting from a hard drive (I used that thing you'd want to make it boot on bios PC's) Android apps work fine but Linux beta refuses to start

sebanc, 2020-05-19:
I don't have much control over crostini as it only globally consist in enabling the official crostini configs in the kernel so I am no sure what could be the issue on that case. If you have parameters in your bios related to graphics/virtualisation... Could you try maxing them out ?

sebanc, 2020-05-20:
If it does not work, could you attempt to install crostini and attach to this issue the "/var/log/message" file just after it fails ?

B3ATDROP3R, 2020-05-20:
I'll have to install it back to a flash drive because I whiped it for Debian.

sebanc, 2020-06-02:
Found out that for crostini to launch on Gen 1 intel, you need to remove "virtio_gpu" from /etc/ui_use_flags.txt

B3ATDROP3R, 2020-06-02:
Ooh. What does that do?

sebanc, 2020-06-02:
It disables crostini gpu acceleration, it appears that modifying the flag in guest mode is not working.

B3ATDROP3R, 2020-06-06:
I forgot to say that fixed it. Thank you

closed 2020-06-06 by B3ATDROP3R

B3ATDROP3R, 2020-06-20:
after using this more i have come to the conclusion that doing this will break android apps. Some randomly crash or just wont open.

diego9090dd, 2020-08-25:
how did you do to use a chrome os in a legacy bios (mbr) ????? @B3ATDROP3R