sebanc/brunch #157: Sleep not working correctly

tungmeister, 2020-05-17:
Hi, I've just setup on my Dell E5480 however sleep isn't working correctly. Everything else works perfectly (other than microphone which I've added to the other issue) Once the device goes to sleep I'm unable to resume - blank screen and no activity. The only way to recover is by holding down the power button and restarting. I've tried enabling the S3 suspend option but the result is still the same. Happy to debug if you can point me in the right direction.

tungmeister, 2020-05-17:
to add, whenever I shutdown it disables my keyboard backlight so I have to go back into the bios config to re-enable it for windows or turn it back on in ubuntu settings.

tungmeister, 2020-05-19:
also noticed that cpu usage at idle seems pretty high - 8 to 10% in chrome os vs 0 to 2% idle in windows. It appears as though multithreading isn't functional either (see sceesnshots)
[image]
[image]

sebanc, 2020-05-20:
Hi,

Regarding performance / battery life I get the same result as with other OS. Which recovery image did you use ?
ChromeOS sets brightness quite high by default which is a known source for battery drain. Also, the linux kernel manages the cpu differently than windows so there can be differences (even though I highly doubt Windows cpu usage being 0 to 2%). I have a chromebook and cpu usage is the same.

Regarding suspend, could you try the release I have just uploaded in the "brunch-testing" repo (not this one), I have changed settings related to usb which might have an impact on your issue.

tungmeister, 2020-05-20:
hi, those screenshots were from linux and chrome os so it still appears that chrome os isn't seeing the multiple core threads. with regard to the CPU usage I'm in windows now and here's a shot of task manager:
[image]
I'll try the testing branch and report back shortly.

sebanc, 2020-05-20:
Forgot to mention that Hyperthreading is disabled by default in chromeos:
https://support.google.com/chromebook/answer/9340236?hl=en

tungmeister, 2020-05-20:
ah that explains it. I've been on the testing branch but the only thing available is the read me?

sebanc, 2020-05-20:
Have you looked at the release tab ?
https://github.com/sebanc/brunch-testing/releases

SoFLazerTag, 2020-05-20:
I too am having the issue described here on a Lenovo Yoga 900 laptop. It has an i7-6500u.  I've tried both rammus and lars recovery images and they behave the same. If the unit suspends, sleeps, or turns off the display it just doesn't come back and has to be powered off. It does not mess with my keyboard backlight.  I was going to try the brunch-testing you suggest but get a 404 error form the link. If you have any suggests I am will to fuss about.

tungmeister, 2020-05-21:
@sebanc Sorry, I'd had a bit too much to drink when I was looking last night and missed the release. Happy to report that my sleep issues appear to be fixed and switching to performance has enabled hyperthreading. As far as I can tell everything is now working perfectly other than the keyboard backlight getting disabled when I shutdown and the microphone requiring the startup script. I have noticed that the keyboard backlight doesn't get disabled if I reboot from the terminal though, any thoughts?

sebanc, 2020-05-21:
No worries

> Other than the keyboard backlight getting disabled when I shutdown

This one is not fixed ? Just to confirm as I attempted something to fix it

sebanc, 2020-05-21:
Also, I forgot to mention that even though you enable Hyperthreading, it will be disabled automatically if you launch crostini (it's a ChromeOS behaviour).

tungmeister, 2020-05-21:
@sebanc my bad, the keyboard backlight is behaving as expected, I'd just not reenabled it since putting on the test build. With this build my cpu utilisation at idle is back around the 0-2% mark as in windows. I'm guessing the USB issues you fixed were also potentially causing excessive CPU usage as everything is perfect now. Currently on the 83 beta at the bottom of my garden using instant tethering. Thanks for all your hard work! One minor request - I'm going to deploy this on an old laptop for my father - is there any way to disable verbose output on brunch so he doesn't see all the startup text?

sebanc, 2020-05-21:
Nice to know that most issues are fixed with this release :)

For the microphone issue, I will keep following it in #56 until we figure out the real cause.

To avoid verbose, you can add "console=" to the kernel command line (after cros_debug).

tungmeister, 2020-05-23:
@sebanc This morning when I woke the laptop after a night of sleeping the system was very sluggish and it took 5 minutes for wifi and Bluetooth to reinitialise.  I tried to get into shell to have a look at what was going on but the system was too unresponsive so I had to shut down. I'm running the latest testing brunch.

sebanc, 2020-05-23:
Indeed, I have other issues related to suspend, not sure yet what it might be related to but I am looking at it.

sebanc, 2020-06-02:
Some suspend issues appears to have been solved in the latest brunch testing release, don't hesitate to try it.

closed 2020-06-26 by sebanc