sebanc/brunch #158: Can’t expand hdd partition free space

blonqe, 2020-05-17:
Installed chrome os to hdd partition (64gb)

Img maxed out at 14gb and can’t access the rest of remaining hdd space?

Using usb method to expand data doesn’t work

sebanc, 2020-05-17:
Hi, 

You had to select the disk image size at image creation time (when running chromeos-install) by using the -s option.

The only solution is to re-create the image with the correct size.

blonqe, 2020-05-17:
Thanks for the reply, unfortunately every time I tried to specify size it stated “not enough room” so Installed it without -s

Do we have to take off the 10gb required? Therefore quote 54gb rather then the full 64gb?

Cheers

blonqe, 2020-05-17:
Was being stupid! I was inputting kB instead of gb (should have taken more notice, sorry)

Though I had 62gb free I had to state 58gb

closed 2020-05-17 by blonqe