sebanc/brunch #159: Scan device failed. Fallback to shell. iMac

MarkStruik, 2020-05-17:
Hi, thanks for al the great work here to make this happen. 

I am trying to run this on my old iMac core 2 duo. But it seems not to boot from my usb. 

See output in image. Any thought??

[image]

MarkStruik, 2020-05-17:
It’s the kingston usb. I see it loading. But it’s unable to find it somehow?

MarkStruik, 2020-05-17:
I am now also creating an image with a Samus base image

sebanc, 2020-05-17:
Hi,

The issue you are facing is related to chromeos being built with CPU instructions which are not supported by your CPU ("trap invalid opcode" messages). All current official chromeos images are supported only by 1st generation Intel CPU (referring to https://en.wikipedia.org/wiki/Intel_Core).

I have actually just tried to include a patch from android x86 to emulate those instructions, i don't think it will work but you might want to try it.

If you are interested, could you download the release in "brunch-testing" repository (not this one) and use samus recovery image with it ?

MarkStruik, 2020-05-17:
Ah i misread the wiki page :S my bad, i will try with the updated emulation. ill let you know the results

MarkStruik, 2020-05-17:
No unfortunately that doesn’t work. 

[image]

sebanc, 2020-05-17:
Thanks for trying it !

I have found one last thing to try but it's very unlikely to succeed, you can follow the comments in issue #77 to see if it is worth testing or not.

MarkStruik, 2020-05-17:
I tried with newest Ubuntu but that doesn’t boot. I will try with the mint version to check if that will work

sebanc, 2020-05-17:
Don't bother, for now it will not work. I will upload my last attempt in around 1 hour if all goes well.

sebanc, 2020-05-17:
I have just uploaded the new test release in "brunch-testing" branch if you want to try

MarkStruik, 2020-05-17:
do i need to add any special options or just rebuild the image with the new test release?

MarkStruik, 2020-05-17:
``` bash
 mark  DESKTOP-I5RNGUE  ~  chromeos  $  tar zxvf brunch_r81_k4.19_testing_20200517_v2.tar.gz
chromeos-install.sh
efi_legacy.img
efi_secure.img
rootc.img
 mark  DESKTOP-I5RNGUE  ~  chromeos  $  sudo bash chromeos-install.sh -src chromeos_12739.111.0_samus_recovery_stable-channel_mp-v3.bin -dst chromeos.img
[sudo] password for mark:
Creating image file
0+0 records in
0+0 records out
0 bytes copied, 7.11e-05 s, 0.0 kB/s
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
F29F8D67-A25A-E54F-A161-4216506E7D25
       start        size    part  contents
           0           1          PMBR (Boot GUID: F29F8D67-A25A-E54F-A161-4216506E7D25)
           1           1          Pri GPT header
           2          32          Pri GPT table
    19255304    10104776       1  Label: "STATE"
                                  Type: Linux data
                                  UUID: B6DA3D86-2AF4-7942-AA43-895AE8945C6D
     2117638       65536       2  Label: "KERN-A"
                                  Type: ChromeOS kernel
                                  UUID: 71A01515-83B7-6045-9123-26A146654987
                                  Attr: priority=15 tries=15 successful=0
    10866696     8388608       3  Label: "ROOT-A"
                                  Type: ChromeOS rootfs
                                  UUID: 96658CBB-2180-FC45-B07F-1E1DF3171CC0
     2183174       65536       4  Label: "KERN-B"
                                  Type: ChromeOS kernel
                                  UUID: B2F53706-169C-9444-8E72-1DFB4738A394
                                  Attr: priority=0 tries=15 successful=0
     2478088     8388608       5  Label: "ROOT-B"
                                  Type: ChromeOS rootfs
                                  UUID: 871DA7BB-904D-7746-8FEE-DE837E05CDDC
       16448           1       6  Label: "KERN-C"
                                  Type: ChromeOS kernel
                                  UUID: C0E37FAD-2D1B-6740-B1C3-519BC14ACB83
                                  Attr: priority=0 tries=15 successful=0
       16456     2097152       7  Label: "ROOT-C"
                                  Type: ChromeOS rootfs
                                  UUID: 8B83AE91-28D6-EE4C-84AB-9237CAC5786E
     2248712       32768       8  Label: "OEM"
                                  Type: Linux data
                                  UUID: 7F27FED5-F79E-7F45-A544-A9A7C506FF52
     2113608           1       9  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 8A8C3A0A-2E81-E146-9E20-78D39A2D37DC
     2113609           1      10  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 2BE9B356-E83B-AF43-BB37-F87D3E9E6166
          64       16384      11  Label: "RWFW"
                                  Type: ChromeOS firmware
                                  UUID: 6A9D037E-94D4-DA4C-9143-CFA2F1CEC3FE
     2412552       65536      12  Label: "EFI-SYSTEM"
                                  Type: EFI System Partition
                                  UUID: F29F8D67-A25A-E54F-A161-4216506E7D25
    29360095          32          Sec GPT table
    29360127           1          Sec GPT header
```

MarkStruik, 2020-05-17:
Seems to give less errors but still doesn’t work 

[image]

sebanc, 2020-05-17:
Indeed, I can see that the module I added tried to do something but failed... I will describe my attempts in the issue #77  and follow it there. Hopefully at some point someone will comes up with a solution to manage those unsupported cpu instructions.

MarkStruik, 2020-05-17:
Thanks for all the support! great effort! i will keep following it there. For now i might need to install Linux Mint to get some performance back on that machine.

MarkStruik, 2020-05-17:
Btw. Not sure if you can borrow something from cloudready. That does seem to boot

[image]

sebanc, 2020-05-17:
Unfortunately, cloudready builds chromiumos from source without including those cpu instructions whereas the principle of brunch is to use the official recovery images made by google and which are built with those cpu instructions.

closed 2020-05-19 by sebanc