sebanc/brunch #160: File system is read-only

alt3r3dgd, 2020-05-17:
So, I have installed the OS image like in the instructions, booted up, it was building like said, and then it started rapidly outputting errors of mkdir and rm that the filesystem is read-only, and then it says "chrome-self-repair" and my USB drive stops blinking (which means that it doesn't transfer files). I have tried reinstalling the OS and rebooting multiple times.

If this may help, I am on Ubuntu 20.04 and using samus recovery.

sebanc, 2020-05-17:
Hi, which recovery image did you try ?

alt3r3dgd, 2020-05-17:
> Hi, which recovery image did you try ?

First I tried rammus, then samus, same results.

sebanc, 2020-05-17:
You have applied the dual boot procedure right ? Did you install the image on a ntfs or ext4 partition ? Have you disabled fast startup in windows ?

alt3r3dgd, 2020-05-17:
> You have applied the dual boot procedure right ? Did you install the image on a ntfs or ext4 partition ? Have you disabled fast startup in windows ?

I didn't do a dual boot procedure yet, I have just installed it on my USB drive. Also I don't have Windows, I use Ubuntu.

sebanc, 2020-05-17:
Ok, I would need a short video of the boot process to really understand what has happened but I would say the issue occurred at image creation time. Did you see any error other than "Primary/Secondary gpt not found" warnings ?

sebanc, 2020-05-17:
Also, what's your usb flashdrive size ?

alt3r3dgd, 2020-05-17:
My usb is 16 gb. Will send the video of the boot process asap.

alt3r3dgd, 2020-05-17:
https://drive.google.com/open?id=1cwnkuImAO09P7sws82KQmp_Zg-rMoh70
Well, now it's some other problem: as it's not the first boot, it now says something about inodes and is stuck in boot loop i guess

sebanc, 2020-05-17:
Indeed the video only shows partition ROOT-A is corrupt. If you want you can press "e" quickly when you see grub and add "options=rebuild" after cros_debug.

If I were you, I would just run chromeos-install.sh again, I am nearly 100% sure the issue occurred at the time the image was created/flashed.

alt3r3dgd, 2020-05-17:
I ran chromeos-install.sh multiple times, getting the same result as for the first time. I'm gonna try the "options=rebuild" thing + make a recording of it.

alt3r3dgd, 2020-05-17:
https://drive.google.com/open?id=1BHxFcibarZHKry-vTb6EGHj3Y9H3-7PD
Now this is what I was initially seeing. The build process goes for 15 minutes, then this HUGE bunch of errors, but now instead of being stuck it just rebooted.

sebanc, 2020-05-17:
The root cause of the error is a shared library which is not loading with "invalid elf header" message which usually means that the architecture of the recovery image is incorrect. Are you sure you have downloaded the right recovery image (it should be samus in your case) ?

alt3r3dgd, 2020-05-17:
Yes, I did. I went to the link from readme and chose samus recovery version 80.

sebanc, 2020-05-17:
The only explanation in that case would be that there is a bad sector on the usb flashdrive at the exact place of the library header. Could you try the dual boot procedure ?

alt3r3dgd, 2020-05-17:
Okay, will try that tomorrow and reply whether its successful or not.

alt3r3dgd, 2020-05-18:
I tried, and as I have run the command it said "Not enought space to create disk image". The partition size is 73 GB.

sebanc, 2020-05-18:
What is the command that you have run ?
Could you also post the output of "lsblk" and "mount" commands ?

alt3r3dgd, 2020-05-18:
This is what I used:
```bash
$ mkdir -p ~/tmpmount
$ sudo mount /dev/sda3 ~/tmpmount
[sudo] password for altered: 
$ sudo bash chromeos-install.sh -src '/home/altered/Загрузки/chromeos_12739.111.0_samus_recovery_stable-channel_mp-v3.bin' -dst ~/tmpmount/chromeos.img -s 73
Not enought space to create image file
```

The output of mount command:
```bash
$ mount
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
udev on /dev type devtmpfs (rw,nosuid,noexec,relatime,size=8112920k,nr_inodes=2028230,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,noexec,relatime,size=1628212k,mode=755)
/dev/sda5 on / type ext4 (rw,relatime,errors=remount-ro)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup2 on /sys/fs/cgroup/unified type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,name=systemd)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/net_cls,net_prio type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls,net_prio)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/hugetlb type cgroup (rw,nosuid,nodev,noexec,relatime,hugetlb)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,nosuid,nodev,noexec,relatime,pids)
cgroup on /sys/fs/cgroup/rdma type cgroup (rw,nosuid,nodev,noexec,relatime,rdma)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=28,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=2405)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
hugetlbfs on /dev/hugepages type hugetlbfs (rw,relatime,pagesize=2M)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
/var/lib/snapd/snaps/code_32.snap on /snap/code/32 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/core_9066.snap on /snap/core/9066 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/core18_1754.snap on /snap/core18/1754 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/discord_109.snap on /snap/discord/109 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/gtk-common-themes_1506.snap on /snap/gtk-common-themes/1506 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/core18_1705.snap on /snap/core18/1705 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/gnome-3-28-1804_116.snap on /snap/gnome-3-28-1804/116 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/auryo_197.snap on /snap/auryo/197 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/inkscape_7627.snap on /snap/inkscape/7627 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/intellij-idea-community_221.snap on /snap/intellij-idea-community/221 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/skype_123.snap on /snap/skype/123 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/snapd_7264.snap on /snap/snapd/7264 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/zoom-client_76.snap on /snap/zoom-client/76 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/zoom-client_78.snap on /snap/zoom-client/78 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/simplescreenrecorder_1.snap on /snap/simplescreenrecorder/1 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/scrcpy_211.snap on /snap/scrcpy/211 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/intellij-idea-community_226.snap on /snap/intellij-idea-community/226 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/obs-studio_1068.snap on /snap/obs-studio/1068 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/code_33.snap on /snap/code/33 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/scrcpy_217.snap on /snap/scrcpy/217 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/gifski_58.snap on /snap/gifski/58 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/obs-studio_1071.snap on /snap/obs-studio/1071 type squashfs (ro,nodev,relatime,x-gdu.hide)
[... cut ...]
loop16   7:16   0 583,4M  1 loop /snap/intellij-idea-community/226
loop17   7:17   0 228,5M  1 loop /snap/obs-studio/1068
loop18   7:18   0 142,9M  1 loop /snap/code/33
loop19   7:19   0  81,5M  1 loop /snap/scrcpy/217
loop20   7:20   0   736K  1 loop /snap/gifski/58
loop21   7:21   0 176,7M  1 loop /snap/obs-studio/1071
sda      8:0    0 232,9G  0 disk 
├─sda1   8:1    0   512M  0 part /boot/efi
├─sda2   8:2    0     1K  0 part 
├─sda3   8:3    0    74G  0 part /home/altered/tmpmount
└─sda5   8:5    0 158,4G  0 part /
```

alt3r3dgd, 2020-05-18:
Well, I just had to remove the `-s 73` part and it works now.

sebanc, 2020-05-18:
Without -s 73, It will create an image with the default size (14 GB)

The issue is that the ext4 filesystem itself needs space + some distro apply by default reserved space on partitions:
https://odzangba.wordpress.com/2010/02/20/how-to-free-reserved-space-on-ext4-partitions/

Apply the command mentioned on the above link and try with -s 70.

alt3r3dgd, 2020-05-18:
Okay, so I have installed the system to the .img file, added the grub entry, booted, but the system didn't start at all, it just kicked me back to grub.
Previously the installation mentioned that primary and secondary GPT tables are corrupt or something.

sebanc, 2020-05-18:
the primary and secondary GPT tables are corrupt warning is normal, do you see an error message in grub ?

For ubuntu the easy way is to create a file /etc/grub.d/41_chromeos with the below content:
```
#!/bin/sh
exec tail -n +3 $0
# This file provides an easy way to add custom menu entries.  Simply type the
# menu entries you want to add after this comment.  Be careful not to change
# the 'exec tail' line above.

menuentry "ChromeOS" {
	img_part=/dev/sda3
	img_path=/chromeos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,gpt7)/initramfs.img
}
```
And then run `sudo update-grub`

alt3r3dgd, 2020-05-18:
GRUB said absolutely nothing, just silently kicked me back to menu. The 41_chromeos file changed nothing.

sebanc, 2020-05-18:
Ok sometimes the ubuntu grub does not work for some reason, try to replace the "grubx64.efi" file in /boot/efi/efi/ubuntu/ by the one that is in efi_secure.img of the brunch package (you can mount efi_secure.img this file to access its content)

alt3r3dgd, 2020-05-18:
The /boot/efi folder is empty. Could that be that I have installed Ubuntu with legacy instead of UEFI?

sebanc, 2020-05-18:
try opening it with "sudo su"

alt3r3dgd, 2020-05-18:
No, it's literally empty:
```bash
$ sudo su
[sudo] password for altered: 
# cd /boot/efi
# ls
#
```

sebanc, 2020-05-18:
Well, it's very weird because the /dev/sda1 is an efi partition but it's empty. Indeed ubuntu might have been installed in bios mode, else I don't see how uefi could boot.

alt3r3dgd, 2020-05-18:
Okay, I will try to reinstall Ubuntu in UEFI mode.

alt3r3dgd, 2020-05-18:
Well, I suffered a lot, but finally got Ubuntu on UEFI and restored most of my apps and settings.
I did the whole installation process again, and now GRUB outputs the following:
```
error: no such device: /chromeos.img.
error: file `chomeos.img' not found.
error: disk `loop,gpt7' not found.
error: you need to load kernel first.

Press any key to continue...
```
How am I supposed to load the kernel first?

sebanc, 2020-05-18:
Ok that's already better.

Could you try to replace the "grubx64.efi" file in /boot/efi/efi/ubuntu/ by the one that is in efi_secure.img of the brunch package (you can mount efi_secure.img this file to access its content) ?

alt3r3dgd, 2020-05-18:
I replaced the file and... OMG it launched!!!
The only thing is that now it shows white screen, and I pretty much know why. Can I get access to some sort of bash from the welcome screen, and is there anything like xrandr?

sebanc, 2020-05-18:
You can try CTRL+ALT+F2 to get a bash screen (not sure it will work, it depends where you are in the boot process)
However, there is no xrandr as chromeos does not use X11. What do you think is causing chromeos not to boot ?

alt3r3dgd, 2020-05-18:
It's that some "Unknown Display" is my primary display while my single HDMI display is the "second". I am now seeking for changing that in UEFI. Thanks for helping me to resolve the original issue tho!

sebanc, 2020-05-18:
In that case, could you try adding "fbcon=map:1" to your grub config kernel command line (after "cros_debug") ?

alt3r3dgd, 2020-05-18:
This didn't work. I need to get a way to swap the "monitors" order.

sebanc, 2020-05-18:
As far as I know, there is no command to control displays from shell. I think you need to search for chromeos keyboard shortcuts, I don't use them so I don't know them well. I saw somewhere that CTRL+F4 might help but there might be others.

sebanc, 2020-05-18:
Or you can try to add something like "video=HDMI-A-1:D" to the kernel command line if you know your display reference (should be the same one as with xrandr in linux)

alt3r3dgd, 2020-05-19:
Oh, wow, ctrl+f4 worked! Thanks!!!
I guess we can close the issue now..?

sebanc, 2020-05-19:
Sure ! Nice to know that CTRL+F4 worked :)

closed 2020-05-19 by sebanc