sebanc/brunch #162: Auto Rotate + Auto Brightness not working on Surface Go 2

petertriho, 2020-05-18:
Thank you for producing this project, it made dual booting win 10 + chromeos a breeze. Any donation links? 

Device: Surface Go 2 M3/8GB/256GB/LTE

Software: Windows 10 + ChromeOS 80 (Rammus) + Grub + Brunch r81 k4.19 stable 20200510
(options=iwlwifi_backport,advanced_als)

Issues: 
- Auto Rotate does not seem to work. Tablet mode seems to activate properly (after disconnecting surface type cover) and shows an auto rotate/lock orientation button in notification but auto rotate does not physically seem to work
- Tried with and without advanced_als option but auto brightness does not seem to work (is there a way for me to test this properly?)

sebanc, 2020-05-18:
Hi,

Happy to hear that you like this project ! Thanks for the donation offer, I really appreciate the gesture :) I am not rich but i would prefer that my work would serve charity/hospitals, if you can make a small donation to one that would be perfect.

To test als, I think the best way is to use your phone as a flashlight and direct it towards the screen, it should change the screen brightness level.

Regarding sensors, I think the new Surface devices (SP7, SL3) do not have kernel drivers yet, it might be the same for Surface Go 2 but I am not sure. Could you attach a "dmesg" and an "lsmod" to this issue ? I will also try to contact someone from linux-surface team, he might know more about this.

Personal question: Are Surface Go 2 cameras working ? (I guess they are not but just to confirm)

petertriho, 2020-05-18:
> i would prefer that my work would serve charity/hospitals, if you can make a small donation to one that would be perfect.

Will do!

> To test als, I think the best way is to use your phone as a flashlight and direct it towards the screen, it should change the screen brightness level.

Just tried this, brightness seems to stay the same so probably doesn't work 

> Regarding sensors, I think the new Surface devices (SP7, SL3) do not have kernel drivers yet, it might be the same for Surface Go 2 but I am not sure. Could you attach a "dmesg" and an "lsmod" to this issue ? I will also try to contact someone from linux-surface team, he might know more about this.

[lsmod.txt](https://github.com/sebanc/brunch/files/4643386/lsmod.txt)
[dmesg.txt](https://github.com/sebanc/brunch/files/4643387/dmesg.txt)

> Personal question: Are Surface Go 2 cameras working ? (I guess they are not but just to confirm)

Happy to test anything, just tested and you're right, cameras do not work

sebanc, 2020-05-18:
The good news is that the sensors driver is detected and seems in theory compatible, the bad one is that it fails quite quickly after start. I tried to have a look at the surface linux repo but could not find a similar issue (even though I suspect SP7 suffers from the same one).

Could you try installing ubuntu and the most recent linux-surface kernel and see If the accelerometer works well ? In that case, it would mean that I can just look for an existing patch.

sebanc, 2020-05-25:
Hi,

I discussed your issue with kitakar5525 and he pointed at a few patches which could help. Could you try the below test release and if it does not work post another dmesg ?
https://github.com/sebanc/brunch-testing/releases

petertriho, 2020-05-25:
Ah sorry, didn't your comment earlier. I'll test it tomorrow. Would you me to also test the linux-surface + ubuntu (20.04?)

sebanc, 2020-05-25:
Try this test release first, hopefully it will work. If it does not, I would still be interested to know if it works with linux-surface + ubuntu 20.04.

petertriho, 2020-05-25:
Amazing, just tested, auto-rotate is now working! Auto-brightness doesn't appear to be working though (not sure if this was meant to be fixed)

petertriho, 2020-05-26:
Seems like the trackpad does not work if I have the surface typecover connected upon boot (keyboard works). It requires me to disconnect the typecover then reconnect it. I tried the latest stable + the one you sent to test and they both have the same issue. 

I recently upgrade to ChromeOS V81 (rammus) so I'm not sure if this is causing the issue as I don't remember it happening in V80

I've attached the dmesg. A quick look it does seem like the touchpad is detected on boot, so I'm not too sure if it's just a bug with V81

[dmesg (1).txt](https://github.com/sebanc/brunch/files/4680289/dmesg.1.txt)

sebanc, 2020-05-26:
Hi,

I had a similar issue on my Surface Go (1) where the touchpad would randomly be detected. I added the same patch for your type cover, could you try the new test build ?
https://github.com/sebanc/brunch-testing/releases

(I hope it is the same issue because those ones are really complicated to debug)

petertriho, 2020-05-26:
That worked, not sure if I'm doing something wrong but when installing (`sudo chromeos-update -f <brunch_release>`) I saw a bunch of `job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: Bad file descriptor` messages

sebanc, 2020-05-26:
> That seemed to have fixed the issue (adding `options=alt_touchpad_config`)

Actually that's a patch for another issue, you probably don't it (the type cover patch I added is directly in the kernel)

>, not sure if I'm doing something wrong but when installing (`sudo chromeos-update -f <brunch_release>`) I saw a bunch of `job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: Bad file descriptor` messages
> 
> Says "Brunch updated." and seems to have updated just fine though.

Those are harmless warnings that I am not managing to get rid of. You can avoid them by running "chromeos-update" from the default crosh shell directory.

petertriho, 2020-05-26:
Tried it again without `options=alt_touchpad_config` and it works 

Reinstall from home (~) directory and those warnings aren't showing up either.

Thanks for the help!

closed 2020-05-26 by petertriho