sebanc/brunch #164: ChromeOS shutdown from powerd (or any reasons apart from shutdown command) crashes UEFI and breaks Android container for the subsequent reboot with signal 15 (AGAIN!)

khanhtranngoccva, 2020-05-18:
I have already updated UEFI from v1.5 to v1.6, but the error persists. Other Chromium distros might have tweaked powerd to fix the problem. Can you look into shutdown modules? My UEFI BIOS version is H81M-DGS R2 v1.6(latest).

khanhtranngoccva, 2020-05-18:
I also observed that the CMOS clock was reset to UTC time.

sebanc, 2020-05-19:
Hi,

I have added a new test release in my github "brunch-testing" repo (not this one) where I disabled CMOS update in case it could be causing an issue.

Regarding powerd, as I am not building chromeos from source it might be difficult to change things related to it but we will see that later.

Could you try the testing build, confirm that CMOS is not updated anymore and if your issue persist ?

khanhtranngoccva, 2020-05-19:
No, it still doesn't work. ChromeOS still crashes UEFI whenever the system powers off, making the menu entry for the next reboot "??" and the second reboot will still result in a USB timeout. Rebooting with any means isn't affected.

sebanc, 2020-05-19:
Ok, could you confirm the CMOS update part is fixed ?

Regarding powerd, I am not building chromeos from source, so I cannot modify it and it is a core part of the system so I cannot remove it. What you can do:
1) Play with powerd parameters which are located in /usr/share/power_manager/ and see if there is one which fixes your issue.
2) Try other recoveries, they may be compiled with different powerd flags.

Also, as a side note, it still looks like an efi bios issue and it is not because your efi bios is up-to-date that it means it is not bugged.

khanhtranngoccva, 2020-05-19:
[Shutdown_failure_log.zip](https://github.com/sebanc/brunch/files/4650266/Shutdown_failure_log.zip)
Would you mind analyzing the log for errors? My ROM shuts down for the first time at 06:47PM UTC+7, so you will see the log confirming the first shutdown at 11:47AM UTC (due to CMOS reset).

sebanc, 2020-05-19:
I had a close look at the logs, I saw nothing that would explain the menu entry issue after reboot, however there are a lot of warnings regarding the lack of battery.

khanhtranngoccva, 2020-05-19:
What happens when there is no battery in ChromeOS? I am using a custom PC.

sebanc, 2020-05-19:
That's a good question, i don't know actually I have only laptops. Maybe you could try a chromebox recovery image like fizz ?

Also, you could try modifying the battery related settings in /usr/share/power_manager/

sebanc, 2020-05-19:
Also, I just had a look at the powerd source code and I have no idea of what could be causing the menu entry issue:
https://chromium.googlesource.com/chromiumos/platform2/+/master/power_manager

khanhtranngoccva, 2020-05-20:
Well, I tried to set USB and storage settings like this and get some results:
1. Enable Intel Smart Auto: No more disk timeout error.
The Android container still crashes (even after powerwashing) and POST now hangs forever instead of the menu entry error.

khanhtranngoccva, 2020-05-20:
2. Disable USB Legacy support: nothing changes, but the PC bricks.

sebanc, 2020-05-20:
Is your motherboard able to boot bios/mbr devices ? Maybe if the issue is related to UEFI, bios boot would work better ?

khanhtranngoccva, 2020-05-20:
Enabling Intel USB3.0 by default + disable boot failure guard solves the USB timeout bug, but Android container keeps failing to mount.
2020-05-20T20:27:48.602671+07:00 DEBUG arc-kmsg-logger[2157]: vdc: Waited 1030ms for vold
2020-05-20T20:27:48.602856+07:00 ERR arc-kmsg-logger[2157]: vdc: Failed: Status(-8): '0: '
2020-05-20T20:27:48.603218+07:00 INFO arc-kmsg-logger[2157]: init: Service 'exec 2 (/system/bin/vdc --wait cryptfs init_user0)' (pid 15) exited with status 25 waiting took 25.879999 seconds
2020-05-20T20:27:48.603263+07:00 ERR arc-kmsg-logger[2157]: init: Failure (reboot suppressed): init_user0_failed

khanhtranngoccva, 2020-05-20:
Yes, but bios boot with croissant images won't fix the menu entry error, or fix the container.

sebanc, 2020-05-20:
If you know that bios won't work, no need to try. Nice that you are making progress.

Regarding android apps, now your issue really looks like #74, have you tried waiting a bit on the login screen as suggested in this issue ?

khanhtranngoccva, 2020-05-21:
[  107.984682] init: apex-monitor post-stop process (1392) terminated with status 1
[  111.667839] init: cros_healthd main process (1475) terminated with status 254
[  111.668744] init: cros_healthd main process ended, respawning
[  111.766844] init: cros_healthd main process (1614) terminated with status 254
[  111.767716] init: cros_healthd main process ended, respawning
[  111.848701] init: cros_healthd main process (1625) terminated with status 254
[  111.849534] init: cros_healthd main process ended, respawning
[  111.912759] init: cros_healthd main process (1632) terminated with status 254
[  111.913605] init: cros_healthd respawning too fast, stopped
[  118.933247] fscrypt: AES-256-CTS-CBC using implementation "cts(cbc(aes-generic))"
[  119.480284] fscrypt: AES-256-XTS using implementation "xts(ecb(aes-generic))"
[  154.774680] cgroup: run_oci (1361) created nested cgroup for controller "memory" which has incomplete hierarchy support. Nested cgroups may change behavior in the future.
[  154.777215] cgroup: "memory" requires setting use_hierarchy to 1 on the root
[  155.706356] EXT4-fs (sdb1): re-mounted. Opts:
[  162.020863] binder: 2313:2313 ioctl 4018620d 7ffe40b189d0 returned -22
[  183.354079] init: arc-boot-continue pre-start process (2269) terminated with status 1
[  183.683935] init: arc-kmsg-logger main process (2251) killed by TERM signal
[  183.749796] init: usb_bouncer main process (2275) terminated with status 1

sebanc, 2020-05-21:
This one is totally normal:
[ 154.774680] cgroup: run_oci (1361) created nested cgroup for controller "memory" which has incomplete hierarchy support. Nested cgroups may change behavior in the future.

The cros_healthd issues seems more suspicious to me, maybe you can try to run:
sudo mv /etc/init/cros_healthd.conf /etc/init/cros_healthd.backup

and reboot to see if it helps.

sebanc, 2020-05-21:
Could you try the new release I have uploaded on the "brunch-testing" branch ?

khanhtranngoccva, 2020-05-21:
Well, I don't know what happens when the system halts and the disk has not been unmounted. Will it mean a hot plug, and what happens when a Linux Live USB gets hot-plugged?
[umount-encrypted.log](https://github.com/sebanc/brunch/files/4662256/umount-encrypted.log)

khanhtranngoccva, 2020-05-21:
The USB timeout error has already been solved by enabling Intel USB3.0 Mode, but ChromeOS did not unmount properly and this is another issue, which may be the reason for my other issues.

sebanc, 2020-05-21:
Indeed that's weird but I would say it is a consequence of the container not being correctly mounted in the first place.

khanhtranngoccva, 2020-05-21:
What happens if you pull the plug of a USB when it's running Linux Live USB?

sebanc, 2020-05-21:
You have a delay between the moment the write instruction is executed and when it is actually written on the disk. If you unplug an usb you loose the changes that have not been synced on the disk which can be quite bad or not depending on how important the data that have not been synced are.

khanhtranngoccva, 2020-05-21:
How about pulling the plug of a PC?

sebanc, 2020-05-21:
Exact same thing.

If you want to disable shutdown, you can try running:
`echo "exit 0" | sudo tee /usr/bin/powerd_setuid_helper`

sebanc, 2020-05-23:
Also, I again attempted a fix in the release I have just uploaded in "brunch-testing" repository, don't hesitate to try it.

khanhtranngoccva, 2020-05-26:
Well, I threw the ChromeOS logs into the official Google forum.

khanhtranngoccva, 2020-05-27:
Your newer package still fails. Have you inspected arc.log carefully?

sebanc, 2020-05-27:
Yes I did with my limited knowledge of how arc work, knowing that this part of chromeos is not open source.

khanhtranngoccva, 2020-05-27:
NOT open source? I thought there is a page on Chromium source.

sebanc, 2020-05-27:
The source code is not available as far as I know.

I looked at arc.log trying to find differences with other logs I have but there was nothing obvious as to why it fails (just some battery related issues which might actually be normal).

khanhtranngoccva, 2020-05-28:
[dmesg.log](https://github.com/sebanc/brunch/files/4693974/dmesg.log)
How about this? Is there a problem with my PC or software?

khanhtranngoccva, 2020-05-28:
Another thing is that after waiting for one minute and a half, the container boots at first login without fail, but waiting like that destroys the purpose of ChromeOS.

khanhtranngoccva, 2020-05-28:
One more observation relating to delayed sign in: Whenever the fscrypt process runs before run_oci cgroup is created, the container shuts down, and when the fscrypt process run later than when the run_oci cgroup is created, the container starts with no fails

khanhtranngoccva, 2020-05-28:
[image]
[image]

sebanc, 2020-05-28:
I am not sure I understand what you are saying, correct the below if something is incorrect:
- On first boot, android apps always install without problems.
- On second boot android apps do not work if you login directly and you have to wait 1min30 for them to work. Do you have to wait 1 min 30 before login or after login ?

khanhtranngoccva, 2020-05-28:
YES. I see that linux binder takes 3 minutes (176 seconds) from BIOS to start, while getting to log in screen takes only 2 minutes (117 seconds)

sebanc, 2020-05-28:
Do you need to wait 1min 30 specifically at the login screen ?
Or does it work even if you login directly ?

khanhtranngoccva, 2020-05-28:
Yes.

sebanc, 2020-05-28:
yes to which ?

khanhtranngoccva, 2020-05-28:
I had to wait for 1min to let the container work.

sebanc, 2020-05-28:
I understand that you have to wait but you did not answer my questions.
Do you specifically have to wait at the login screen ?
Or does login in and waiting has the same effect ?

khanhtranngoccva, 2020-05-28:
When I log in too soon, the device decryption starts before binder module starts which makes "binder" unusable and Android container crashes (as seen in dmesg). However, if I wait, the binder module starts before decryption and it is usable. Is it correct?

sebanc, 2020-05-28:
The android binder is built-in the kernel, not a module so that's not it.

Moreover, since the 20200526 release, the login screen is not displayed before all modules have been loaded which has already fixed a similar issue for another user as reported in #74.

As far as I can tell, those logs seems more like the consequence of the issue than its origin.

khanhtranngoccva, 2020-05-28:
Then why could I log in before ashmem and binder have been started?

sebanc, 2020-05-28:
I must have missed something, what makes you think you logged in before ashmem and binder had been started ?

khanhtranngoccva, 2020-05-28:
The system decrypts before binder sends a signal.

sebanc, 2020-05-28:
My understanding is that the android container fails, send a bad call to the binder which in turn fails.

Here is the android binder kernel config:
CONFIG_ANDROID=y
CONFIG_ANDROID_BINDER_IPC=y
CONFIG_ANDROID_BINDER_DEVICES="binder,hwbinder,vndbinder"

khanhtranngoccva, 2020-05-28:
Then I don't know why I had to wait in the sign in screen for a minute.

sebanc, 2020-05-28:
I don't know also, what i can tell you for sure is that since 20200522 all modules are loaded before login screen is displayed so that's not it.

Have you tried updating chromeos to see if it helps ?

khanhtranngoccva, 2020-05-28:
I am on latest ChromeOS.

sebanc, 2020-05-28:
r83 ?

khanhtranngoccva, 2020-05-28:
No. I use r80 stable.

khanhtranngoccva, 2020-05-29:
If I log in right after this log line: [  168.224761] capability: warning: `main' uses 32-bit capabilities (legacy support in use), ARC++ works, but if I log in right before the line, it fails.

sebanc, 2020-05-29:
Could you try editing /etc/init/arc-system-mount.conf and replacing:
"start on starting boot-services"
by
"start on started ui"

khanhtranngoccva, 2020-05-29:
No, I still have to wait until that line in dmesg spawns to be able to log in and use ARC++.

sebanc, 2020-05-29:
Could you revert the previous change and try editing /etc/init/ui.conf replacing:
"start on started boot-services and stopped udev-trigger"
by
"start on started boot-services and started arc-system-mount"

khanhtranngoccva, 2020-05-29:
Still not working.
[image]

sebanc, 2020-05-29:
I think this one might break boot but it is worth a try, edit

khanhtranngoccva, 2020-05-29:
The system failed to boot completely. Can you make the process related to the 'capability: warning: `main' uses 32-bit capabilities (legacy support in use)' line start faster?

[47 more replies]