sebanc/brunch #165: Pixel 2013 - sleep mode not working

Tommy35, 2020-05-18:
Hi,

first of all, thanks for this awesome mod. I have Google Chromebook Pixel 2013 and the sleep mode doesnt work when i close the laptop. I have the version installed, it just locks the OS, turn off keyboard and screen but the laptop is still running. Any fix for it? :)

Tommy35, 2020-05-18:
Okay, so now even touchscreen stopped working. :(

sebanc, 2020-05-19:
Hi,

Just to say that I observed some weird behavior with the current chromiumos kernel source regarding networking and suspend for which I have not been able to clearly identify the origin of the issue.
I expect that this is caused by a bad commit which will be corrected in the next kernel release. I will update this issue once a release with a newer kernel is available.

Regarding touchscreen issue, I don't think it is related, how did it happen ? After a certain time of use ? Did you add brunch options ?

Tommy35, 2020-05-19:
It probably happened after updating do 81 build. I had the latest brunch installed. That was on rammus build.

I also tried Samus, but here was a problem with touchpad, I needed to use two fingers to move cursor.

With Eve it didn't even booted.

I was thinking about Hatch (Samsung), it has same keyboard layout, do you think there is a chance?

Should I stick with overally recommended Rammus, or if I have 3rd generation Intel, the Samus and wait for the fix?

Thank you for reply and loving work You do! :)

sebanc, 2020-05-19:
You will have to use "samus" as the other images you mention will not support your cpu from next version (r83 as r82 has been cancelled).

It seems samus must have a specific touchpad setup but we can try reverting it. Could you try installing samus and removing the line "natural_scroll_default" from /etc/ui_use_flags.txt ?

Tommy35, 2020-05-19:
So, I installed samus, updated OS to 81, touchscreen works. Only problems is the touchpad, need to hard press and move, or two-finger move. And the sleep mode still not working, it doesnt go to sleep, and looks like after few minutes the laptop shuts down completely.

About the removing line you suggesting.. can You please guide me trough how to do that? Not really expert in these things.. yet.. 👍

sebanc, 2020-05-21:
Could you try the lulu recovery image and see if it helps with the touchpad ? I think r80 does not have crostini but it was added to r81.

Tommy35, 2020-05-21:
Ok, installed LULU recovery with 20200510 brunch (read about some new testing one from You, but I cant get to it).
Looks like everything works, but sleep still doesn't. :)

sebanc, 2020-05-21:
The testing release is here:
https://github.com/sebanc/brunch-testing/releases

Hope it will fix your suspend issue

Tommy35, 2020-05-21:
Thank You, bu it didn't help, when i close laptop, the led still lights indicator blue, when i open it, display and keyboard start, and wifi is reconnecting. Im testing in Guest mode. Any other help? different recovery, or wait for brunch fix? possibly ChromeOS83 could fix it? For example CloudReady worked fine in sleep mode.

Thank You for Your time. :)

sebanc, 2020-05-22:
Could you try to see if adding "options=suspend_s3" to the kernel command line helps ?

Tommy35, 2020-05-22:
also, didn't help, i even updated to Brunch r81 k4.19 testing 20200522. :/ Cloudready, Gallium OS also worked with sleep mode. There must be some bug, i remember i have to few times on CR with some developer builds.

Tommy35, 2020-05-31:
Can You please respond @sebanc ? Thank You. :)

sebanc, 2020-05-31:
Yes, I confirm that the suspend issue is known even though it does not impact everyone and I am working on it but suspend issues are difficult to debug.

Could you try the build I have posted today in "brunch-testing" branch ?

smilejek, 2020-06-20:
> Can You please respond @sebanc ? Thank You. :)

> also, didn't help, i even updated to Brunch r81 k4.19 testing 20200522. :/ Cloudready, Gallium OS also worked with sleep mode. There must be some bug, i remember i have to few times on CR with some developer builds.
> 
> EDIT: I even tried Shift+Search+L (shoult trigger sleep mode), screen and keyboard lids out, but in a second its back on login screen.

Hi Tommy,
I'm fairly new owner of this Chromebook Pixel 2013,
can you tell me, what's the best recovery for this model and does it have Google Play Store enabled?

sebanc, 2020-06-26:
Not sure if you still have the suspend issue but just found this on arch wiki and thought it could be of interest:
https://wiki.archlinux.org/index.php/Chrome_OS_devices#Fixing_suspend

sebanc, 2021-10-09:
@u9wy Using "kernel-chromebook-5.4" is probably the best way to fix this issue. If you want to use kernel 5.10 you can try the directions in arch wiki: https://wiki.archlinux.org/index.php/Chrome_OS_devices#Fixing_suspend