sebanc/brunch #166: Confirming no AMD support (at least for my model)

TILVince, 2020-05-18:
Performed install with grunt build ver 80 (only one that mentioned AMD) on a Lenovo X120e (AMD processor).  Upon initial reboot, got an error at scanning device /dev/sda:

traps: blkid[115] trap invalid opcode ip:7a1725d98ae9 sp:7ffdc9ecddc0 error:0 in libblkid.so.1[7a1725d85000+24000] Illegal instruction.

Then, "ChromeOS was not found...fallback to shell."

Thanks for this project, though.

sebanc, 2020-05-21:
Hi, thanks for the report :) Indeed, unfortunately your device is much older than the "grunt" recovery image ones and there are cpu instructions built into the image which are unsupported by your cpu.

tornadox, 2020-05-22:
confirmed same issue on AsRock e350m1 (same apu) it's too old actually to run modern os in 1080p. Cloudready of course works there, but thats another story.

closed 2020-06-02 by sebanc

MP715, 2020-11-24:
I'm getting a similar error but I have an Intel Core 2 Duo T5250 / 1.5 GHz 8 gigs ram
[image]