sebanc/brunch #167: Dell E7440 - Touchpad issue

fdotcico, 2020-05-19:
Hi,
i had installed successfully the nani version of Chrome OS on my Latitute E7440. Everything is working correctly but the touchpad got few problems. When pressed, the arrow icon is wobbling (seems too much sensitivity). Is there any method to fix it? 

Thanks in advance

sebanc, 2020-05-19:
Hi,

You can try to modify the first block of the file /etc/gesture/40-touchpad-cmt.conf to better manage your touchpad:
```
# Configure touchpads to use Chromium Multitouch (cmt) X input driver
Section "InputClass"
    Identifier      "touchpad"
    MatchIsTouchpad "on"
    MatchDevicePath "/dev/input/event*"
    Driver          "cmt"
    Option          "AccelerationProfile" "-1"
    Option          "Scroll Buttons" "0"
    Option          "Scroll Axes" "1"

    # CMT devices potentially process keyboard events
    Option          "XkbModel" "pc"
    Option          "XkbLayout" "us"
EndSection
```

As a start you can try to change  `Option          "AccelerationProfile"` value. You can aso look at the other configs found below in that same file or at the ones available at:
https://github.com/hugegreenbug/xf86-input-cmt/tree/master/xorg-conf

If you find a config that works, could you post it here to help others with the same touchpad model ?

au79912, 2020-05-22:
i have the same issue
can you guide me on how to do that?

sebanc, 2020-05-22:
run "sudo nano /etc/gesture/40-touchpad-cmt.conf" in crosh shell and modify "AccelerationProfile" value

au79912, 2020-05-22:
> sudo nano /etc/gesture/40-touchpad-cmt.conf

im sorry but im not really well versed in CLI
can yo guide me as to how i can modify the value?

sebanc, 2020-05-22:
use CTRL+ALT+T to open crosh, type "shell" at the invite, then run "sudo nano /etc/gesture/40-touchpad-cmt.conf".
Modify the value.
Press CTRL+x to quit, "y" to save and ENTER to confirm the file name.

au79912, 2020-06-02:
using options=alt_touchpad_config in the grub config helped a lot in fixing this. the jitter is still there but it has greatly decreased and is gone completely when scrolling with two fingers. using a dell latitude E6440 BTW

closed 2020-06-26 by sebanc