sebanc/brunch #168: Can't first login

MrGKanev, 2020-05-19:
Hi guys,
Firstly I love the idea of this project and thanks for the hard work. I have a "Chuwi LapBook Air" on which I installed with " rammus " with brunch. (The idea is to be the only OS on the laptop)  The install went great, everything works (connected with wifi at least) but I can't sign in. Every time it says that it can't verify the email or password. I tried even creating a new account and still didn't work.
The internet is working because when I try to login I see my avatar showing when writing the password so the internet is connected for sure, and I have checked if the password is correct. I tried "Powerwash" 2-3 times but there isn't any differences.

sebanc, 2020-05-19:
Hi,

I am not sure I correctly understood your issue.

After install you arrive on the initial welcome screen, proceed with wifi, enter your gmail username / password, then the blue circle spins for 5 minutes and fails to login, is that it ?

Did you try it on usb ? If so, were you able to login ?

MrGKanev, 2020-05-19:
Hi,
Sorry If I wasn't understandable. Yes, I installed the os, connected to the WiFi, entered my account details and it spins that is it. (over and over again)
I'm not sure what do you mean by "did you try it on USB?". I directly installed it on the SSD.

sebanc, 2020-05-19:
Usually this issue is related to the TPM, could you try turning off the TPM in the bios or try changing the settings related to it ?

MrGKanev, 2020-05-19:
Removed everything connected with "security" on this machine but the problem is still the same. (fTPM also is off) Should I try reinstalling it with the settings still turned off or there isn't going to be a difference?

sebanc, 2020-05-20:
Could you try to enable fTPM if it was already off ?

MrGKanev, 2020-05-20:
I tried it "off" and "on". (fTPM and it was on the first time) Still no differences, I will install it on a flash drive so I can continue testing on this machine. (Will install a temporary OS just so I can use it for work.)

sebanc, 2020-05-20:
The issue is clearly with swtpm, I would have wanted to find a solution as I don't see what could go wrong with it, it has no specific dependency.

What should definitely work is to use the "samus" recovery as it does not use swtpm.

sebanc, 2020-06-02:
I believe this issue is fixed, don't hesitate to re-open it if it is not the case.

closed 2020-06-02 by sebanc

madhits, 2023-05-19:
same issue for me on my hp touchbook that is very similar to samus, pixel book 2015. Any hope, Im using ramus 107.

See: https://github.com/sebanc/brunch/issues/1549

UPDATE: There is no fix, google apparently made huge changes to something in there code with 101 that Brunch can not overcome (e-waste prevails again, RIP machines from 2013-2016). So Your best bet is Brunch 102 and ramus 100.