sebanc/brunch #169: Thinkpad Helix 2 - Sleep not working

nicogig, 2020-05-19:
Hi all,

First off, thanks for this project!
I am currently facing a sleep issue on my Thinkpad Helix 2. Whenever I try to put the tablet to sleep by pressing the power button, the screen does turn off and the device locks itself, but it continues to run normally (i.e. the red status light on the back does not switch to pulsating). Any ideas of what might cause this?

Thanks!

midi1996, 2020-05-19:
Tried `options=suspend_s3` ?

In my case, having this would put my tablet to sleep after a while (like after a minute or two), keeping the device powered on for a while just in case you need to use it after a quick lock. Closing the screen with the keyboard deck might put it directly to S3 mode (sleep).

nicogig, 2020-05-19:
> Tried `options=suspend_s3` ?
> 
> In my case, having this would put my tablet (not the same one as yours) to sleep after a while (like after a minute or two), keeping the device powered on for a while just in case you need to use it after a quick lock. Closing the screen with the keyboard deck might put it directly to S3 mode (sleep).

Yeah, tried that option. Unfortunately it does not work at all without the keyboard attached and when closing the screen with the keyboard on  it shuts down the tablet entirely.

sebanc, 2020-05-19:
@nicogig Indeed, I think it is a normal behavior, it will suspend:
- If on ac, according to the delay in /usr/share/power_manager/plugged_suspend_ms
- If on battery, according to the delay in /usr/share/power_manager/unplugged_suspend_ms

Could you confirm if it is working as expected ?

nicogig, 2020-05-19:
> @nicogig Indeed, I think it is a normal behavior, it will suspend:
> 
> * If on ac, according to the delay in /usr/share/power_manager/plugged_suspend_ms
> * If on battery, according to the delay in /usr/share/power_manager/unplugged_suspend_ms
> 
> Could you confirm if it is working as expected ?

According to the delays reported there, it should suspend after 51s when plugged in and after 39s when unplugged. Unfortunately my tablet does not suspend at all.

sebanc, 2020-05-20:
I think you are missing a 0, it is 510s and 390s if I am not mistaken. Could you confirm that after pressing the tablet button and waiting 10 minutes the tablet is still not sleeping ?

nicogig, 2020-05-20:
> I think you are missing a 0, it is 510s and 390s if I am not mistaken. Could you confirm that after pressing the tablet button and waiting 10 minutes the tablet is still not sleeping ?

Hi, yes indeed there was a 0 missing, sorry about that. I have tested with the option "suspend_s3" on and the tablet shuts down entirely after 10 minutes, while without the option the tablet turns the screen on again.

midi1996, 2020-05-20:
do you happen to have `Modern Sleep` `Instant Wake` `Modern Standby` `InstantON` or the such in your UEFI Setup options?

nicogig, 2020-05-20:
> do you happen to have `Modern Sleep` `Instant Wake` `Modern Standby` `InstantON` or the such in your UEFI Setup options?

Had a look in the UEFI and no such options are present.

sebanc, 2020-05-21:
@nicogig Could you try this release ?
https://github.com/sebanc/brunch-testing/releases

nicogig, 2020-05-21:
> @nicogig Could you try this release ?
> https://github.com/sebanc/brunch-testing/releases

Just tried with the option suspend_s3 on and on battery. 6m30s after pressing the power button the screen turns back on, stays on for about 10s, and then the screen turns off again. The tablet, however, never goes to sleep. Something I noticed while I was observing this behaviour is that when the screen turns back on, I can see that the Wi-Fi was turned off and it is reconnecting, which doesn't happen if I manually turn the screen on. So at least it seems that it is *trying* to go to sleep.

I will try without the suspend_s3 option and report back if there is any difference.

chenleo92, 2020-05-23:
I have similar problem on two laptops, feels this could be a problem with external drive stopped after suspend and when it wakes up, the system can't find original loop device files to be awake.

sebanc, 2020-05-23:
@chenleo92 Indeed, I have identified an issue regarding to the kernel config CONFIG_USB_DEFAULT_PERSIST not being set which I have corrected in the test version available here:
https://github.com/sebanc/brunch-testing/releases
Could you try this build ?

Unfortunately, even with that, it still does not work for @nicogig so there is apparently another issue in his case.

sebanc, 2020-06-02:
Could you try the latest "brunch-testing" repository release, it appears to have fixed suspend issues for some ?

nicogig, 2020-06-06:
> Could you try the latest "brunch-testing" repository release, it appears to have fixed suspend issues for some ?

Sorry for the late reply. I tried the last testing building (0604) but to no avail unfortunately. Sleep is still not working.

sebanc, 2020-06-06:
That's too bad this version appears to improve suspend for some but thanks for the link to your device arch wiki.

What's the output of "cat /system/usr/share/power_manager/board_specific/suspend_to_idle" ?

nicogig, 2020-06-06:
> That's too bad this version appears to improve suspend for some but thanks for the link to your device arch wiki.
> 
> What's the output of "cat /system/usr/share/power_manager/board_specific/suspend_to_idle" ?

I’m assuming you are referring to /usr/share/power_manager/board_specific/suspend_to_idle since i couldn’t find the system folder.

In any case, the output from the cat is 1.

sebanc, 2020-06-06:
Yes, indeed I meant /usr/share/power_manager/board_specific/suspend_to_idle which is correctly set to 1 so that's not related.

Do you use an sdcard as external storage for android apps ?

nicogig, 2020-06-06:
> Yes, indeed I meant /usr/share/power_manager/board_specific/suspend_to_idle which is correctly set to 1 so that's not related.
> 
> Do you use an sdcard as external storage for android apps ?

No everything is stored on device. Doing cat /usr/share/power_manager/suspend_mode i noticed the output is “deep” which would theoretically lead to S3 sleep. According to the Chromium Docs this shouldn’t affect sleep because s2idle takes precedence but I wonder if that might be an issue?

sebanc, 2020-06-06:
> No everything is stored on device. Doing cat /usr/share/power_manager/suspend_mode i noticed the output is “deep” which would theoretically lead to S3 sleep. According to the Chromium Docs this shouldn’t affect sleep because s2idle takes precedence but I wonder if that might be an issue?
> 

I have the same understanding, you can try to change it but it should make no difference.

> Edit: looking through the docs i also noticed that the Suspender tries ten times to suspend the PC before invoking a shutdown. That would explain why sometimes the tablet just shuts down instead of suspending

Indeed that could explain it. If you see that your device fails while trying to suspend and you manage to make it resume, could you try to grab a dmesg, it could help identifying the culpirit module.

nicogig, 2020-06-06:
> I have the same understanding, you can try to change it but it should make no difference.
>
>
> Indeed that could explain it. If you see that your device fails while trying to suspend and you manage to [...]

I managed to grab the logs, I'm attaching it to this post as it would be too much to put in a comment.
[dmesg.txt](https://github.com/sebanc/brunch/files/4740929/dmesg.txt)

There are some references to thinkpad-acpi and HKEY event 0x4013 being unhandled. This should be the docking event (dock is not connected FYI) and has been reported to the thinkpad-acpi devs already (see https://sourceforge.net/p/ibm-acpi/mailman/message/34083474/) hence maybe the driver is not up-to-date?

sebanc, 2020-06-07:
Actually as far as I can tell this suspend log looks perfect, did you make it resume from suspend or did it wake spontaneously ?

nicogig, 2020-06-07:
> Actually as far as I can tell this suspend log looks perfect, did you make it resume from suspend or did it wake spontaneously ?

It woke up by itself.

sebanc, 2020-06-07:
Those messages are normal as far as I know, I have those on my Surface devices and have no issue with s2idle.

I think the real issue is related to this (from the arch wiki) and that the ec events actually not only resume devices while suspended but completely resume the laptop:
```
Disable embedded-controller wake-ups
By default, s2idle will still exhibit a significant battery drain while suspended (the batteries will be dead within a few hours). It appears that the device suffers from embedded controller wake-ups. For a more reasonable drain while suspended (i.e. you can leave it suspended for days), you must set the acpi.ec_no_wakeup=1 kernel parameter.

Note: If you set the acpi.ec_no_wakeup=1 kernel parameter, the device will not be able to wake up after any lid events (attaching/detaching the tablet from the base). So, it's a trade-off: enable the parameter and enjoy proper suspend behavior as long as you don't remove the device from the base (e.g. to close it), or do not set the parameter and enjoy the ability to close and re-open the laptop but suffer wake-level battery drainage.
```

Could you try to see if suspend works with "acpi.ec_no_wakeup=1" added to the kernel command line ?

nicogig, 2020-06-07:
> Those messages are normal as far as I know, I have those on my Surface devices and have no issue with s2idle.
>
> I think the real issue is related to this (from the arch wiki) and that the ec events [...]

Right, so I feel this is going somewhere. With the option on the ThinkPad light shuts off (!) and the device does not turn the screen back on. But that's it. Light doesn't "breathe" to indicate sleep, power button does nothing, every other button does not wake the device. Windows button reacts to touch (i.e. vibrates) which is not usual behaviour for sleep mode as far as I remember. So it seems that with this option and no base the tablet does not react at all.  Will test with the base and report back.