sebanc/brunch #170: Network adapter not detected

halcut, 2020-05-19:
Installation went well on my desktop which previously had Linux mint. However after booting it is stuck on initial setup as it could not detect any network connection. I cannot proceed any further as I need to select a network connection first. Desktop only has wired connection.. no wifi adapter. Pls help.

sebanc, 2020-05-20:
Hi,

Could you try the release I have just uploaded in the "brunch-testing" repo (not this one), I have added support for more ethernet adapters.

If it still does not work, could you attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads directory)

halcut, 2020-05-21:
Thanks. Let me try it.

halcut, 2020-05-21:
It worked. Thanks a lot.

closed 2020-05-21 by sebanc