sebanc/brunch #173: MBR/BIOS boot

omminp, 2020-05-20:
Thank you very much for your efforts with this project.  It is so worthwhile. And I have been searching for a way to boot brunch created code on non-UEFI machines, so your update was timely.

I have tried the alternative brunch MBR/BIOS install using brunch_81 and samus recovery.  All worked without error, but the USB stick fails to boot with the message 'Missing OS'

I have repeated a few times with the same result.  What can I check to troubleshoot?

sebanc, 2020-05-20:
Hi,

Could you confirm that you have extracted the "mbr_support.tar.gz" package in the same directory as the brunch release before running chromeos-install.sh ?

If so, could you mount partition 12 of your disk in linux and provide the list of files present in "syslinux" directory ?

omminp, 2020-05-20:
Hi,
Yes, mbr_support.tar.gz extracted to same directory, and the
chromeos_install.sh file was overwritten. New .sh  file has code to load
mbr.bin.

pmn@pmn-PC:~$ ls -l  /mnt/new/syslinux
total 15456
-rwxr-xr-x 1 root root  206888 May 19 03:33 core.img
-rwxr-xr-x 1 root root      24 Mar 31 20:19 default.cfg
-r-xr-xr-x 1 root root  115396 Mar 31 11:53 ldlinux.c32
-r-xr-xr-x 1 root root   69632 Mar 31 11:53 ldlinux.sys
-rwxr-xr-x 1 root root    1024 May 19 03:33 lnxboot.img
-rwxr-xr-x 1 root root     172 Mar 31 11:53 README
-rwxr-xr-x 1 root root     531 Mar 31 20:19 root.A.cfg
-rwxr-xr-x 1 root root     531 Mar 31 20:19 root.B.cfg
-rwxr-xr-x 1 root root     112 May 19 03:33 syslinux.cfg
-rwxr-xr-x 1 root root     887 Mar 31 20:19 usb.A.cfg
-rwxr-xr-x 1 root root 7708656 Mar 31 11:53 vmlinuz.A
-rwxr-xr-x 1 root root 7708656 Mar 31 11:53 vmlinuz.B
pmn@pmn-PC:~$

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
20/05/20,
10:08:05 pm

sebanc, 2020-05-20:
Everything seems good, looks like you bios does not like the syslinux mbr.

What are your laptop/pc specs ?

Maybe you could try the package below instead of mbr_support.tar.gz. It's the previous package before I did some cleanup in the code:
[testmbr7.zip](https://github.com/sebanc/brunch/files/4656773/testmbr7.zip)

Same principle, just you have to add "-m" argument when you run "chromeos-install.sh"

omminp, 2020-05-21:
G'day, 
laptop specs  
1. Intel core2 T5600  Intel graphics
2. Intel core2 Duo P9600 Intel graphcs
3. AMD A6

Don't expect this build to run on AMD but it does find the BIOS boot (and no UEFI) and gives the same boot fail message.

Build with testmbr.zip with -m resulted in successful build to USB,

USB failed to boot on all laptops, with the same message "Missing OS"  ,but differs from previous builds in that after the "Missing OS" message, it jumps straight to boot from system HDD.  Previously it would just hang after the message.

Each of these machines will boot linux from BIOS. Anything else I can look for?

sebanc, 2020-05-21:
Could you attach a picture of the missing OS screen ?

omminp, 2020-05-21:
[image]

omminp, 2020-05-21:
The cursor is blinking on the start of the next line down.

sebanc, 2020-05-21:
Could you provide the output of "cgpt show /dev/<your_flashdrive> ? What size is it ?

omminp, 2020-05-21:
[cgpt.txt](https://github.com/sebanc/brunch/files/4660834/cgpt.txt)
[sdb.txt](https://github.com/sebanc/brunch/files/4660835/sdb.txt)

The USB is 32GB Sandisk
Thanks for your patience and support.

sebanc, 2020-05-21:
Could you try to run the below command in a linux distro ?

sudo sgdisk -A 12:set:2 /dev/<your_flashdrive>

omminp, 2020-05-21:
Done, attached files show output of command and boot screen after.
[sgdisk.txt](https://github.com/sebanc/brunch/files/4662345/sgdisk.txt)
[image]

sebanc, 2020-05-21:
Could you try this one ?
[mbr_new.zip](https://github.com/sebanc/brunch/files/4662769/mbr_new.zip)

First without running "sudo sgdisk -A 12:set:2 /dev/<your_flashdrive>" and then with it if it does not work.

omminp, 2020-05-22:
no sgdisk - same 'Missing OS' message
after ssgdisk  see photo.  accepts keyboard input, but tried linuxboot , core , extlinux  all errors...
[image]

Z3r0shin, 2020-05-22:
Sorry for butting in, I am following similar issues as mine #63 .

I am having the same problems as @omminp with the same last result.
Using a i5 450M old laptop.

sebanc, 2020-05-22:
My mistake, could you mount partition 12 of your usb and rename /extlinux/extlinux.cfg to /extlinux/extlinux.conf and try to boot ?

EnzoRimapa, 2020-05-22:
> My mistake, could you mount partition 12 of your usb and rename /extlinux/extlinux.cfg to /extlinux/extlinux.conf and try to boot ?

Good afternoon, I have tried all the steps that have been mentioned on a Toshiba satellite laptop directly on the hard disk, but it always throws the same boot failure message that I can hardly see because it takes me to the boot menu.

EnzoRimapa, 2020-05-22:
> Could you try this one ?
> [mbr_new.zip](https://github.com/sebanc/brunch/files/4662769/mbr_new.zip)
> 
> First without running "sudo sgdisk -A 12:set:2 /dev/<your_flashdrive>" and then with it if it does not work.

I wanted to try this step but since the first one did not work I tried to format and use the command but it tells me that the USB is in read only and it does not allow me to format it apparently it has damaged the USB because not even in windows it lets me format it.

omminp, 2020-05-23:
Some test results.  Good news first.  Rename the file to extlinux.conf boots on my machines, after sgdisk to set boot flag.  The same can be achieved using gparted, set legacy_boot flag for sdx12.

After boot starts, on core 2 duo P9600 machine there is good progress with brunch construction of the system, then it starts throwing "traps: invalid opcode" errors and stops with a mesage "chromeos-boot-alert: self_repair"    see later fail screenshot.

On the other machine, core 2 T5600, it boots but fails early after "traps: invalid opcode" and falls back to a shell.

On the AMD machine, it boots and gets through to show the chrome splash screen - more than I expected.  

Is it worth trying a different recovery (now on samus), or is it more likely that the opcodes causing grief are the result of brunch code?   It is so close....
[image]
[image]

EnzoRimapa, 2020-05-23:
> Algunos resultados de las pruebas. Buenas noticias primero. Cambie el nombre del archivo a extlinux.conf boots en mis equipos, después de sgdisk para establecer la marca de arranque. Lo mismo se puede lograr usando gparted, establecer legacy_boot bandera para sdx12.
> 
> Después de que se inicia el arranque, en el núcleo 2 duo P9600 máquina hay un buen progreso con la construcción del brunch del sistema, entonces comienza a lanzar "trampas: código de operación no válido" errores y se detiene con un mensaje "chromeos-boot-alert: self_repair" ver más tarde fail captura de pantalla.
> 
> En la otra máquina, el núcleo 2 T5600, arranca pero falla temprano después de "trampas: opcode no válido" y cae de nuevo a un shell.
> 
> En la máquina AMD, arranca y se pone a través de mostrar la pantalla de bienvenida cromada - más de lo que esperaba.
> 
> ¿Vale la pena probar una recuperación diferente (ahora en samus), o es más probable que los códigos de operación que causan dolor sean el resultado del código de brunch? Está tan cerca....

I have a core two duo that also remains as the first photo on the screen doing the normal steps mentioned in the documentation.

sebanc, 2020-05-23:
Sorry but what was not normal and I wanted to troubleshoot in this issue is the mbr bios not booting, however core 2 duos and all cpu prior to Intel 1st gen (according to https://en.wikipedia.org/wiki/Intel_Core) will not boot as all current google recovery images are build with cpu instructions included which are not supported by those cpus.

Regarding, your AMD A6, it is close enough from the "grunt" recovery image devices that it might actually work (it won't work with other recovery images as AMD graphics are not supported on intel recovery images).
Could you try to boot the AMD device using the modified mbr support package below and with grunt recovery image ?
[mbr_new2.zip](https://github.com/sebanc/brunch/files/4671158/mbr_new2.zip)

omminp, 2020-05-26:
Sorry for the delayed reply. I did a build using grunt, and it worked as expected on the AMD machine using BIOS boot. The legacy-boot flag was set on partition 12 out of the box.  I am the one who reported success using grunt with AMD, so it was expected that the alternative boot method would run too, as long as the BIOS boot worked.
I didn't know that the Google recoveries used CPU instructions not valid for older CPUs. But it makes sense to optimize the code for the specific target machine when that is the only machine meant to run it.  After my good fortune in having success with AMD contrary to the limitations set for the project, I was bold enough to think the older CPUs could be made to work, notwithstanding the clear advice to the contrary. I fear I may have wasted your time chasing after BIOS boot, since machines with BIOS and no UEFI boot just won't run brunch.
The only idea that occurred to me was to use a "recovery" built using only CPU instructions valid for older CPUs.  Arnold came to mind, and also WayneOS which I had running on my older machines.  I was able to download a .bin file for WayneOS and use that with brunch.  It built the USB stick ok, and booted on the Core 2 duo.  But on the Core 2, it throws 'trap invalid opcode' errors when brunch is searching for chromeos on the partitions. So I guess the brunch code has some CPU instructions that are invalid on older machines as well.
Maybe there is something here that might help you dissuade others from trying the impossible.
Thanks for your help and best regards...

sebanc, 2020-05-26:
No worries, thanks for the update and for confirming that the AMD laptop works with brunch and grunt image.

Indeed your findings are correct, I will just clarify one point, brunch reuses some binaries from the image it was built with (currently samus which is the current most compatible image in terms of cpu instructions). For that reason the computers which do not support those "minimal" cpu instructions will fail at brunch stage and fallback to shell.

EnzoRimapa, 2020-05-28:
[image]
I also stayed in that problem but look for an alternative like cloudready_x64 and if it started on an old pc with an intel centrino duo CPU, I was testing the system and it is normal that it should be at brunch, does it not have that compatibility?

sebanc, 2020-05-28:
It looks perfectly normal for brunch with this cpu. The difference is:
- Cloudready builds chromiumos from source without specific cpu instructions for maximum compatibility.
- Brunch uses images prebuilt by google for specific devices which contain all the cpu instructions supported by the device for optimal performance.

There are currently no prebuilt google images compatible with your device cpu.

EnzoRimapa, 2020-05-28:
apparently this is why the greatest compatibility in any case the only option I have that for that device although cloudready does not have google apps do you know of any way to be able to make the facilities to have the play store please?

sebanc, 2020-05-28:
Fydeos is built from chromiumos source so it might work.

Else you can also directly install an android-x86 fork, someone on the Telegram group recently mentioned "PrimeOS Classic".

omminp, 2020-06-02:
Thanks for your support.  One last thing, I have for some reason been excluded from posting to the Telegram group. I can't find any way to contact the group to ask for review. Maybe you could get that fixed.   username @OldPierre .

closed 2020-06-02 by omminp

sebanc, 2020-06-02:
Just wrote to you on Telegram