sebanc/brunch #174: Error "Provided disk image size is not numeric"

Pascallec, 2020-05-20:
Hello
Thanks for your works

When I follow your tutorial; I arrive at the time when Chrome OS must install and there is an error "provided disk image size is not numeric"
How do I proceed?

thank you in advance

sebanc, 2020-05-20:
Hi, what are you typing after "-s" argument ?

sebanc, 2020-05-20:
Well, actually could you provide the complete command line you use to install ?

Pascallec, 2020-05-20:
I respect a tutorial found on youtube "https://www.youtube.com/watch?v=FEY9MtznYGU"

Pascallec, 2020-05-20:
this tutorial is based on your work.
In your database "https://sites.google.com/view/brunch-project" there is an error for "ERROR = SIZE_NOT_NUMERIC This error says that you have specified disk image size that is not numeric, eg 163.2 instead of 163. " but I don't understand how to solve it
thank you for your reply

sebanc, 2020-05-20:
Sorry but I am not related to those videos/tutorials, could you open an issue on the github account of the guy who made the video / script ?

Else you can follow the official instructions on the main page and re-open this issue if you trouble following them.

closed 2020-05-20 by sebanc