sebanc/brunch #176: Shutdowns suddenly on log screen

leosel, 2020-05-21:
After successful installation and login of gmail account and using the brunch_r81 patch to be able to update, then restart, but after turning it on back again Shutdowns suddenly on log screen. please help

sebanc, 2020-05-21:
Hi, what are your laptop/pc specs ? to which version of chromeos have you updated ?

leosel, 2020-05-21:
I'm using an acer spin 1 with model: sp111-31-c2w3, Celeron N3350, HD Graphics 500, 4GB ram, 32gb emmc. After installing chrome os v80 using brunch r80 without problems and installed some apps like youtube in playstore, then I attempted to enable the update to chrome os v81, so I install the brunch r81 then restart the chromebook, after the boot doing some update on the images text displaying on the screen, then finish. At boot and went on the Log In screen, it shuts off, tried to power on again, but after the log in screen it suddenly shuts off.

closed 2020-05-21 by leosel

sebanc, 2020-05-21:
There was an issue in brunch r80, try the same with brunch r81