sebanc/brunch #177: Screen resolution is wrong after updating.

theportal2, 2020-05-21:
I installed chrome os and everything was good but i was running chrome os 80 so i updated to 81.  After that my screen size is small, see photos [https://imgur.com/Vi7k2kI](url)
[https://imgur.com/ZduJBfs.](url) Any fix?

sebanc, 2020-05-21:
Hi,

With r81, google changed the shelf size, you can restore it with a flag I don't remember right now but that you will find easily on chromeos reddit.
Aside from that maybe google changed something with dpi calculation but it is not related to brunch. You can also use the resolution slider in display settings to adapt your screen size.

FaiekA, 2020-05-21:
Hi - I installed Chrome - everything's working great - thanks! - one issue i can't get a higher resolution then 1024×768 - running my PC with my TV so at that resolution it's not great.
Any suggestions will be appreciated 
Thanks

sebanc, 2020-05-21:
How is your pc connected to your tv ? hdmi or usb hub display port ?

FaiekA, 2020-05-21:
Hi - via VGA

sebanc, 2020-05-21:
Your graphic card is i915 ?

FaiekA, 2020-05-21:
Intel HD Graphics Family 
HD graphics 2500

theportal2, 2020-05-21:
i don't think i have resolution slider in display settings
[image]

FaiekA, 2020-05-21:
Thanks- I moved the slider to tiny - doesn't really make a differance.
- Windows is obviously fone 
- Linux I had the same problem used  Xrandr commands to fix that. 
Only chrome now - I know it would look great if I can get this sorted ....

FaiekA, 2020-05-21:
Your screen shot looks great 1980x1080 if I could only get there...

sebanc, 2020-05-22:
Could you try this new release ?
https://github.com/sebanc/brunch-testing/releases

FaiekA, 2020-05-22:
Ok - Thanks will do 
Will let you know once installed

FaiekA, 2020-05-22:
Hi - Thanks for the assistance
Still the same 1024×768 highest setting
( triple boot System ) - with win2grub
Procedure Followed 
Downloaded brunch-testing/release from link - copied in to Chrome OS folder on USB drive.
Ran in Linux Ubuntu DDE ..
Run multi_boot.sh and install as per Kedar instructions
Completed installations successfully. 
You think it could be my internal video card ?
I don't know -

sebanc, 2020-05-23:
Actually I did not really understand that there were 2 different reports in this issue and kind of mixed them both.

@theportal2 The slider can be seen on your printscreen (between Tiny / Huge settings, currently set on tiny), the native resolution indicates 1920x1080 is it correct ?

@FaiekA If it is the same with the linux kernel, could you try booting with "vga=ask" added to the kernel command line ? It should let you select the resolution from a list

FaiekA, 2020-05-23:
@sebanc it's my fault I should have opened a new issue -  instead of jumping in on thePortal2 thread --- Yes I hade the same issue with Ubuntu DDE and fixed it via xrandr.
If you could explain where and how to add this as I'm fairly new to this.
I would really appreciate it - again this is excellent - I just cant get enough of it ( hooking up with the rest of my google home stuff )

sebanc, 2020-05-23:
@FaiekA No worries, open crosh (CTRL+ALT+T), type "shell", run "sudo edit-grub-config" and add a space and "vga=ask" after cros_debug. Then CTRL+x, "y" to save and enter to confirm.

On next reboot, you should get a prompt to choose which resolution to use.

FaiekA, 2020-05-23:
@sebanc - Hi no luck - didnt prompt at all  - here's the code from crosh 
`menuentry "ChromeOS" {
  regexp --set disk "(^.+)(,gpt)" $root
  linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug vga=ask
  initrd ($disk,7)/initramfs.img
}
`

As well as a screen shot - is it normal for the display to be unknown .

[image]
 As you can see it just too big  my screens a  42"

FaiekA, 2020-05-23:
It might be time to invest in a graphics card and see what happens.

sebanc, 2020-05-24:
@FaiekA Could you try adding to the kernel command line "video=1920x1080-32" ?

FaiekA, 2020-05-24:
@sebanc  Ok that will be in the place of vga=ask ?

sebanc, 2020-05-24:
yes, exactly

FaiekA, 2020-05-24:
@sebanc will do - Thanks

FaiekA, 2020-05-27:
@sebanc - Hi been a bit busy - finally got it working with "video=1920x1080-32" 
Had to update the detail in the grub customiser app.
But thank a million for your assistance - keep up the great work, cant wait to see whats next.
[image]

Thank again

closed 2020-06-29 by sebanc