sebanc/brunch #178: remap function keys?

tungmeister, 2020-05-21:
@sebanc is there any way to remap the function keys so that they correspond with the correct keys?

sebanc, 2020-05-21:
No, sorry I don't know if it is possible

tungmeister, 2020-05-25:
No worries, it's the only thing that's slightly annoying - having to remember to press F3 for page refreshes and trying to remember which buttons correspond to brightness and volume. Are they hard coded?

kitakar5525, 2020-05-25:
I guess you mean this setting?
Settings -> Device -> Keyboard -> "Treat top-row keys as function keys"

sebanc, 2020-05-25:
Yes, it's hardcoded:
https://support.google.com/chromebook/answer/1047364?hl=en

Apparently though there is a setting to “Treat top-row keys as function keys”:
https://beebom.com/how-enable-function-keys-chromebook/

sebanc, 2020-05-25:
@kitakar5525 Thanks ! beat me to it

closed 2020-06-02 by sebanc