sebanc/brunch #180: Installation hangs during boot up.

weikevinhc, 2020-05-21:
Hi,
I followed the instructions and installed the rammus recovery onto a 64GB USB flash drive, with my XPS-13 (icelake i5 CPU, developer's edition). 
After selecting to boot with USB drive, the installation process seems to hang at the line:

    [120.074339]  input: HDA Intel PCH HDMI/DP,pca=3 as /devices/pci0000:00:1f.3/sound/card0/input20

Not sure if I am doing something wrong or there's a hardware incompatibility.

sebanc, 2020-05-22:
Hi,

Intel Iris Plus Graphics are not properly supported by the 4.19 kernel that brunch currently uses and I have some issues currently with kernel 5.4. However, Some work is being done to bring kernel 5.4 support in issue #18 (even though it is not regularly updated, there is significant work going on).

sebanc, 2020-05-22:
I invite you to follow and participate to issue #18 and hope that a kernel compatible with 10th gen graphics will be available soon.

closed 2020-05-22 by sebanc

weikevinhc, 2020-05-22:
Thanks a lot of the reply and this project in general! I will follow issue #18.