sebanc/brunch #183: Asus VivoTab 8 (M81C)

bustamantemarlon, 2020-05-21:
Hey,

This project is great. I already brought new life to a few of my devices using Brunch.

This tablet is running an Intel Atom Z3745 with the rammus recovery. Almost everything seems to work.
From what I tested these are what work:
- Touch screen
- Volume Rocker
- Power Button (Only works as power on/off)
- Sleep/Screen lock using tablet cover
- 2.4GHz WiFi
- Bluetooth

What doesn't work:
- 5GHz WiFi
- Auto rotation
- Camera

Auto rotation is probably the biggest issue for me. I can cope without the rest. 5GHz works on the latest Mint and Ubuntu versions.

Let me know if more information is needed. Thank you.

sebanc, 2020-05-22:
Hi, could you attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads directory)

bustamantemarlon, 2020-05-22:
Here you go.
[dmesg.txt](https://github.com/sebanc/brunch/files/4666159/dmesg.txt)

sebanc, 2020-05-22:
Sorry I forgot to mention that the dmesg should be taken after a fresh boot, could you attach another one ?

bustamantemarlon, 2020-05-22:
No problem. Thanks for the help.
[dmesg.txt](https://github.com/sebanc/brunch/files/4666263/dmesg.txt)

sebanc, 2020-05-22:
Thanks for the dmesg. Unfortunately the news are not so good.

ChromeOS only supports usb cameras for x86 devices so yours will not work. 

Regarding rotation, unfortunately your device does not use the common ISH hid sensors which are the only ones that have been ported (it uses InvenSense MPU-6050). Unless a dev with the right hardware ports those sensors to work with chromeos, there is nothing I can do.

The 5ghz issue is probably related to the firmware files I have added, could you try replacing both brcm/brcmfmac43340-sdio firmwares by the ones you use in linux ?

bustamantemarlon, 2020-05-22:
That's unfortunate. Luckily I don't use the webcam. But it's really unfortunate that the sensor for the rotation will not work for the moment. That is fine, I will just manual switch the rotation when needed.

As for the 5GHz, I tried replacing the brcmfmac43340-sdio.bin from the one used in Mint but that was not able to activate the WiFi band. These are the firmware files used in Mint.
[brcm_mint.txt](https://github.com/sebanc/brunch/files/4669434/brcm_mint.txt)

sebanc, 2020-05-23:
Could you try to replace the brcmfmac43340-sdio.txt in /lib/firmware/brcm/ with this one ?

[brcmfmac43340-sdio.txt](https://github.com/sebanc/brunch/files/4671428/brcmfmac43340-sdio.txt)

bustamantemarlon, 2020-05-23:
Thanks so much! That appears have have fixed it.

closed 2020-06-02 by sebanc