sebanc/brunch #184: Play store not opening

au79912, 2020-05-22:
Non of the google play stores apps are opening after the firsst boot
I deleted the google sync but its still not working
Any ideas on how to fix this?

au79912, 2020-05-22:
im using a dell lattitude E6440

sebanc, 2020-05-22:
Hi,

Are they working correctly on first boot and not after or are they not even opening on first boot ?

au79912, 2020-05-22:
it was working fine on the first boot. after shutting down and rebooting its stuck on loading

sebanc, 2020-05-22:
Your issue seems similar to #74, could you try letting the system rest 2 minutes on the login screen before logging ?

Narayanbhat166, 2020-05-22:
Same issue here, but it's not working on the first boot itself. It says setting up Google Play and then fails

au79912, 2020-05-22:
> Your issue seems similar to #74, could you try letting the system rest 2 minutes on the login screen before logging ?

waiting 2 mins after booting worked
thanks!

sebanc, 2020-05-22:
@Narayanbhat166 Could you try deleting google sync ?
https://support.google.com/chromebook/answer/2914794?hl=en

Also, what are your laptop/pc specs ?

Narayanbhat166, 2020-05-22:
Also there are frequent lags, it becomes unresponsive for about 30 seconds. Iam unable to install linux virtual machine on it

Laptop specs:
Intel i5 4th gen
Rammus recovery release 80

sebanc, 2020-05-22:
Try deleting google sync and updating to r81 stable brunch if you are not on it. Also, do you have a dual gpu ?

closed 2020-05-23 by au79912