sebanc/brunch #186: HP Spectre X2 tablet/laptop support

tornadox, 2020-05-22:
Hi **sebanc**! First of all thanks for this project and guide, it's very user friendly (found it on youtube) and simple!
So I've installed it on my HP Spectre X2 table/laptop. It has a i5-4202y with Intel HD Graphics 4200. Everything runs fine except a few issues: 
1) when detaching the keyboard the OS doesn't care. Fix is to enable on-screen keyboard. I saw there is a boot argument, would that fix the issue?
2) probably related to 1st issue, no rotation, in some android games it's possible to use the gyroscope for steering.
3) graphics artefacts in some android games like Asphalt 9. In some games there is a yelow 1pixel boarder.
Just FYI, I've tried a precompiled image from 4pda, I think it was based on eve-caroline recovery + chrome 70 or 74, graphics issues were also present there with the same games.
4) sound is low - purely related to linux kernel as on windows there is a special beats driver that increases the volume. Everything works fine, except that the batter doesn't show ETA time.

midi1996, 2020-05-22:
1) check the options in the readme (scroll down) and check #154 (you want to read and understand how udev rules work and make your own rules and apply them)
2) that just means gyro is working but tablet mode isnt set up (check 1)
3) your CPU's iGPU is fairly slow (and low power). try enabling csm/legacy in your uefi and see if that helps (not expecting anything amazing but worth the try). And please do not use random images you find on the web, it's risky and who knows what could be in there.
4) if it's on all linux distros then thats just sad
5) you may want to edit your acpi and load it to linux to fix this, on some models, hp chose to remove battery time calculation from the acpi. Also try checking your firmware setup if it has "Battery Reading" or some related name and either disable or enable it.

good luck.

sebanc, 2020-05-23:
@midi1996 Thanks for the help !

@tornadox I maintain a list of detachable keyboard ids so that tablet mode can be applied automatically when using tablet mode options, could you post the output of "dmesg | grep input" just after a reboot and with the keyboard attached.

tornadox, 2020-05-23:
@sebanc thanks for the reply, here's the output. I also noticed that there is no reaction for the power button (I do know how it works on my HP x360 14") and wireless button.
`[    0.685529] input: Lid Switch as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0D:00/input/input0
[    0.685665] input: Power Button as /devices/LNXSYSTM:00/LNXPWRBN:00/input/input1
[    0.852021] input: AT Translated Set 2 keyboard as /devices/platform/i8042/serio0/input/input2
[    1.644270] hid-generic 0003:0483:91D1.0001: hiddev0,hidraw0: USB HID v1.10 Device [STMicroelectronics ST_SENSOR_HUB] on usb-0000:00:1d.0-1.1/input0
[    1.970042] input: Synaptics Touch Pad V 103u9 Mouse as /devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:06CB:5711.0002/input/input3
[    1.972612] hid-generic 0003:06CB:5711.0002: input,hiddev1,hidraw1: USB HID v1.11 Mouse [Synaptics Touch Pad V 103u9] on usb-0000:00:1d.0-1.6/input0
[    2.183778] input: HP TrueVision Full HD: HP TrueV as /devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/input/input5
[    2.352960] input: ELAN Touchscreen as /devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.8/1-1.8:1.0/0003:04F3:0217.0003/input/input6
[    2.355733] hid-multitouch 0003:04F3:0217.0003: input,hiddev2,hidraw2: USB HID v1.10 Device [ELAN Touchscreen] on usb-0000:00:1d.0-1.8/input0
[   24.736985] input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input10
[   26.246972] snd_hda_codec_idt hdaudioC1D0:    inputs:
[   26.253114] input: HDA Intel HDMI HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:03.0/sound/card0/input12
[   26.253390] input: HDA Intel HDMI HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:03.0/sound/card0/input13
[   26.253637] input: HDA Intel HDMI HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:03.0/sound/card0/input14
[   26.253892] input: HDA Intel HDMI HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:03.0/sound/card0/input15
[   26.254181] input: HDA Intel HDMI HDMI/DP,pcm=10 as /devices/pci0000:00/0000:00:03.0/sound/card0/input16
[   26.298536] input: HDA Intel PCH Mic as /devices/pci0000:00/0000:00:1b.0/sound/card1/input17
[   26.299080] input: HDA Intel PCH Front Mic as /devices/pci0000:00/0000:00:1b.0/sound/card1/input18
[   26.299322] input: HDA Intel PCH Headphone as /devices/pci0000:00/0000:00:1b.0/sound/card1/input19
[   26.299659] input: HDA Intel PCH Front Headphone as /devices/pci0000:00/0000:00:1b.0/sound/card1/input20
[   26.577958] input: Synaptics s3203 as /devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:06CB:5711.0002/input/input11
[   26.593620] hid-rmi 0003:06CB:5711.0002: input,hiddev0,hidraw0: USB HID v1.11 Mouse [Synaptics Touch Pad V 103u9] on usb-0000:00:1d.0-1.6/input0
[   28.242741] input: HP Wireless hotkeys as /devices/virtual/input/input21
[   29.441687] input: HP WMI hotkeys as /devices/virtual/input/input22`

@midi1996 thanks for the reply, enabling CSM didn't change anything. I think the graphics issue is related specifically to the game and video driver as I works on windows on this tablet - so let's just ignore that. However I connected to an external monitor with 2560x1080 and max resolution was 1920x1080 - but I seen that AMD opensource drives also strugles with this resolution.
The tablet has two batteries, one in the tablet, other in the keyboard which might be the reason. Linux itself correctly sees both batteries and their charge levels however I don't remember seeing charge ETA even in Windows and I don't use with batteries at home=) Not sure if I would want to play around with acpi, the batteries are quite old.

sebanc, 2020-05-24:
@tornadox What is the output of "lsmod | grep acer" in crosh shell ?

sebanc, 2020-05-24:
You can also try the build here:
https://github.com/sebanc/brunch-testing/releases

Tablet mode should work if you add "options=force_tablet_mode" to the kernel command line.

tornadox, 2020-05-27:
Thanks @sebanc I will check, what is the best way to do so?
sudo chromeos-update -f < path to the Brunch release archive > - is this enough or should I better install from scratch?

sebanc, 2020-05-27:
> Thanks @sebanc I will check, what is the best way to do so?
> sudo chromeos-update -f < path to the Brunch release archive > - is this enough or should I better install from scratch?

Using chromeos-update should be enough

> 
> > What is the output of "lsmod | grep acer" in crosh shell ?
> 
> on the latest release it's empty

Ok, that's what i wanted to verify

tornadox, 2020-05-27:
Works like a charm with Brunch r81 k4.19 testing 20200526 + options=force_tablet_mode
immediately when I disconnect the keyboard it enters tablet mode and orientation also works!
Is it possible to get the power off/sleep button to get fixed?

sebanc, 2020-05-27:
try "options=force_tablet_mode,acpi_power_button"

tornadox, 2020-05-27:
It works!

tornadox, 2020-05-27:
I think that the boot process got longer on the new testing package, but maybe it's related to amout of apps I installed.

midi1996, 2020-05-29:
i think `no_ui_delay` should help if you update to the latest release (now that it is packing everything from the testing branch). If you happen to have android issues, then remove that option and you'll have to live with that extra delay.

sebanc, 2020-05-29:
Thanks @midi1996 Indeed the boot delay got longer with latest release to fix a global module loading issue. You can bypass it by using "options=no_ui_delay".

tornadox, 2020-05-29:
Can you give some example of issue with android apps so I can understand better? For example I have black graphics in resource hungry games like asphalt 9. Other apps that I start on Spectre with brunch run very similar compared to a chromebook with the performance differnce in hardware (HP Chromebook has an i3-8130u which is faster)
Also in some other games there are some graphical issues with violet color (perhaps violet is a result of non-working transparency or so) but this is not related to brunch as I have them on a real chromebook with intel graphics.

sebanc, 2020-05-29:
No, it is not that, the issue is android apps not working, not issues within android apps.

Usually the cause of black screen in apps/games is the absence of "drm" due to brunch starting in dev mode.
You can try to boot without dev mode but it often fails due to selinux issues. Also if non-dev mode works for you, you will then need to powerwash to get the drm working.

Regarding chromebooks android compatibility issues, there is not much I can do.

tornadox, 2020-05-29:
I will try, thanks for the explanation

tornadox, 2020-05-31:
I have a question regarding separate battery charge data showing, as this laptop/tablet has two batteries I get this:
```
chronos@localhost / $ cat /sys/class/power_supply/BAT0/uevent 
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_STATUS=Charging
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11100000
POWER_SUPPLY_VOLTAGE_NOW=12495000
POWER_SUPPLY_POWER_NOW=0
POWER_SUPPLY_ENERGY_FULL_DESIGN=30336000
POWER_SUPPLY_ENERGY_FULL=30336000
POWER_SUPPLY_ENERGY_NOW=28726000
POWER_SUPPLY_CAPACITY=94
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_MODEL_NAME=TR03033XL
POWER_SUPPLY_MANUFACTURER=13-2B
POWER_SUPPLY_SERIAL_NUMBER=42384
chronos@localhost / $ cat /sys/class/power_supply/BAT1/uevent 
POWER_SUPPLY_NAME=BAT1
POWER_SUPPLY_STATUS=Unknown
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11100000
POWER_SUPPLY_VOLTAGE_NOW=12529000
POWER_SUPPLY_POWER_NOW=0
POWER_SUPPLY_ENERGY_FULL_DESIGN=17634000
POWER_SUPPLY_ENERGY_FULL=17634000
POWER_SUPPLY_ENERGY_NOW=16161000
POWER_SUPPLY_CAPACITY=91
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_MODEL_NAME=IEP001
POWER_SUPPLY_MANUFACTURER=13-17
POWER_SUPPLY_SERIAL_NUMBER=  597
```
As I understand chrome os shows a summary of both batteries, is it possible to show them separately in chrome os?

sebanc, 2020-05-31:
Nice to see that the latest build fixed some issues.

Unfortunately, for the battery indicator the only options are to show only the main battery or to have an average of multiple batteries.

closed 2020-06-02 by sebanc