sebanc/brunch #187: Boot Failed. Support MRB/BIOS

EnzoRimapa, 2020-05-22:
hello good afternoon I have a problem with the latest support for mbr when installing it on a Toshiba satellite laptop with corei3-350M everything indicates that it is installed correctly from linux mint, I restart it but when it starts it throws me a boot error along with the name of the device redirecting me to the boot menu, I also tried installing from USB but I get the same problem.
I attach some photos.
[image]

midi1996, 2020-05-22:
did you follow that yt guide or the instructions in this repo?

EnzoRimapa, 2020-05-22:
actually guided me from this  [#77 ](https://github.com/sebanc/brunch/issues/77)
and from [#173 ](https://github.com/sebanc/brunch/issues/173)
following all the steps but still throws the same error.

EnzoRimapa, 2020-05-22:
@midi1996 in case it is important to do tests, I have erased everything from the disk from a windows installer leaving the hdd without assigned partitions and it throws me the same error so what I think is that the grub or some boot file is not being recognized?

midi1996, 2020-05-22:
🤔 i installed it on an sdcard for a legacy system and it didnt install grub properly, for that i had to boot to some linux distro (in this case ubuntu 20.04), mounted the EFI partition (i guess it is partition 12, check for that) and ran:
```
grub-install --recheck --target=i386-pc --boot-directory=/path/to/mounted/part /dev/<your_disk_id>
```
for example the efi partition is mounted in /mnt and my target disk sda
```
grub-install --recheck --target=i386-pc --boot-directory=/mnt /dev/sda
```
Once done, you will find grub installed in /mnt/boot/grub, you can copy over the grub.cfg from /mnt/efi/boot/ (which is the original path for the grub configuration from the release archives). You can also try `--boot-directory=/mnt/efi` but idk what that could yield. Once copied you can open it and configure it if you need to and then try booting.

EnzoRimapa, 2020-05-22:
@midi1996 I get that message as an error, why? I am using a USB live because I have the installation in my hdd.

<pre>Mounted /dev/sda12 at /media/root/EFI-SYSTEM.
root@mint:/home/mint# grub-install --recheck --target=i386-pc --boot-directory=/media/root/EFI-SYSTEM /dev/sda
Installing for i386-pc platform.
grub-install: warning: this GPT partition label contains no BIOS Boot Partition; embedding won&apos;t be possible.
grub-install: warning: Embedding is not possible.  GRUB can only be installed in this setup by using blocklists.  However, blocklists are UNRELIABLE and their use is discouraged..
grub-install: error: will not proceed with blocklists.
root@mint:/home/mint# 
</pre>

midi1996, 2020-05-22:
try using `--root-directory` instead

EnzoRimapa, 2020-05-22:
now give me this message
<pre><font color="#8AE234"><b>mint@mint</b></font>:<font color="#729FCF"><b>~</b></font>$ grub-install --recheck --target=i386-pc --root-directory=/mnt /dev/sda
Installing for i386-pc platform.
grub-install: error: failed to get canonical path of `/mnt/boot/grub&apos;.
<font color="#8AE234"><b>mint@mint</b></font>:<font color="#729FCF"><b>~</b></font>$ 
</pre>

midi1996, 2020-05-23:
is the efi partition really mounted?

EnzoRimapa, 2020-05-23:
if it is mounted in / mnt and there is this in the /efi/boot/grub.cfg folder

midi1996, 2020-05-23:
try adding `--force`

EnzoRimapa, 2020-05-23:
<pre>root@mint:/home/mint# grub-install --recheck --target=i386-pc --root-directory=/mnt /dev/sda --force
Installing for i386-pc platform.
grub-install: warning: this GPT partition label contains no BIOS Boot Partition; embedding won&apos;t be possible.
grub-install: warning: Embedding is not possible.  GRUB can only be installed in this setup by using blocklists.  However, blocklists are UNRELIABLE and their use is discouraged..
Installation finished. No error reported.
root@mint:/home/mint# 
</pre>

sebanc, 2020-05-23:
@EnzoRimapa Your issue should be solved by using the mbr_new2.zip package I have just posted in the issue #173

EnzoRimapa, 2020-05-23:
@sebanc would I do it with samus or with grunt?

EnzoRimapa, 2020-05-23:
<pre><font color="#8AE234"><b>mint@mint</b></font>:<font color="#729FCF"><b>/cdrom/chrome os antiguo</b></font>$ time sudo bash chromeos-install.sh -src chromeos_12739.111.0_samus_recovery_stable-channel_mp-v3.bin -dst /dev/sda
All data on device /dev/sda will be lost, are you sure ? (type yes to continue) yes
umount: /dev/sda: not mounted.
umount: /dev/sda1: not mounted.
umount: /dev/sda10: not mounted.
umount: /dev/sda11: not mounted.
umount: /dev/sda12: not mounted.
umount: /dev/sda2: not mounted.
umount: /dev/sda3: not mounted.
umount: /dev/sda4: not mounted.
umount: /dev/sda5: not mounted.
umount: /dev/sda6: not mounted.
umount: /dev/sda7: not mounted.
umount: /dev/sda8: not mounted.
umount: /dev/sda9: not mounted.
B4F5A273-A3FF-F948-917B-0C8BA3C9C4F7
The operation has completed successfully.
       start        size    part  contents
           0           1          PMBR (Boot GUID: B4F5A273-A3FF-F948-917B-0C8BA3C9C4F7)
           1           1          Pri GPT header
           2          32          Pri GPT table
    19255304   957517816       1  Label: &quot;STATE&quot;
                                  Type: Linux data
                                  UUID: DE3A61A3-9558-5144-A85E-761A06F2E8E4
     2117638       65536       2  Label: &quot;KERN-A&quot;
                                  Type: ChromeOS kernel
                                  UUID: E4406B1B-3CCE-484E-8984-8D6F40D4A5A4
                                  Attr: priority=15 tries=15 successful=0 
    10866696     8388608       3  Label: &quot;ROOT-A&quot;
                                  Type: ChromeOS rootfs
                                  UUID: 26889899-1D2E-8148-90D7-045E3B59B85D
     2183174       65536       4  Label: &quot;KERN-B&quot;
                                  Type: ChromeOS kernel
                                  UUID: 70D2AC40-7B8C-DD47-8C00-E6A7F42E5184
                                  Attr: priority=0 tries=15 successful=0 
     2478088     8388608       5  Label: &quot;ROOT-B&quot;
                                  Type: ChromeOS rootfs
                                  UUID: 98F02E52-8021-1B48-8590-86A2A7E362E1
       16448           1       6  Label: &quot;KERN-C&quot;
                                  Type: ChromeOS kernel
                                  UUID: 787D4826-1907-3049-A6FF-E06CA2E4D6FB
                                  Attr: priority=0 tries=15 successful=0 
       16456     2097152       7  Label: &quot;ROOT-C&quot;
                                  Type: ChromeOS rootfs
                                  UUID: 866561E4-C6C8-C84F-B8A0-61CFE9A37445
     2248712       32768       8  Label: &quot;OEM&quot;
                                  Type: Linux data
                                  UUID: 8842917D-DF86-BB43-834E-03DD5B8C2531
     2113608           1       9  Label: &quot;reserved&quot;
                                  Type: ChromeOS reserved
                                  UUID: B34C2008-E9A0-FD46-9754-64E0E5F67F96
     2113609           1      10  Label: &quot;reserved&quot;
                                  Type: ChromeOS reserved
                                  UUID: 5570C0F0-086F-D349-83A6-9E73689478D5
          64       16384      11  Label: &quot;RWFW&quot;
                                  Type: ChromeOS firmware
                                  UUID: 3872C669-BC05-584F-9C7F-8CB37F061A69
     2412552       65536      12  Label: &quot;EFI-SYSTEM&quot;
                                  Type: EFI System Partition
                                  UUID: B4F5A273-A3FF-F948-917B-0C8BA3C9C4F7
                                  Attr: legacy_boot=1 
   976773135          32          Sec GPT table
   976773167           1          Sec GPT header
Writing partition 1
10.0MiB 0:00:00 [19.2MiB/s] [================================&gt;] 100%            
Writing partition 2
16.0MiB 0:00:00 [18.2MiB/s] [================================&gt;] 100%            
Writing partition 3
2.00GiB 0:01:58 [17.3MiB/s] [================================&gt;] 100%            
Writing partition 4
16.0MiB 0:00:00 [17.7MiB/s] [================================&gt;] 100%            
Writing partition 5
2.00GiB 0:01:55 [17.7MiB/s] [================================&gt;] 100%            
Writing partition 6
 512 B 0:00:00 [ 440KiB/s] [=================================&gt;] 100%            
Writing partition 7
1.00GiB 0:00:56 [18.0MiB/s] [================================&gt;] 100%            
Writing partition 8
16.0MiB 0:00:00 [18.0MiB/s] [================================&gt;] 100%            
Writing partition 9
 512 B 0:00:00 [1.38MiB/s] [=================================&gt;] 100%            
Writing partition 10
 512 B 0:00:00 [2.05MiB/s] [=================================&gt;] 100%            
Writing partition 11
8.00MiB 0:00:00 [18.5MiB/s] [================================&gt;] 100%            
Writing partition 12
32.0MiB 0:00:01 [18.7MiB/s] [================================&gt;] 100%            
ChromeOS installed.

real	5m23.017s
user	0m0.555s
sys	0m12.307s
<font color="#8AE234"><b>mint@mint</b></font>:<font color="#729FCF"><b>/cdrom/chrome os antiguo</b></font>$ </pre>

sebanc, 2020-05-23:
Could you take a photo of the boot message and attach it here ?

EnzoRimapa, 2020-05-23:
[image]

it throws that message at me for a second and it returns me to the boot menu, I tried deleting the disk with a windows installer and it throws me the same message so I think it does not recognize the starter

sebanc, 2020-05-23:
The previous version was a bit of a hack but in this one I properly reinstalled the latest syslinux and added a compatibility flag (not conform to gpt specifications but needed by some bios). I don't really see what could be the issue now...

If you write a linux live on the same usb flashdrive does it boot ? Could you try it on a usb flashdrive ?

EnzoRimapa, 2020-05-23:
Do you know what this is all about? I have tried to write it and tbm format it from windows but I still get those errors on the USB

<pre><font color="#8AE234"><b>mint@mint</b></font>:<font color="#729FCF"><b>/cdrom/ChromeOs antiguo</b></font>$ time sudo bash chromeos-install.sh -src chromeos_12739.111.0_samus_recovery_stable-channel_mp-v3.bin -dst /dev/sdb
All data on device /dev/sdb will be lost, are you sure ? (type yes to continue) yes
umount: /dev/sdb: not mounted.
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
WARNING: One of the GPT headers/entries is invalid

ERROR: please run &apos;cgpt repair&apos; before adding anything.
WARNING: One of the GPT headers/entries is invalid

ERROR: please run &apos;cgpt repair&apos; before adding anything.
WARNING: One of the GPT headers/entries is invalid

ERROR: please run &apos;cgpt repair&apos; before adding anything.
WARNING: One of the GPT headers/entries is invalid

ERROR: please run &apos;cgpt repair&apos; before adding anything.
WARNING: One of the GPT headers/entries is invalid

ERROR: please run &apos;cgpt repair&apos; before adding anything.
WARNING: One of the GPT headers/entries is invalid

ERROR: please run &apos;cgpt repair&apos; before adding anything.
WARNING: One of the GPT headers/entries is invalid

ERROR: please run &apos;cgpt repair&apos; before adding anything.
00000000-0000-0000-0000-000000000000
WARNING: One of the GPT headers/entries is invalid

ERROR: please run &apos;cgpt repair&apos; before adding anything.
Warning! Main and backup partition tables differ! Use the &apos;c&apos; and &apos;e&apos; options
on the recovery &amp; transformation menu to examine the two tables.

Warning! One or more CRCs don&apos;t match. You should repair the disk!

****************************************************************************
Caution: Found protective or hybrid MBR and corrupt GPT. Using GPT, but disk
verification and recovery are STRONGLY recommended.
****************************************************************************
The operation has completed successfully.
       start        size    part  contents
           0           1          PMBR
           1           1          Pri GPT header
           2          32          Pri GPT table
     2117638       65536       2  Label: &quot;KERN-A&quot;
                                  Type: ChromeOS kernel
                                  UUID: 7C33C952-C53F-C348-8374-1E3A48088A9A
                                  Attr: priority=0 tries=0 successful=0 
     2183174       65536       4  Label: &quot;KERN-B&quot;
                                  Type: ChromeOS kernel
                                  UUID: 715DB847-0A9C-D24A-B0F3-A8D9F2CF8FF3
                                  Attr: priority=0 tries=0 successful=0 
       16448           1       6  Label: &quot;KERN-C&quot;
                                  Type: ChromeOS kernel
                                  UUID: 3F6BCCFA-170A-0548-BB20-3EFB458DF36F
                                  Attr: priority=0 tries=0 successful=0 
       16456     2097152       7  Label: &quot;ROOT-C&quot;
                                  Type: ChromeOS rootfs
                                  UUID: 69D0A2BD-67E0-A443-B8D7-249392DEF86B
     2248712       32768       8  Label: &quot;OEM&quot;
                                  Type: Linux data
                                  UUID: 8E6D4FBA-18C0-1B4C-ABDF-3D6AE02BE77B
     2113608           1       9  Label: &quot;reserved&quot;
                                  Type: ChromeOS reserved
                                  UUID: 943BEC09-007C-B149-8FB2-80450ECC33A2
     2113609           1      10  Label: &quot;reserved&quot;
                                  Type: ChromeOS reserved
                                  UUID: 240682B5-EE06-4E48-BB43-A8BCB6C94106
          64       16384      11  Label: &quot;RWFW&quot;
                                  Type: ChromeOS firmware
                                  UUID: E24D209D-21F3-CE45-9DF5-DA76171F6022
    65535967          32 INVALID  Sec GPT table
           1           1 INVALID  Sec GPT header
WARNING: One of the GPT headers/entries is invalid

Writing partition 1
10.0MiB 0:00:00 [16.8MiB/s] [================================&gt;] 100%            
Writing partition 2
16.0MiB 0:00:01 [15.9MiB/s] [================================&gt;] 100%            
Writing partition 3
1.79GiB 0:02:01 [15.1MiB/s] [=============================&gt;    ] 89% ETA 0:00:14
pv: write failed: No space left on device
Writing partition 4
16.0MiB 0:00:01 [15.1MiB/s] [================================&gt;] 100%            
Writing partition 5
pv: write failed: No space left on device
Writing partition 6
 512 B 0:00:00 [ 426KiB/s] [=================================&gt;] 100%            
Writing partition 7
1.00GiB 0:01:35 [10.8MiB/s] [================================&gt;] 100%            
Writing partition 8
16.0MiB 0:00:01 [15.3MiB/s] [================================&gt;] 100%            
Writing partition 9
 512 B 0:00:00 [ 413KiB/s] [=================================&gt;] 100%            
Writing partition 10
 512 B 0:00:00 [2.65MiB/s] [=================================&gt;] 100%            
Writing partition 11
8.00MiB 0:00:00 [16.5MiB/s] [================================&gt;] 100%            
Writing partition 12
pv: write failed: No space left on device
ChromeOS installed.

real	4m52.282s
user	0m0.327s
sys	0m7.156s
<font color="#8AE234"><b>mint@mint</b></font>:<font color="#729FCF"><b>/cdrom/ChromeOs antiguo</b></font>$ 
</pre>

sebanc, 2020-05-24:
The cgpt warnings are due to the previous write of primary / secondary headers gpt from windows and should have been corrected now that the script has been launched from linux.

Despite what it says, the second issue is not related to the lack of space. Those are write errors which occur for a reason I have not yet been able to identify, It has notably be discussed in issue #111 and different fixes have been found:
- Removing all partitions in gparted or windows, creating a single ext4 partition, rebooting and launching the script again
- Using another usb live distro to launch the chromeos-install script

Z3r0shin, 2020-05-24:
@sebanc , doing the fixes you mentionned worked! I booted into Brunch for the first time on my i5 450m laptop.

I still have some issues though :
-I can't access the chrome://flags. I mean, it stays blank and won't load the flags, even if I powerwash, remove/add developer mode and open the flags in incognito mode. It does show when using the guest sign in, but obviously no changes made that way stays. I read elsewhere that it may be due to my build (samus) not supporting TPM and therefore not making flags available.
-I can't install the linux container. It tries, but doesn't complete, saying there's an error. Tried to install it while having my internet plugged in to remove the chance of it being my wi-fi the culprit.

How can I help?

sebanc, 2020-05-25:
@Z3r0shin Nice that you could boot :)

Samus uses TPM1.2 which is not properly emulated  so it has 2 known issues (and workarounds):
- flags can be changed but only from guest mode.
- update channel has to be changed from command line.

Regarding crostini, could you remove it, reboot and launch a new install. After it fails could you run "cp /var/log/messages ~/Downloads/messages.txt" and attach messages.txt from your downloads folder to this issue ?

Z3r0shin, 2020-05-25:
Here's the output!
[messages.txt](https://github.com/sebanc/brunch/files/4677051/messages.txt)

Regarding the TPM, what can be done? Update channel, I don't mind since I will do it manually with the method you listed. Can I command-line this thing? Or Something more permanent?

sebanc, 2020-05-25:
Apparently crostini fails as it cannot enable gpu acceleration, could you use guest mode to disable crostini gpu acceleration in chrome://flags and try again ?

I looked at the tpm issue but did not manage to fix it, nevertheless ownership works in dev branch if you really need it.

Z3r0shin, 2020-05-25:
[messages.txt](https://github.com/sebanc/brunch/files/4677823/messages.txt)
Still does not work. Here is an updated messages.txt.
Does changing the flags in guest mode stays when I go back to my normal account? I thought that everything done in guest mode stays only in guest mode for that session only.

Also, what is the command to change to dev channel? I've tried "update engine client --channel=dev-channel --update" and "channel_update -dev" but none are current commands.

sebanc, 2020-05-25:
> Does changing the flags in guest mode stays when I go back to my normal account? I thought that everything done in guest mode stays only in guest mode for that session only.
That's a good question, i never tried myself but from what I have heard it changes for all users (but that may depends on the concerned setting).

In this case, the issue is still exactly the same, try updating to r84 with "sudo update_engine_client --channel=dev-channel -update", deactivate crostini gpu acceleration in chrome://flags and try crostini.

Z3r0shin, 2020-05-25:
When trying to update, I get this line : 

> crosh> shell
> chronos@localhost / $ sudo update_engine_client --channel=dev-channel -update
> [0525/114745.045065:INFO:update_engine_client.cc(422)] Channel permanently set to: dev-channel
> [0525/114745.045191:INFO:update_engine_client.cc(472)] Forcing an update by setting app_version to ForcedUpdate.
> [0525/114745.045209:INFO:update_engine_client.cc(474)] Initiating update check.
> [0525/114745.045983:INFO:update_engine_client.cc(503)] Waiting for update to complete.
> [0525/114815.159404:ERROR:update_engine_client.cc(204)] Update failed, current operation is UPDATE_STATUS_IDLE, last error code is ErrorCode::kOmahaErrorInHTTPResponse(37)
> chronos@localhost / $

sebanc, 2020-05-25:
You have to enable updates first by adding "options=enable_updates" to the kernel command line.

Run "sudo edit-grub-config" in crosh shell and add "options=enable_updates" after "cros_debug".

Z3r0shin, 2020-05-25:
Well, the dev channel makes it impossible to use my laptop. I can open tabs but not close them (if I do, my laptop freezes). I can open apps too, but not close them. I can't access my app drawer for some reason. BUT, crostini installed. Too bad I can't access it lol.
Do you think it is only due to the dev channel or something you can fix?
Also, I've just noticed this, but my trackpad doesn't work. Didn't try anything to fix this yet, if you have an idea on why it doesn't.

Z3r0shin, 2020-05-25:
Beta channel does the same thing, without the access to flags.

sebanc, 2020-05-26:
Ok, let's try something else, 1st install chromeos with "lulu" as recovery image, it might fix the touchpad. (if not there are other things to try but better be on lulu first).

You will have the same ownership issue but I just remembered another way to disable crostini acceleration:
Remove the line "virtio_gpu" from /etc/ui_use_flags.txt

Z3r0shin, 2020-05-26:
Lulu doesn't fix the trackpad. Is there is other recoveries to try? I will try them one at a time.

I opened the ui_use_flags.txt, but there is no "virtio_gpu" in it. See attached photo.
[image]

sebanc, 2020-05-26:
Other recoveries won't help further for the touchpad, it's just that someone had a specific touchpad issue with samus and I wanted to verify first it was not the same case. 

Could you add the below options in the first section of /etc/gesture/40-touchpad-cmt.conf ?
```
Option          "Touchpad Stack Version" "1"
Option          "Horizontal Resolution" "33"
Option          "Vertical Resolution" "33"
Option          "Integrated Touchpad" "1"
Option          "Input Queue Delay" "0.017"
Option          "Two Finger Vertical Close Distance Thresh" "35.0"
Option          "Fling Buffer Suppress Zero Length Scrolls" "0"
```

Regarding crostini, is it present on your chromeos install ? I think it has only been added to lulu image with r81 so you might need to update first.

Z3r0shin, 2020-05-26:
So, I've tried updating Lulu to beta channel (since that, indeed, Crostini wasn't present), but the same problems plaguing Samus' beta and dev channels kept happening (can't open launcher, can't close browser/tabs, etc.). I reinstalled Samus instead and removed the virtio_gpu thingie. Crostini installed and works well so far. You think I can re-instate virtio_gpu so I can have gpu acceleration in Crostini?
The trackpad hack doesn't work though. I did put it in the first section of the conf file, at the end of the options that were there already.

Z3r0shin, 2020-05-26:
It seems the container crashes if virtio_gpu is on. That sucks.
Would there be a way to enable it/make me old i5 450m work more for the non-gpu-intensive games I want to play on it? I am not even looking at Minecraft, more in the lines of GOG games (the real old games) and such, but it lags so much. The laptop could easily play those games in Ubuntu and Windows 10 before, so I guess there is something we can do.

sebanc, 2020-05-26:
The lag you encounter is more related to the fact that crostini is slow, even with gpu acceleration on. Try crouton instead.

For the trackpad, could you describe exactly the behavior ? Is the mouse pointer moving ? can you click things ?
Also, could you try booting with "pci=noacpi" added to the kernel command line ?

Z3r0shin, 2020-05-27:
For the trackpad, just nothing comes out of it. No movement, no clicks are registered, simply nothing.
If by kernel command line you mean adding it to the grub config file, it changed nothing. If not, then please point me where since I am a very enthousiatic person with bare basic knowledge of what I am doing.

EnzoRimapa, 2020-05-27:
@sebanc @midi1996  How do you know what this message is about? and how could I solve it? Maybe that's why you don't access the grub.

<pre>root@ubuntu:/home/ubuntu# mount /dev/sda12 /mnt
root@ubuntu:/home/ubuntu# grub-install --recheck --root-directory=/mnt /dev/sda --force
Installing for i386-pc platform.
grub-install: warning: this GPT partition label contains no BIOS Boot Partition; embedding won&apos;t be possible.
grub-install: warning: Embedding is not possible.  GRUB can only be installed in this setup by using blocklists.  However, blocklists are UNRELIABLE and their use is discouraged..
Installation finished. No error reported.
root@ubuntu:/home/ubuntu# 
</pre>

sebanc, 2020-05-27:
@EnzoRimapa From what I have read, there are 2 methods to have a hybrid mbr/gpt grub install:
- 1) (recommended) Create a specific partition to install the grub bootloader.
- 2) Install grub on any partition and hardcode the first bootloader block in the mbr.

For method 1, you would have needed to make changes to the partitions layout. What you are attempting when typing this is method 2. Grub installer just warns you that it is not the recommended method, so I would say it is just harmless warnings (but I cannot tell if the full command is correct).

This is actually why I chose to go with syslinux:
- method 1 requires to change the partition table layout
- method 2 is not robust

sebanc, 2020-05-27:
> For the trackpad, just nothing comes out of it. No movement, no clicks are registered, simply nothing.
> If by kernel command line you mean adding it to the grub config file, it changed nothing. If [...]

Yes to the grub config file, adding a space and "pci=noacpi" after "cros_debug" text.

I see the touchpad detected in the log file, it's weird that it does absolutely nothing, could you check if it is not blocked by a function key (FN + F<1-12>) ?

EnzoRimapa, 2020-05-27:
> @EnzoRimapa Por lo que he leído, hay 2 métodos para tener un híbrido mbr/gpt grub instalar:
> 
> * 1. (recomendado) Cree una partición específica para instalar el gestor de arranque grub.
> * 1. Instale grub en cualquier partición y codifique el primer bloque bootloader en el mbr.
> 
> Para el método 1, habría necesitado realizar cambios en el diseño de particiones. Lo que está intentando al escribir este es el método 2. Grub instalador sólo le advierte que no es el método recomendado, por lo que yo diría que es sólo advertencias inofensivas (pero no puedo decir si el comando completo es correcto).
> 
> Esta es realmente la razón por la que elegí ir con syslinux:
> 
> * método 1 requiere cambiar el diseño de la tabla de particiones
> * método 2 no es robusto

What do you recommend me to do in this case? I'm going to be honest I have 3 laptops a corei3 350m, a dual core and a core i5 5 °, in the i5 probe ramus and samus from usb and it works normal for the same reason that it is uefi and supports legacy inheritance but in that I do not want to use Chrome os but on the i3 that is older, can you do something?

sebanc, 2020-05-27:
Well, syslinux has been properly installed and I have also added the "tweaks" that are not standard but improve compatibility with bios machine. I have applied all the things I found online to make it the most compatible so I don't know how to make it boot on your laptop.

What I can suggest:
- Verify in your bios settings that booting hybrid mbr is allowed.
- Look at online resources for tips on how to get an hybrid mbr to boot and try them (I have applied all the ones I found but apparently it is not sufficient for this laptop)

Z3r0shin, 2020-05-27:
In fact, after installing Crouton, the trackpad works in Crouton, but not in Brunch.
I tried using the FN key to enable/disable the trackpad, but to no avail.
There is one thing I could try though : reinstall with Lulu, keep the stable channel r80 on and then try the fixes you mentionned. If Crouton works well, then that will be how the problems will be solved in my case.
I do hope that what I did help you too haha, I would hate to have taken up much of your time and energy for nothing!

sebanc, 2020-05-27:
No worries, i am always learning new things :) Learned that crostini gpu acceleration is not compatible with 1st gen intel and right now I just learned that installing crouton is a great way to see if the touchpad not working in chromeos is a kernel issue or a config issue.

In your case, it clearly shows that the kernel works fine so it's a chromeos touchpad configuration issue. In that case, it is very difficult to guess which config might be missing but you could try some of the options available in the below repository and see if some help:
https://github.com/hugegreenbug/xf86-input-cmt/tree/master/xorg-conf

Z3r0shin, 2020-05-27:
Will do and report!
Do you have any idea how to change flags without access to chrome://flags? More specifically the chrome://flags#scheduler-configuration thing to enable hyperthreading? That would help with playing games.

sebanc, 2020-05-27:
add the line "--scheduler-configuration=performance" to /etc/chrome_dev.conf

closed 2020-06-02 by sebanc

simplyrajatgupta, 2021-05-31:
Hello!
I visited this page before I found a new method
[https://github.com/sebanc/brunch/issues/1088]()