sebanc/brunch #188: VS Code fails to run with GPU / XServer error

keithnlarsen, 2020-05-23:
First off, thanks so much for your work, it's amazing.

I'm running this on a Dell XPS 15 i7 2-in-1 9575. It has the AMD Radeon RX Vega M GL graphics chip.

I've tried this with nami, sarien, and rammus all with the same result. I've also tried other linux apps and they work, for example Sublime Text 3.  However, this is the only app based on Atom that I've tried maybe something to do with that.

I ran code with the following flags:
`code --verbose --disable-gpu`

And got the following debug info:
[output.txt](https://github.com/sebanc/brunch/files/4670855/output.txt)

sebanc, 2020-05-23:
Hi,

Could you try blacklisting the amd gpu by booting with "module_blacklist=amdgpu" to the kernel command line ?

keithnlarsen, 2020-05-23:
> Hi,
> 
> Could you try blacklisting the amd gpu by booting with "module_blacklist=amdgpu" to the kernel command line ?

Hey that worked!

Thank you so much!

closed 2020-05-23 by keithnlarsen