sebanc/brunch #190: multi finger gestures

au79912, 2020-05-23:
Im using a dell lattitude E6440 and the track pad works fine except the usual hiccup here and there
but the three finger gestures are not work at all is this a trackpad issue or is this something that can be fixed?
thanks!

sebanc, 2020-05-23:
Hi,

I can confirm is that 3 fingers gestures are supposed to work out of the box (it works on my Surface Go) but I don't really have experience in debugging touchpad issues, all i can recommend is trying some of the configs provided there:
https://github.com/hugegreenbug/xf86-input-cmt/tree/master/xorg-conf

sebanc, 2020-05-23:
Sorry I forgot to add a bit of context, the default touchpad configuration is located in the first block of the file /etc/gesture/40-touchpad-cmt.conf:

```
Section "InputClass"
    Identifier      "touchpad"
    MatchIsTouchpad "on"
    MatchDevicePath "/dev/input/event*"
    Driver          "cmt"
    Option          "Tap Minimum Pressure" "0"
    Option          "AccelerationProfile" "-1"
    Option          "Scroll Buttons" "0"
    Option          "Scroll Axes" "1"

    # CMT devices potentially process keyboard events
    Option          "XkbModel" "pc"
    Option          "XkbLayout" "us"
EndSection
```

You can try adding some of the "Option" line you find on the below github repository (don't replace the full config as they miss some chromeos necessary features), just add the "Option" lines you want to try in this config.
https://github.com/hugegreenbug/xf86-input-cmt/tree/master/xorg-conf

If you find a config that works, could you post it here to help others with the same touchpad model ?

harun911, 2020-05-23:
Is their a command you can use to find out which config is currently being used for the touchpad?

sebanc, 2020-05-23:
No, there is no direct way but nearly certain that it is the first general config that is used, you can see that the other ones are for very specific devices (often external) identified by usb matching rules like `MatchUSBID      "XXXX:XXXX"`

hbarsaiyan, 2020-06-22:
Can confirm Three finger gestures are also not working on Dell Latitude E6320

CarlitoisLoco, 2020-08-01:
@sebanc, 3/4 finger-gestures do not seem to work on my pavilion x360 (skylake) so I tried to follow this [tutorial](https://docs.google.com/document/u/1/d/e/2PACX-1vSV4dCvvx6aZmOeMWDmrT_1GwtEJlkzRk12E8nkNL5VaesMr55Hk09neKbaHu-gsXQlM6SwKLiRnHPK/pub)  mentioned by the developer of Cloudready in a [thread](https://neverware.zendesk.com/hc/en-us/community/posts/360038407154-Touchpad-thinkpad-t470). Essentially, the steps for Synaptics devices are:
1. Disable verity
2. Reboot
3. Make system rewritable
4. Make the necessary changes to grub.efi (using vim or nano)
5. Save
6. Reboot
But I'm not able to make changes to grub.efi since it doesn't exist under /tmp/t/efi/boot/. Do you know what might be helpful here?

CarlitoisLoco, 2020-08-01:
> Can confirm Three finger gestures are also not working on Dell Latitude E6320. Seems like an issue with Alps touchpads.

Add psmouse.synaptics_intertouch=1 in grub config after options=[....]

librehat, 2020-11-17:
> > Can confirm Three finger gestures are also not working on Dell Latitude E6320. Seems like an issue with Alps touchpads.
> 
> Add psmouse.synaptics_intertouch=1 in grub config after options=[....]

I can confirm that this has fixed my issue (Lenovo Yoga 700-14ISK). On top of this, I've also added `SmoothAccel` option to the firsrt block in the file `/etc/gesture/40-touchpad-cmt.conf`.

CarlitoisLoco, 2020-11-17:
> I can confirm that this has fixed my issue (Lenovo Yoga 700-14ISK). On top of this, I've also added `SmoothAccel` option to the firsrt block in the file `/etc/gesture/40-touchpad-cmt.conf`.

that's great!

closed 2021-12-09 by au79912