sebanc/brunch #192: Unable to Boot Into Installation Media (Fails to Build)

xeu100, 2020-05-24:
Using m80 Rammus with Brunch m81 4.19 stable (latest). 

I have tried this across different brunch builds, but it fails. Everything happens so fast that I couldn't take a proper photo, these are screenshots from a slow-motion recording. They include all the details. It looks like one of the partitions fails.

[image]
[image]
[image]
[image]

sebanc, 2020-05-24:
Hi,

Thanks for the screenshots.

The first screenshot error warning can be normal and harmless if you have created the usb from windows.

The second one is the real culprit as it shows that ROOT-B partition is corrupt which is not normal.

The third one is a consequence of the second screenshot issue.

We therefore need to identify why ROOT-B partition is corrupt:
- Did you get error messages while creating the chromeos image ?
- Which tool have you used to write it on the usb flashdrive ?

xeu100, 2020-05-24:
> Hi,
>
> Thanks for the screenshots.
>
> The first screenshot error warning can be normal and harmless if you have created the usb from windows.
>
> The second one is the real culprit as it shows that ROOT-B [...]

I didn't get any error messages while compiling brunch, except for some size issues related to 512 multiplier. I used Balena Etcher and Chromebook Recovery Utility with the same images and the same results.

sebanc, 2020-05-24:
The "512 multiplier" errors are not normal, it can either mean that the recovery image was not unzipped or that you wrote to a partition instead of a block device.

closed 2020-05-24 by xeu100

xeu100, 2020-05-27:
In-case anyone experiences this issue, it was related to file validity. My download of the recovery image was corrupt, attempt a redownload.