sebanc/brunch #195: BIOS Dual boot: ChromeOS not found

XeGrox, 2020-05-24:
Hello, I'm using the dual boot method in the Linux section but I'm getting the following error:

```
brunch: Scanning device /dev/sda
brunch: Chromeos was not found on any device, fallback to shell... 
```

I do know that mbr/bios dual boot is not supported, but why is brunch searching for Chromeos partitions in my drive instead of the chromeos.img file? Thanks! 😊

sebanc, 2020-05-24:
Hi,

brunch will search for an image if both "img_part" and "img_path" variables are set in the kernel command line from the grub config.

This means that either those variables are not set in your grub config or that grub does not manage it correctly (actually my issue for dual boot with bios is that grub does not seem to handle variables the same way as the efi one)

XeGrox, 2020-05-24:
I've gotten it to work by manually setting the variables, but there are some errors that I came across

`losetup -P /mnt/loop15 /mainroot/chromeos.img` 
ends up with `loop480p*` being created instead of `loop15p*` (really weird and I have no explanation to this)

`exec switch_root /system /sbin/init` requires a `-c /dev/console` argument in order to work

I've successfully booted using the USB method without the need for any fix, so I'm not sure why these are happening

Cheers! 😁

sebanc, 2020-05-25:
There should be only one "\" in the output from the script as the "cat" command which generates it remove one.
Have you generated the script by copying it from the installer script ?

XeGrox, 2020-05-25:
Oops, the grub config was missing from the mbr install script so I copied it directly...
But there isn't an issue with the variables and I've booted just fine

sebanc, 2020-05-25:
Good to know that it worked, from which linux distro did you apply the dual boot procedure ?

XeGrox, 2020-05-25:
Arch

closed 2020-05-25 by XeGrox