sebanc/brunch #196: Power Button not Working on Asus Laptop X509FJ

Khang5687, 2020-05-24:
I managed to do triple boot on my device: Asus Laptop X509FJ i3 model. But the problem is that pressing the power button does nothing at all! Can you help me to get it to work? I used rammus recovery image version 80 and latest release of brunch. Thanks ^^

sebanc, 2020-05-24:
Hi,

Could you attach an lsmod to this issue ? (run "lsmod > ~/Downloads/lsmod.txt" in crosh shell and attach lsmod.txt from your downloads folder)

Khang5687, 2020-05-24:
[lsmod.txt](https://github.com/sebanc/brunch/files/4673616/lsmod.txt)
Here you go

sebanc, 2020-05-24:
The modules seem correctly loaded, could you try replacing "—force-tablet-power-button" by "--aura-legacy-power-button" in /etc/chrome_dev.conf ?

Khang5687, 2020-05-24:
Doesn't work my friend

tungmeister, 2020-05-24:
I'm experiencing the same on my E5480. I'd not tried it until seeing this issue. my power button does nothing other than when holding it down long enough to force kill the system.

sebanc, 2020-05-25:
In both cases the modules seem correctly loaded, could you both paste the output of "cat /proc/bus/input/devices" ?

tungmeister, 2020-05-25:
here you go.
[input.txt](https://github.com/sebanc/brunch/files/4676465/input.txt)

sebanc, 2020-05-25:
@tungmeister Could you post the output of running "sudo evtest", selecting device 1 and pressing power ?

And do the same for device 3 also.

tungmeister, 2020-05-25:
doesn't look like anything gets detected for button 1 when pressing the power button however button 3 appears to work fine:
```
Input device ID: bus 0x19 vendor 0x0 product 0x1 version 0x0
Input device name: "Power Button"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 116 (KEY_POWER)
Properties:
Testing ... (interrupt to exit)
Event: time 1590420319.427738, type 1 (EV_KEY), code 116 (KEY_POWER), value 1
Event: time 1590420319.427738, -------------- SYN_REPORT ------------
Event: time 1590420319.427748, type 1 (EV_KEY), code 116 (KEY_POWER), value 0
Event: time 1590420319.427748, -------------- SYN_REPORT ------------
```

sebanc, 2020-05-25:
Indeed input event 3 should work, could you post "/var/log/power_manager/powerd.LATEST" ?

tungmeister, 2020-05-25:
here you go:
[latest.txt](https://github.com/sebanc/brunch/files/4677929/latest.txt)

sebanc, 2020-05-25:
Well, the power button receives events and is correctly registered by the power manager, I have currently no idea why it does not work but I will keep thinking about it.

tungmeister, 2020-05-25:
@sebanc  should this `0524/230130.836175:INFO:input_watcher.cc(393)] Watching power button: /dev/input/event4` not be `event3` I don't see any reference to event3 in the powerd log, unless I'm misunderstanding?

sebanc, 2020-05-25:
You are right indeed, I got confused with event 4.

Could you post "udevadm info /sys/devices/LNXSYSTM:00/LNXPWRBN:00/input/input3" ?

tungmeister, 2020-05-25:
```
P: /devices/LNXSYSTM:00/LNXPWRBN:00/input/input3
E: DEVPATH=/devices/LNXSYSTM:00/LNXPWRBN:00/input/input3
E: EV=3
E: ID_INPUT=1
E: ID_INPUT_KEY=1
E: ID_PATH=acpi-LNXPWRBN:00
E: ID_PATH_TAG=acpi-LNXPWRBN_00
E: KEY=10000000000000 0
E: MODALIAS=input:b0019v0000p0001e0000-e0,1,k74,ramlsfw
E: NAME="Power Button"
E: PHYS="LNXPWRBN/button/input0"
E: PRODUCT=19/0/1/0
E: PROP=0
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: USEC_INITIALIZED=16298313
```

sebanc, 2020-05-25:
I see no obvious difference with mine, I will try to have a look at powerd code

tungmeister, 2020-05-25:
@sebanc are your sources available anywhere? I'm not that familiar with linux coding but nows as good a time as any to try and get some better understanding.

sebanc, 2020-05-25:
My source code is available in version branches of this repo (i.e. r81) but all I do is applying patches to the prebuilt chromeos images.

In that case I will look at the chromiumos power manager code:
https://chromium.googlesource.com/chromiumos/platform2/+/refs/heads/master/power_manager/

sebanc, 2020-05-25:
Also, which recovery image do you use ?

tungmeister, 2020-05-25:
I'm using rammus. How does brunch interface with chrome os powerd then?

sebanc, 2020-05-25:
Brunch just uses powerd as built in chromeos, I can only modify some powerd parameters (in /usr/share/power_manager/) or write complementary udev rules.

sebanc, 2020-05-25:
Could you try to run "echo 1 | sudo tee /usr/share/power_manager/board_specific/legacy_power_button" and rebooting to see if it helped ?

tungmeister, 2020-05-25:
yep, the power button now works.

sebanc, 2020-05-25:
Nice :)

@Khang5687 Could you try the same command ?

sebanc, 2020-05-25:
Just to keep history on this, ACPI power buttons are ignored unless "legacy_power_button" is set.

tungmeister, 2020-05-25:
Can this be set by default?

sebanc, 2020-05-25:
Not by default because it would probably break non-acpi power buttons but I can create an option to enable it.

tungmeister, 2020-05-25:
@sebanc I've just updated to the latest testing branch and now when I echo `echo 1 | sudo tee /usr/share/power_manager/board_specific/legacy_power_button` the power button turns off the screen instantly rather than bringing up the power menu, it appears that all it does is turn off the screen as the laptop stays on no matter how long I wait and pressing it again immediately turns the screen back on.

sebanc, 2020-05-26:
You actually need 2 things for the power menu to be displayed:
- running "echo 1 | sudo tee /usr/share/power_manager/board_specific/legacy_power_button" for the power button to be recognized.
- replacing "—force-tablet-power-button" by "--aura-legacy-power-button" in /etc/chrome_dev.conf to bring up the power menu

sebanc, 2020-05-26:
@tungmeister I have added an option "acpi_power_button" in the new testing release:
https://github.com/sebanc/brunch-testing/releases

Could you verify that it works as expected ?

tungmeister, 2020-05-26:
@sebanc Yes I can confirm it's working as expected. Also tested on another old laptop I've got and it's also working there.

sebanc, 2020-05-26:
Thanks for the feedback :)

alt_touchpad_config is a different config which seems to solve issues for some users which have either non-working touchpads or no gestures support.

Khang5687, 2020-05-26:
> Nice :)
> 
> @Khang5687 Could you try the same command ?

Thanks I will try and see if it'll work ^^

sebanc, 2020-05-26:
@Khang5687 I was not very clear in my previous comment, you actually need 2 things for the power menu to be displayed:
- running "echo 1 | sudo tee /usr/share/power_manager/board_specific/legacy_power_button" for the power button to be recognized.
- replacing "—force-tablet-power-button" by "--aura-legacy-power-button" in /etc/chrome_dev.conf to bring up the power menu

Khang5687, 2020-05-26:
It worked flawlessly! Thank you sebanc <3

closed 2020-05-26 by Khang5687

sebanc, 2020-05-26:
Great ! When the next brunch release will be out, you will just have to add "options=acpi_power_button" to the kernel command line.

SL-Pirate, 2021-03-09:
> @Khang5687 I was not very clear in my previous comment, you actually need 2 things for the power menu to be displayed:
>
> * running "echo 1 | sudo tee [...]

Can someone please tell me exactly how to do the second action. btw I did the 1st one by execution the command in crosh and it worked (turns off screen and lock) for some period but again my power button is doing nothing again. I tried executing the command again but nada. please someone help me. I'm lost.
Thank you in advance.

Khang5687, 2021-03-09:
@SL-Pirate use nano to edit the chrome_dev.conf file. More specifically, "sudo nano /etc/chrome_dev.conf"