sebanc/brunch #197: Boot without USB

EldarAron, 2020-05-24:
Hello!

Thanks for a great "hack"! I managed to install this on mye MS Surface Go 2 after a lot of try and failure. Tried all suggestions. Didn't get the internal card reader to work. Tried different SD cards with readers and USB keys.  Got different errors. 

Finally able to install it on an partition on the Go2 with my external Samsung T5 SSD. But I have to use the SSD to boot into ChromeOS.

Is there any way to boot into ChromeOS which is "installed" on a NTFS partition? Now I have to have the Samsung T5 SSD with we to get into ChromeOS .

This is the Grub code generated after "installed" ChromeOS on the NTFS partition:

> menuentry "ChromeOS (boot from disk image)" {
        img_part=/dev/nvme0n1p4
        img_path=/chromeos.img
        search --no-floppy --set=root --file $img_path
        loopback loop $img_path
        linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
        initrd (loop,gpt7)/initramfs.img
}

>

sebanc, 2020-05-25:
Hi,

This part is undocumented because I have not yet found a safe method to do this, messing with the bootloader is always a risk for the windows install.

Globally, the safest procedure is probably:
- Install refind from windows (refer to online resources)
- Copy the brunch /efi/boot folder located in partition 12 of your usb hdd to /efi/brunch on your windows efi partition where refind has been installed.

Normally refind should pick both brunch and windows and allow you to boot them.

EldarAron, 2020-05-25:
Thanks for you input.

I got it working by using refind and your tips. I found this guide on [schdck's blog](https://schdck.github.io/Installing-refind-from-Windows-10), but he recommended to try it as described on [refinds install guide](https://www.rodsbooks.com/refind/installing.html). That worked for a few reboots, then Windows did something and always booted into Windows. No boot menu. So I had to follow schdck's guide. 

If anyone follows these guides, it's at your own risk.

sebanc, 2020-05-25:
Usually windows only changes the boot priority by restoring windows bootloader as first entry in the bios instead of "internal storage" (or something like that). Restoring Internal storage as first boot priority might solve your issue.

EldarAron, 2020-05-25:
I can try that later. At least now it works. :)
Again, thanks for a great "hack" and keep up the good work! 👍 :)

closed 2020-05-25 by EldarAron