sebanc/brunch #198: No sound and microphone on Lenovo Ideapad Slim 1-11AST-05 A6

progloto, 2020-05-24:
Laptop is 
Lenovo IP Slim 11AST-05 - 81VR0024IV
image built from brunch r81 and grunt r80
volume on minimum. if I move it to any non minimum it get back to minimum
tried to connect bluetooth speaker - connected - no sound
pressing vol- put it to mute vol+ set it to minimum and not moving up

cat /proc/asound/cards
0 [HDMI           ]: HDA-Intel - HDA ATI HDMI
                     HDA ATI HDMI at 0xe8e60000 irq 40
1 [Generic        ]: HDA-Intel - HD-Audio Generic
                     HD-Audio Generic at 0xe8e64000 irq 41

$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: HDMI [HDA ATI HDMI], device 3: HDMI 0 [HDMI 0]
 Subdevices: 1/1
 Subdevice #0: subdevice #0
card 1: Generic [HD-Audio Generic], device 0: ALC257 Analog [ALC257 Analog]
 Subdevices: 1/1
 Subdevice #0: subdevice #0

What we tried with sebanc:
Try to add "snd.slots=,snd-hda-codec-hdmi" to the kernel command line and see if it changes the order of sound cards in "aplay -l"
did not helped card 0 still HDMI
try adding "module_blacklist=snd_hda_codec_hdmi" to the kernel command line, it's a bit radical but a good way to test if the issue is related to card ordering
strange but did not helped at all

 aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: HDMI [HDA ATI HDMI], device 3: Generic Digital [Generic Digital]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Generic [HD-Audio Generic], device 0: ALC257 Analog [ALC257 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

$ sudo rmmod snd_hda_codec_hdmi

rmmod: ERROR: Module snd_hda_codec_hdmi is not currently loaded
it is not loaded

ok so the blacklisting worked but it is another module responsible for the hdmi audio. Let's try something else
Try adding "snd_hda_intel.index=0" to the kernel command line
If it does not work adding "snd_hda_intel.enable=1,0" instead

$ lsmod
Module                  Size  Used by
tun                    36864  2
vhost_vsock            20480  6
vmw_vsock_virtio_transport_common    24576  1 vhost_vsock
vsock                  36864  8 vmw_vsock_virtio_transport_common,vhost_vsock
vhost                  32768  1 vhost_vsock
snd_seq_dummy          16384  0
snd_seq                69632  5 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
ccm                    20480  3
cmac                   16384  1
uinput                 20480  0
btusb                  53248  0
btrtl                  20480  1 btusb
btbcm                  16384  1 btusb
btintel                24576  1 btusb
bluetooth             458752  29 btrtl,btintel,btbcm,btusb
ecdh_generic           20480  2 bluetooth
input_polldev          16384  0
sparse_keymap          16384  0
wmi_bmof               16384  0
wmi                    24576  1 wmi_bmof
nf_nat_tftp            16384  0
nf_conntrack_tftp      16384  1 nf_nat_tftp
nf_nat_ftp             16384  0
nf_conntrack_ftp       16384  1 nf_nat_ftp
esp6                   20480  0
ah6                    16384  0
xfrm6_mode_tunnel      16384  0
xfrm6_mode_transport    16384  0
xfrm4_mode_tunnel      16384  0
xfrm4_mode_transport    16384  0
ip6t_REJECT            16384  2
ip6t_ipv6header        16384  0
joydev                 24576  0
ath10k_pci             40960  0
ath10k_core           438272  1 ath10k_pci
zram                   24576  1
mac80211              675840  1 ath10k_core
ath                    32768  1 ath10k_core
snd_hda_codec_realtek    98304  1
snd_hda_codec_generic    69632  1 snd_hda_codec_realtek
snd_hda_codec_hdmi     49152  1
cfg80211              577536  3 ath,mac80211,ath10k_core
snd_hda_intel          36864  5
snd_intel_nhlt         16384  1 snd_hda_intel
i2c_hid                28672  0
snd_hda_codec         110592  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
snd_hda_core           65536  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
snd_hwdep              16384  1 snd_hda_codec
i2c_piix4              24576  0
i915                 1474560  0
intel_gtt              20480  1 i915
video                  40960  1 i915
amdgpu               2695168  59
chash                  16384  1 amdgpu
gpu_sched              24576  1 amdgpu
ttm                    81920  1 amdgpu
input_leds             16384  0

sebanc, 2020-05-25:
Hi,

Thanks for opening the issue here.

As all kernel modules attempts to change the sound card order failed, could you try to replace the text in /etc/asound.conf with:
```
pcm.!default {
        type cras
        card 1
}
ctl.!default {
        type cras
        card 1
}
```

progloto, 2020-05-26:
Did it.
Unfortunately it didn't help.

sebanc, 2020-05-26:
Latest release link:
https://github.com/sebanc/brunch-testing/releases