sebanc/brunch #199: Intel Dual Band Wireless-AC 3165 Not working

PixelOFC, 2020-05-24:
any fixes mayb?

sebanc, 2020-05-25:
Have you tried adding "options=iwlwifi_backport" to the kernel command line ?

PixelOFC, 2020-05-25:
where and how? Im new to ChromeOS on brunch, I only tried FydeOS 2 years ago.

PixelOFC, 2020-05-25:
@sebanc ?

sebanc, 2020-05-25:
Run "sudo edit-grub-config" in crosh shell (CTRL+ALT+T and type "shell" at the invite). Then add a space and "options=iwlwifi_backport" after "cros_debug". Press CTRL+X to save, "y" and ENTER to confirm the file name.

PixelOFC, 2020-05-25:
Ill try

PixelOFC, 2020-05-25:
@sebanc Cant open crosh bcuz um stuck on the ChromeOS setup, i need to put an network but it doesn't give me any network option, already tried my RPi0 wifi dongle, and i dont have any eth cables

sebanc, 2020-05-25:
Press CTRL+ALT+F2 type "root" as user.

Run "edit-grub-config" in crosh shell. Then add a space and "options=iwlwifi_backport" after "cros_debug". Press CTRL+X to save, "y" and ENTER to confirm the file name. Reboot

PixelOFC, 2020-05-25:
now the networks appear for minus than one sec

PixelOFC, 2020-05-25:
@sebanc i hate my crappy lenovo

sebanc, 2020-05-25:
Press CTRL+ALT+F2, then type "root" as user.

What does "dmesg | grep firmware" returns ? (you can attach a photo if it is easier)

PixelOFC, 2020-05-25:
[image]
Image ^^
and im uploading an video on YT showing what happen (exactly) btw soory for my english, im brazillian

sebanc, 2020-05-25:
what is the output of "rfkill list all" ?

PixelOFC, 2020-05-25:
https://www.youtube.com/watch?v=E19PLDCTDp8 vid

sebanc, 2020-05-25:
Ok, I definitely need the output of "rfkill list all"

PixelOFC, 2020-05-25:
@sebanc image
[image]

sebanc, 2020-05-25:
run "rfkill unblock all" and then press CTRL+ALT+F1 to return to the main screen, you should see the networks

PixelOFC, 2020-05-25:
ok

PixelOFC, 2020-05-25:
god bless you, im writing this on my Chrome OS, how many years you work on this

sebanc, 2020-05-25:
Probably around one year but chromeos is mainly linux in the end and I use linux since longer than that.

Which brunch version did you use ?
Could you attach an "lsmod" to this issue ? (open crosh with CTRL+ALT+T, type "shell", run "lsmod > ~/Downloads/lsmod.txt" and attach lsmod.txt from your downloads folder to this issue)

sebanc, 2020-05-25:
I don't think you want to close the issue yet, your wifi will still be blocked after each reboot

PixelOFC, 2020-05-25:
[lsmod.txt](https://github.com/sebanc/brunch/files/4678287/lsmod.txt)
here it is

sebanc, 2020-05-25:
You used brunch r80 stable ?

PixelOFC, 2020-05-25:
yes

PixelOFC, 2020-05-25:
im feeling an lag, that is normal?

sebanc, 2020-05-25:
I think the wifi issue should be solved if you update to r81 stable, download it and run "sudo chromeos-update -f <path_to_r81_release_package>".

The lag might be normal if you run it on usb or if it is still setting up android apps.

PixelOFC, 2020-05-25:
i was doing some stuff, ill try rn

PixelOFC, 2020-05-25:
i feel lag on the location when u open apps (idk name in english)

sebanc, 2020-05-25:
Try installing it on hdd or dual boot procedure, the experience is not very good on usb

PixelOFC, 2020-05-25:
it is triple booted

sebanc, 2020-05-25:
What's your laptop model ?

PixelOFC, 2020-05-25:
lenovo ideapad 320 80yh0008br, updated Brunch too

PixelOFC, 2020-05-25:
btw lag is only on animations

sebanc, 2020-05-25:
Your laptop should be powerful enough, I have no specific tweaks to recommend.

However, chromeos animations are maybe not as smooth as you would have expected, you can look for ways to disable them on chromeos reddit/forums.

PixelOFC, 2020-05-25:
ty, should i close this now? Wifi problem fixed, ill try changing my region to US bcuz google assistant isnt avaliable in PT-BR

sebanc, 2020-05-25:
Yes, I think it can be closed, you can still re-open it if you notice other issues.

closed 2020-05-25 by sebanc

PixelOFC, 2020-05-26:
@sebanc can i update using settings app?