sebanc/brunch #200: No microphone on Dell Venue 11 Pro 7140

kbnhvn, 2020-05-25:
I have sound but no microphone on Dell Venue 11 Pro 7140
Latest Brunch and Rammus r80

cat /proc/asound/cards
 0 [HDMI           ]: HDA-Intel - HDA Intel HDMI
                      HDA Intel HDMI at 0xf7218000 irq 45
 1 [broadwellrt286 ]: broadwell-rt286 - broadwell-rt286
                      broadwell-rt286

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: HDMI [HDA Intel HDMI], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: broadwellrt286 [broadwell-rt286], device 0: System Playback/Capture (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: broadwellrt286 [broadwell-rt286], device 1: Offload0 Playback (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: broadwellrt286 [broadwell-rt286], device 2: Offload1 Playback (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

lsmod
Module                  Size  Used by
hid_logitech_hidpp     32768  0
hid_logitech_dj        20480  0
snd_seq_dummy          16384  0
snd_seq                69632  5 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
cmac                   16384  1
rfcomm                 61440  4
uinput                 20480  0
wl                   6377472  0
btusb                  53248  0
btrtl                  20480  1 btusb
btbcm                  16384  1 btusb
btintel                24576  1 btusb
bluetooth             458752  34 btrtl,btintel,btbcm,btusb,rfcomm
ecdh_generic           20480  2 bluetooth
ccm                    20480  3
intel_rapl             20480  0
x86_pkg_temp_thermal    16384  0
intel_powerclamp       16384  0
dell_laptop            24576  0
dell_wmi               16384  0
dell_smbios            16384  2 dell_wmi,dell_laptop
wmi_bmof               16384  0
dell_wmi_descriptor    16384  2 dell_wmi,dell_smbios
snd_soc_sst_broadwell    20480  3
snd_soc_sst_haswell_pcm    73728  2 snd_soc_sst_broadwell
snd_soc_sst_dsp        16384  1 snd_soc_sst_haswell_pcm
snd_soc_sst_ipc        16384  1 snd_soc_sst_haswell_pcm
snd_soc_sst_firmware    24576  1 snd_soc_sst_haswell_pcm
wmi                    24576  4 dell_wmi,wmi_bmof,dell_smbios,dell_wmi_descriptor
intel_vbtn             16384  0
soc_button_array       16384  0
sparse_keymap          16384  2 dell_wmi,intel_vbtn
snd_soc_sst_acpi       16384  0
snd_soc_acpi_intel_match    28672  1 snd_soc_sst_acpi
snd_soc_acpi           16384  2 snd_soc_sst_acpi,snd_soc_acpi_intel_match
acpi_pad               16384  0
nf_nat_tftp            16384  0
nf_conntrack_tftp      16384  1 nf_nat_tftp
nf_nat_ftp             16384  0
nf_conntrack_ftp       16384  1 nf_nat_ftp
esp6                   20480  0
ah6                    16384  0
xfrm6_mode_tunnel      16384  0
xfrm6_mode_transport    16384  0
xfrm4_mode_tunnel      16384  0
xfrm4_mode_transport    16384  0
ip6t_REJECT            16384  2
ip6t_ipv6header        16384  0
iio_trig_sysfs         16384  2
hid_rmi                16384  0
rmi_core               73728  1 hid_rmi
hid_sensor_cros_compat    16384  2
industrialio_triggered_buffer    16384  1 hid_sensor_cros_compat
kfifo_buf              16384  2 industrialio_triggered_buffer,hid_sensor_cros_compat
hid_sensor_iio_common    16384  1 hid_sensor_cros_compat
industrialio           57344  6 industrialio_triggered_buffer,kfifo_buf,iio_trig_sysfs,hid_sensor_cros_compat
snd_soc_rt298          45056  0
iwlmvm                376832  0
hid_sensor_hub         16384  2 hid_sensor_iio_common,hid_sensor_cros_compat
mac80211              675840  1 iwlmvm
snd_hda_codec_hdmi     49152  1
mei_wdt                16384  0
snd_hda_intel          36864  2
snd_intel_nhlt         16384  1 snd_hda_intel
iwlwifi               274432  1 iwlmvm
snd_hda_codec         110592  2 snd_hda_codec_hdmi,snd_hda_intel
zram                   24576  1
snd_hda_core           65536  3 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec
snd_soc_rt286          45056  2 snd_soc_sst_broadwell
mei_me                 24576  1
snd_hwdep              16384  1 snd_hda_codec
cfg80211              577536  4 wl,iwlmvm,iwlwifi,mac80211
snd_soc_rl6347a        16384  2 snd_soc_rt298,snd_soc_rt286
mei                    65536  3 mei_wdt,mei_me
i2c_hid                28672  0
i915                 1474560  36
intel_gtt              20480  1 i915
video                  40960  3 dell_wmi,dell_laptop,i915
joydev                 24576  0
input_leds             16384  0

sebanc, 2020-05-25:
Hi,

Could you try to replace the "broadwell-rt286" in /usr/share/alsa/ucm/ with the below one and see if that helps ?
[broadwell-rt286.zip](https://github.com/sebanc/brunch/files/4676826/broadwell-rt286.zip)

kbnhvn, 2020-05-25:
Hi,

Still no microphone, sound works well.

PixelOFC, 2020-05-25:
Congrats u got issue #200

sebanc, 2020-05-25:
Just saving the link for later use:
https://bugzilla.kernel.org/show_bug.cgi?id=150601

PixelOFC, 2020-05-25:
there are still people using Broadwell, lmao

kbnhvn, 2020-05-25:
I just tried a BIOS update, and sound is not working anymore...

sebanc, 2020-05-26:
Hi, I used the patch which is mentioned in my above comment in this testing build. Could you give it a try ?
https://github.com/sebanc/brunch-testing/releases

(The build is in a private repo, you should have received an email from github to validate the access)

kbnhvn, 2020-05-26:
Hi, thanks ! I will try.

kbnhvn, 2020-05-26:
Sound is working, but mic doesn't.

cat /proc/asound/cards
 0 [HDMI           ]: HDA-Intel - HDA Intel HDMI
                      HDA Intel HDMI at 0xf7218000 irq 51
 1 [broadwellrt286 ]: broadwell-rt286 - broadwell-rt286
                      broadwell-rt286

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: HDMI [HDA Intel HDMI], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: broadwellrt286 [broadwell-rt286], device 0: System Playback/Capture (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: broadwellrt286 [broadwell-rt286], device 1: Offload0 Playback (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: broadwellrt286 [broadwell-rt286], device 2: Offload1 Playback (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

lsmod
Module                  Size  Used by
snd_seq_dummy          16384  0
snd_seq                69632  5 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
tcp_diag               16384  0
inet_diag              20480  1 tcp_diag
rfcomm                 61440  4
cmac                   16384  1
uinput                 20480  0
nf_nat_tftp            16384  0
nf_conntrack_tftp      16384  1 nf_nat_tftp
nf_nat_ftp             16384  0
nf_conntrack_ftp       16384  1 nf_nat_ftp
esp6                   20480  0
ah6                    16384  0
xfrm6_mode_tunnel      16384  0
xfrm6_mode_transport    16384  0
xfrm4_mode_tunnel      16384  0
xfrm4_mode_transport    16384  0
ip6t_REJECT            16384  2
ip6t_ipv6header        16384  0
hid_rmi                16384  0
rmi_core               73728  1 hid_rmi
wl                   6377472  0
intel_rapl             20480  0
x86_pkg_temp_thermal    16384  0
intel_powerclamp       16384  0
crct10dif_pclmul       16384  0
btusb                  53248  0
crc32_pclmul           16384  0
btrtl                  20480  1 btusb
ghash_clmulni_intel    16384  0
btbcm                  16384  1 btusb
btintel                24576  1 btusb
pcbc                   16384  0
bluetooth             458752  34 btrtl,btintel,btbcm,btusb,rfcomm
ecdh_generic           20480  2 bluetooth
aesni_intel           200704  7690
aes_x86_64             20480  1 aesni_intel
crypto_simd            16384  1 aesni_intel
cryptd                 20480  3847 crypto_simd,ghash_clmulni_intel,aesni_intel
glue_helper            16384  1 aesni_intel
snd_soc_sst_broadwell    20480  3
wmi_bmof               16384  0
dell_wmi               16384  0
snd_soc_sst_haswell_pcm    73728  2 snd_soc_sst_broadwell
snd_soc_sst_dsp        16384  1 snd_soc_sst_haswell_pcm
snd_soc_sst_ipc        16384  1 snd_soc_sst_haswell_pcm
dell_laptop            24576  0
snd_soc_sst_firmware    24576  1 snd_soc_sst_haswell_pcm
dell_smbios            16384  2 dell_wmi,dell_laptop
dell_wmi_descriptor    16384  2 dell_wmi,dell_smbios
serio_raw              16384  0
wmi                    24576  4 dell_wmi,wmi_bmof,dell_smbios,dell_wmi_descriptor
intel_vbtn             16384  0
sparse_keymap          16384  2 dell_wmi,intel_vbtn
soc_button_array       16384  0
dw_dmac                16384  0
snd_soc_sst_acpi       16384  0
snd_soc_acpi_intel_match    28672  1 snd_soc_sst_acpi
snd_soc_acpi           16384  2 snd_soc_sst_acpi,snd_soc_acpi_intel_match
acpi_pad               16384  0
ccm                    20480  5
iio_trig_sysfs         16384  2
hid_sensor_cros_compat    16384  2
industrialio_triggered_buffer    16384  1 hid_sensor_cros_compat
iwlmvm                376832  0
kfifo_buf              16384  2 industrialio_triggered_buffer,hid_sensor_cros_compat
hid_sensor_iio_common    16384  1 hid_sensor_cros_compat
industrialio           57344  6 industrialio_triggered_buffer,kfifo_buf,iio_trig_sysfs,hid_sensor_cros_compat
mac80211              770048  1 iwlmvm
hid_sensor_hub         16384  2 hid_sensor_iio_common,hid_sensor_cros_compat
snd_hda_codec_hdmi     49152  1
snd_soc_rt298          45056  0
snd_hda_intel          36864  2
hid_logitech_hidpp     32768  0
iwlwifi               274432  1 iwlmvm
snd_intel_nhlt         16384  1 snd_hda_intel
zram                   24576  1
snd_hda_codec         110592  2 snd_hda_codec_hdmi,snd_hda_intel
mei_wdt                16384  0
snd_hda_core           65536  3 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec
snd_soc_rt286          45056  2 snd_soc_sst_broadwell
mei_me                 24576  1
snd_hwdep              16384  1 snd_hda_codec
snd_soc_rl6347a        16384  2 snd_soc_rt298,snd_soc_rt286
i2c_hid                28672  0
mei                    65536  3 mei_wdt,mei_me
hid_logitech_dj        20480  0
cfg80211              577536  4 wl,iwlmvm,iwlwifi,mac80211
i915                 1474560  30
intel_gtt              20480  1 i915
video                  40960  3 dell_wmi,dell_laptop,i915
joydev                 24576  0
input_leds             16384  0
mac_hid                16384  0

kbnhvn, 2020-05-26:
So sound works but mic doesn't.

cat /proc/asound/cards
 0 [HDMI           ]: HDA-Intel - HDA Intel HDMI
                      HDA Intel HDMI at 0xf7218000 irq 51
 1 [broadwellrt286 ]: broadwell-rt286 - broadwell-rt286
                      broadwell-rt286

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: HDMI [HDA Intel HDMI], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: broadwellrt286 [broadwell-rt286], device 0: System Playback/Capture (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: broadwellrt286 [broadwell-rt286], device 1: Offload0 Playback (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: broadwellrt286 [broadwell-rt286], device 2: Offload1 Playback (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

lsmod
Module                  Size  Used by
snd_seq_dummy          16384  0
snd_seq                69632  5 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
tcp_diag               16384  0
inet_diag              20480  1 tcp_diag
rfcomm                 61440  4
cmac                   16384  1
uinput                 20480  0
nf_nat_tftp            16384  0
nf_conntrack_tftp      16384  1 nf_nat_tftp
nf_nat_ftp             16384  0
nf_conntrack_ftp       16384  1 nf_nat_ftp
esp6                   20480  0
ah6                    16384  0
xfrm6_mode_tunnel      16384  0
xfrm6_mode_transport    16384  0
xfrm4_mode_tunnel      16384  0
xfrm4_mode_transport    16384  0
ip6t_REJECT            16384  2
ip6t_ipv6header        16384  0
hid_rmi                16384  0
rmi_core               73728  1 hid_rmi
wl                   6377472  0
intel_rapl             20480  0
x86_pkg_temp_thermal    16384  0
intel_powerclamp       16384  0
crct10dif_pclmul       16384  0
crc32_pclmul           16384  0
ghash_clmulni_intel    16384  0
pcbc                   16384  0
btusb                  53248  0
btrtl                  20480  1 btusb
btbcm                  16384  1 btusb
btintel                24576  1 btusb
bluetooth             458752  34 btrtl,btintel,btbcm,btusb,rfcomm
ecdh_generic           20480  2 bluetooth
aesni_intel           200704  9254
aes_x86_64             20480  1 aesni_intel
crypto_simd            16384  1 aesni_intel
cryptd                 20480  4629 crypto_simd,ghash_clmulni_intel,aesni_intel
glue_helper            16384  1 aesni_intel
snd_soc_sst_broadwell    20480  3
wmi_bmof               16384  0
dell_wmi               16384  0
snd_soc_sst_haswell_pcm    73728  2 snd_soc_sst_broadwell
snd_soc_sst_dsp        16384  1 snd_soc_sst_haswell_pcm
snd_soc_sst_ipc        16384  1 snd_soc_sst_haswell_pcm
snd_soc_sst_firmware    24576  1 snd_soc_sst_haswell_pcm
dell_laptop            24576  0
dell_smbios            16384  2 dell_wmi,dell_laptop
dell_wmi_descriptor    16384  2 dell_wmi,dell_smbios
serio_raw              16384  0
wmi                    24576  4 dell_wmi,wmi_bmof,dell_smbios,dell_wmi_descriptor
soc_button_array       16384  0
intel_vbtn             16384  0
sparse_keymap          16384  2 dell_wmi,intel_vbtn
snd_soc_sst_acpi       16384  0
snd_soc_acpi_intel_match    28672  1 snd_soc_sst_acpi
snd_soc_acpi           16384  2 snd_soc_sst_acpi,snd_soc_acpi_intel_match
dw_dmac                16384  0
acpi_pad               16384  0
ccm                    20480  5
iio_trig_sysfs         16384  2
iwlmvm                376832  0
hid_sensor_cros_compat    16384  2
industrialio_triggered_buffer    16384  1 hid_sensor_cros_compat
kfifo_buf              16384  2 industrialio_triggered_buffer,hid_sensor_cros_compat
hid_sensor_iio_common    16384  1 hid_sensor_cros_compat
mac80211              770048  1 iwlmvm
industrialio           57344  6 industrialio_triggered_buffer,kfifo_buf,iio_trig_sysfs,hid_sensor_cros_compat
snd_hda_codec_hdmi     49152  1
snd_hda_intel          36864  2
hid_sensor_hub         16384  2 hid_sensor_iio_common,hid_sensor_cros_compat
snd_soc_rt298          45056  0
snd_intel_nhlt         16384  1 snd_hda_intel
hid_logitech_hidpp     32768  0
iwlwifi               274432  1 iwlmvm
snd_hda_codec         110592  2 snd_hda_codec_hdmi,snd_hda_intel
mei_wdt                16384  0
zram                   24576  1
snd_hda_core           65536  3 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec
mei_me                 24576  1
snd_soc_rt286          45056  2 snd_soc_sst_broadwell
snd_hwdep              16384  1 snd_hda_codec
mei                    65536  3 mei_wdt,mei_me
hid_logitech_dj        20480  0
cfg80211              577536  4 wl,iwlmvm,iwlwifi,mac80211
i2c_hid                28672  0
snd_soc_rl6347a        16384  2 snd_soc_rt298,snd_soc_rt286
i915                 1474560  24
intel_gtt              20480  1 i915
video                  40960  3 dell_wmi,dell_laptop,i915
joydev                 24576  0
input_leds             16384  0
mac_hid                16384  0

kbnhvn, 2020-05-26:
Tried BIOS Update, I have sound, no mic.

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: HDMI [HDA Intel HDMI], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: HDMI [HDA Intel HDMI], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: broadwellrt286 [broadwell-rt286], device 0: System Playback/Capture (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: broadwellrt286 [broadwell-rt286], device 1: Offload0 Playback (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: broadwellrt286 [broadwell-rt286], device 2: Offload1 Playback (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

sebanc, 2020-05-27:
Ok, It seems that the soundcard is correctly managed by the kernel now. Unfortunately for the microphone, I think you have the same issue as #56. Could you try to apply similar tweaks to the ones mentioned there ?

kbnhvn, 2020-05-27:
chronos@localhost / $ amixer scontrols
Simple mixer control 'Master',0
Simple mixer control 'Capture',0
chronos@localhost / $ amixer -c0 set "Capture" toggle
amixer: Unable to find simple control 'Capture',0

chronos@localhost / $ amixer set "Capture" toggle
Simple mixer control 'Capture',0
  Capabilities: cvolume cvolume-joined cswitch cswitch-joined
  Capture channels: Mono
  Limits: Capture 0 - 100
  Mono: Capture 0 [0%] [off]
chronos@localhost / $ amixer set "Capture" toggle
Simple mixer control 'Capture',0
  Capabilities: cvolume cvolume-joined cswitch cswitch-joined
  Capture channels: Mono
  Limits: Capture 0 - 100
  Mono: Capture 0 [0%] [on]
chronos@localhost / $ amixer set "Capture" 40% 
Simple mixer control 'Capture',0
  Capabilities: cvolume cvolume-joined cswitch cswitch-joined
  Capture channels: Mono
  Limits: Capture 0 - 100
  Mono: Capture 40 [40%] [on]
chronos@localhost / $ amixer
Simple mixer control 'Master',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback 0 - 100
  Mono: Playback 67 [67%] [on]
Simple mixer control 'Capture',0
  Capabilities: cvolume cvolume-joined cswitch cswitch-joined
  Capture channels: Mono
  Limits: Capture 0 - 100
  Mono: Capture 0 [0%] [on]

kbnhvn, 2020-05-27:
When I set Capture to 40%, it goes back to 0%

kbnhvn, 2020-05-27:
I found this: https://github.com/alsa-project/alsa-ucm-conf/commit/98c6eabdfb0c98350bdaa1f491935a86fd9872da

Already used in your files?

sebanc, 2020-05-27:
Actually I currently let the standard alsa ucm files that come with chromeos and just add missing ones. In your case, the config is one that comes with chromeos and it appears to not have the patch you found here: https://github.com/alsa-project/alsa-ucm-conf/commit/98c6eabdfb0c98350bdaa1f491935a86fd9872da

Try to replace the files and see if it helps, if it does I can add them by default in brunch.

kbnhvn, 2020-05-27:
Will I have to rename HiFi.conf to HiFi in file name and code?

sebanc, 2020-05-27:
If you replace both files in "broadwell-rt286" with the ones below it should work:
https://github.com/alsa-project/alsa-ucm-conf/tree/master/ucm2/broadwell-rt286

kbnhvn, 2020-05-27:
I replaced both files, now no sound no mic

sebanc, 2020-05-27:
Could you try the config in comment 22 from https://bugzilla.kernel.org/show_bug.cgi?id=150601 ?

kbnhvn, 2020-05-27:
Which are the original/changes? right or left column?

sebanc, 2020-05-27:
Left is old, right is new

kbnhvn, 2020-05-27:
It's already like right column

sebanc, 2020-05-27:
Ok, nevermind then.

Unfortunately I am not very good at debugging audio issues and I don't really know what else to suggest now that the kernel patch is included and that the different ucm files available have been tested.

kbnhvn, 2020-05-27:
Ok, thanks a lot for your help.
I didn't tried 
"sudo nano /etc/modprobe.d/alsa.conf
Then add the following at the end of the file:
options snd-hda-intel model=generic"
from the other issue thread, do you think it can helps?

sebanc, 2020-05-27:
Yes, I think It's worth testing

kbnhvn, 2020-05-27:
Didn't work.
So i revert all files from your testing release as sound works on it at least.
Thanks a lot for helping !

closed 2026-04-27 by github-actions[bot]