sebanc/brunch #201: No audio on bytcr-rt5640 (Baytrail Tablet)

intialonso, 2020-05-25:
I was able to install Chrome OS with Rammus in this old Windows tablet (Lenovo Miix 2 8), everything seems to work except the audio. The chip is a "bytcr-rt5640".

Already tried booting with the "options=baytrail_chromebook" and there was no change. The sound works ok in Android X86 9 with kernel 4.19.110, and in Ubuntu 20.04 with kernel 5.4.

This is my aplay -l output:
[image]

clacalderonc, 2020-05-26:
Hi, i'm running the same setup (miix 2 8), but using octopus recovery (i did find it works stable with this build) and sound is working ok, maybe you cant try it.

intialonso, 2020-05-26:
> 
> 
> Hi, i'm running the same setup (miix 2 8), but using octopus recovery (i did find it works stable with this build) and sound is working ok, maybe you cant try it.

Ey thats great! Gonna try right now, thank you!

intialonso, 2020-05-26:
> Hi, i'm running the same setup (miix 2 8), but using octopus recovery (i did find it works stable with this build) and sound is working ok, maybe you cant try it.

Just finished installing with Octopus and still no sound. What Brunch version did you usted? Did you make anything else?

clacalderonc, 2020-05-26:
> > Hi, i'm running the same setup (miix 2 8), but using octopus recovery (i did find it works stable with this build) and sound is working ok, maybe you cant try it.
> 
> Just finished installing with Octopus and still no sound. What Brunch version did you usted? Did you make anything else?

No, just installed and worked without changes, I did some testing to have more details:

- Tablet speaker output audio works OK.
- Input audio jack detect earphones but i dont get output/input working.
- Bluetooth Audio device (airpods) output works but i could get the input mic audio working.

intialonso, 2020-05-27:
I was able to fix it removing all the bytcr-rt5640 files and folders from /her/share/alsa/ucm and using the ones from https://github.com/plbossart/, the UCM and asound.state, like this:

sudo cp -rf ~/UCM/bytcr-rt5640 /usr/share/alsa/ucm
sudo cp ~/UCM/bytcr-rt5651/asound.state /var/lib/alsa

With this I have speaker sound, but no headphones.

kbnhvn, 2020-05-27:
How it is running on BayTrail tablet?
I've got an old Miix 3 830 not used.

intialonso, 2020-05-27:
> 
> 
> How it is running on BayTrail tablet?
> I've got an old Miix 3 830 not used.

Well, it is stable, works much better than on Windows. Some things doesnt work as the audio headphones, autorotate and the cameras. Those things except the cameras work fine on Android X86, but is more unstable. For media consumption is a viable choice.

kbnhvn, 2020-05-27:
Thanks, Android app work too?
For rotation, maybe this app will help :
https://github.com/azw413/ChromeOSRotate

intialonso, 2020-05-27:
> 
> 
> Thanks, Android app work too?
> For rotation, maybe this app will help :
> https://github.com/azw413/ChromeOSRotate

Yeah, Android apps are working (I played some games), and you can rotate from the settings. no need for an app, its juts that it doesnt do it automatically.

kbnhvn, 2020-05-27:
No issue with 32 bit uefi / 64 bit CPU of baytrail? For installation

intialonso, 2020-05-27:
> 
> 
> No issue with 32 bit uefi / 64 bit CPU of baytrail? For installation

Not at all.

kbnhvn, 2020-06-09:
What about battery life?

tornadox, 2020-07-10:
I have a TV box Wintel CX-8, it has the same baytrail CPU Z3735, audio is bytcr-rt5640. 
HDMI - doesn't not work, only crackle 
Headphone - output doesn't work
Mic - doesn't work
No internal speaker to test.
None of the fixes above helped, rammus & octopus, baytrail_chromebook, disable_intel_hda, replacing files from above link also.
Every sound output works fine on xubuntu 20 (hdmi, headphone and mic)

tornadox, 2020-07-11:
I tried to edit the baytrail_chromebook patch but it didn't help. The device performs really well on chrome os in comparison to ubuntu

`# Specific patch to allow sound on baytrail chromebooks with the option "baytrail_chromebook"

baytrail_chromebook=0
for i in $(echo "$1" | sed 's#,# #g')
do
        if [ "$i" == "baytrail_chromebook" ]; then baytrail_chromebook=1; fi
done

ret=0
if [ "$baytrail_chromebook" -eq 1 ]; then
        echo "blacklist snd_hda_intel" > /system/etc/modprobe.d/alsa-hda.conf
        if [ ! "$?" -eq 0 ]; then ret=$((ret + (2 ** 0))); fi
        cat >/system/etc/init/bytcr-rt5640.conf <<BYTCR-RT5640
start on stopped udev-trigger

script
        alsaucm -c bytcr-rt5640 set _verb HiFi set _enadev Headphone
        alsaucm -c bytcr-rt5640 set _verb HiFi set _enadev Speakers
        alsaucm -c bytcr-rt5640 set _verb HiFi set _enadev HeadsetMic
        alsaucm -c bytcr-rt5640 set _verb HiFi set _enadev InternalMic
        mkdir -p /var/lib/alsa
        cp /usr/share/alsa/ucm/bytcr-rt5651/asound.state /var/lib/alsa/
        alsactl restore
end script
BYTCR-RT5640
        if [ ! "$?" -eq 0 ]; then ret=$((ret + (2 ** 1))); fi
fi`

DevHamid, 2020-09-09:
> I was able to fix it removing all the bytcr-rt5640 files and folders from /usr/share/alsa/ucm and using the ones from https://github.com/plbossart/, the UCM and asound.state, like this:
> 
> sudo cp -rf ~/UCM/bytcr-rt5640 /usr/share/alsa/ucm
> sudo cp ~/UCM/bytcr-rt5651/asound.state /var/lib/alsa (needed to create that folder)
> 
> With this I have speaker sound, but no headphones.

i try on my bytcr-rt5651 but still no sound, only on tws works